WouterJD/FortiusANT #162: Not able to pair with "Controllable" in ZWIFT

Sat2Freak, 2020-12-10:
Hello Wouter,

I have a working TTS 3.10 with a Tacx Fortius (Head unit T1932)

When starting FortiusAnt, everything seems to go well.
Connected to Tacx Trainer T1932
FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipment device (FE-C), id=57591
FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), ID+57596
FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595

However, when I want to pair with ZWIFT, I only get the Powermeter (57596), Heart Rate, and Cadence (57595) paired.
The Controllable (FE-C) device is not found...

What am I doing wrong?
Is there anything I forgot along the way?

Kind regards from Belgium,
Rudy.

rcerrie, 2020-12-10:
@Sat2Freak, I have exactly the same question. Basically I have the same setup (T1932 head), I just run 2x ANT+ dongles on one laptop. Did you do a test on Zwift? After the initial pairing etc. I did a 5km ride and could feel resistance when doing some inclines and easing out when leveling out on Watopia. So it seems to "work" but not sure about the controllable not showing. @WouterJD

WouterJD, 2020-12-10:
@Sat2Freak and @rcerrie Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

At first sight I would say that FortiusANT broadcasts the signals for powermeter AND FE-C at the same time, there is option to do one and not the other. If I recall well, there are two ways to pair in Zwift: using PowerMeter OR FE-C. Please check there.

If it stilll does not work, run FortiusANT with -d127 flag to crete logfile and add it to this issue.

Sat2Freak, 2020-12-10:
Hi Wouter,

There is no way to connect to the FE-C, even if all sensors are unpaired.  So that's not the right direction.
Attached, you'll find the log file you asked for.
Thanks in advance!
[fortius.txt](https://github.com/WouterJD/FortiusANT/files/5671832/fortius.txt)

10:01:28,922: FortiusANT started
10:01:28,922: -d 127 (0b1111111)
10:01:28,922: -H None
10:01:28,922: -p 1.0
10:01:28,922: ------------------
10:01:29,156: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
10:01:30,159: Dongle - No (free) ANT-dongle found
10:01:30,190: Connected to Tacx Trainer T1932
10:01:30,190: Tacx   - Connected to Tacx Trainer T1932
10:01:30,190: AntHRM - Heartrate expected from Tacx Trainer
10:01:30,190: FortiusAnt ended

Kind regards,
Rudy.

WouterJD, 2020-12-10:
10:01:30,159: Dongle - No (free) ANT-dongle found

This is another ussue; the dongle is in use by another application; take care that Garmin is not running.
Otherwise, remove dongle and re-insert and retry

rcerrie, 2020-12-10:
@Sat2Freak How did you run the "FortiusANT with -d127 flag" option to create the logfile? Sorry I am very new to this.

WouterJD, 2020-12-10:
@rcerrie the manual should explain this; please have a look

rcerrie, 2020-12-10:
> 
> 
> @rcerrie the manual should explain this; please have a look

Got it thanx

Sat2Freak, 2020-12-10:
Hello Wouter,

I have only one dongle on that computer.  That's why I connect with my smartphone to Zwift.
I can find the speed en the power sensor, so the dongle is doing what should be done.
Strange that another dongle should be necessary to broadcast the FE-C signal.

I do understand that I can't link ZWIFT without a second dongle, on that same laptop, but that is not the issue here.

Hope you understand what I mean,
Otherwise, shoot...

Thanks,

Rudy.

WouterJD, 2020-12-10:
One dongle for FortiusANT; one dongle for Zwift <IF ON PC> so one dongle needed

Sat2Freak, 2020-12-10:
Wouter,

And that's what I have.

GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)

So the problem seems to be elsewhere.  Strange this isn't an issue with more people.
Could it be the dongle isn't fully compatible?

Rudy.

WouterJD, 2020-12-10:
> 10:01:29,156: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
> 10:01:30,159: Dongle - No (free) ANT-dongle found

The dongle type is good (unless broken of course)
But the dongle is in use by another process.

Check that no other processes are running (Garmin Express is a notorious invisible one)
Reboot to be sure

Sat2Freak, 2020-12-10:
Seems there is no solution.
Normal behaviour as FortiusANT is concerned, only - No pairing possible with ZWIFT;

And the strangest thing is - when searching for sensor  with my Garmin Edge 520 - The indoor trainer is found immediately (ID 57591)
[fortiusant.txt](https://github.com/WouterJD/FortiusANT/files/5672735/fortiusant.txt)

Any Ideas?  Not with FortiusANT I presume...

WouterJD, 2020-12-10:
@Sat2Freak I think you're too fast. If Garmin can see FortiusAnt, another application could. I cannot answer whether Zwift/android works; I do not have that hardware.

Succes in further getting this combination work!

switchabl, 2020-12-10:
Unfortunately, it appears that Zwift on Android does not support ANT FE-C: https://forums.zwift.com/t/ant-update/195323/119

@WouterJD Something to add to the manual?

Some options you might have:
- get a second dongle and install Zwift on a PC/laptop (avoid "CYCPLUS", "ANSELF" and similar looking dongles, they cause only problems; best is another Dynastream like you have, they are sold under many brands: Garmin, LifeLine, hLine, Saris, Elite, Rotor, Wahoo, CycleOps among others. The older ones labelled Garmin, Suunto or Tacx are fine too if you can get a cheap used one).
- get a North Pole Engineering CABLE, that is a small box that translates the ANT signal to Bluetooth
- give Rouvy a try, Rouvy seems to support 

Also you should probably complain to Zwift about this and tell them this is keeping you from using (and paying for) their service. If enough people do so, they might improve that.

WouterJD, 2020-12-10:
I added a note to the manual.
Unfortunately for you guys; but I think we can close this issue?

Sat2Freak, 2020-12-10:
Thanks a lot for thinking with me.  I must admit I learned a lot.
This issue can be closed, as it's a problem than needs to be solved by Zwift, not by FortiusANT.

Wouter, thanks again for your fantastic work, and I guess a second dongle will be the simpliest solution ;)

closed 2020-12-10 by Sat2Freak