WouterJD/FortiusANT #163: Dongle types: Suunto vs Anself/Dynastream

BrRoBo, 2020-12-10:
Opening new issue for discussion or perhaps further investigation...

I have 2 dongles of which one is a Suunto Movestick Mini I purchased maybe in 2016, and the second dongle is a new **Anself USB ANT+ Stick which is reported by FortiusAnt as a Dynastream** as confirmed by this debug output:

`16:54:58,970: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)`

This is the Anself/Dynastream stick purchase source from Amazon...
[https://www.amazon.com/gp/product/B01M3VQP6Z/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1](url)  

In my setup, I run one Windows 10 PC with these 2 dongles inserted in a USB hub.  

The order of the dongles in the USB ports seems to matter.  If the Suunto is picked up first and used by FortiusAnt (reported as being used in the GUI), then in Zwift I assume it utilizes the Anself/Dynastream and all the Ant+ devices show up ok in the pairing screen, including FE-C Controllable.  

However, if the dongle order is reversed in the USB hub and FortiusAnt picks up the Anself/Dynastream (as reported in the GUI), all the "broadcast" messages appear correct and seemingly to work ok.  This will leave the other, Suunto Movestick Mini dongle for Zwift, but **in the pairing screen now there are NO devices found at all** :(

WouterJD, 2020-12-10:
I could implement -D antDongle, where you can specifiy which dongle you want. (in this case -D 4104) which would select one dongle if there are two different dongles.

Please experiment first and let me know whether such a selection would help.

switchabl, 2020-12-10:
That's a curious case. The Suunto is supposed to be completely fine with Zwift. There have been reports that newer ANSELF dongles are now identical to the (problematic) CYCPLUS ones, but I think those report as CYCPLUS when connected. So yours should be fine.

Maybe there is something in the Zwift logs that can shed some light:
https://support.zwift.com/locating-log-files-BykKO63fr

BrRoBo, 2020-12-10:
@switchabl Yes, this is a curiosity.  Not sure which dongle is the actual problematic one.  I'll grab a new Zwift log with the non-working setup (scenario 2) and see if the Ant scanning has some issue reported.

BrRoBo, 2020-12-10:
> 
> 
> I could implement -D antDongle, where you can specifiy which dongle you want. (in this case -D 4104) which would select one dongle if there are two different dongles.
> 
> Please experiment first and let me know whether such a selection would help.

Yes that might be a good potential workaround, especially if someone has a system where they don't seem to be able to get the dongle order changed.  I think I just got lucky the enumeration in the external USB hub was a simple fix for me.  

I am happy the setup works with scenario 1but digging into the underlying issue to determine cause might be worthwhile.  .

switchabl, 2020-12-10:
> 
> Yes that might be a good potential workaround, especially if someone has a system where they don't seem to be able to get the dongle enumeration order changed.
> 

If all else fails, there is always the possibility to 1) plug in just one dongle 2) start FortiusANT 3) plug in second second dongle 4) start Zwift (or whatever you use). But I guess that might be annoying if you had to do it every time.

WouterJD, 2020-12-10:
Agree, that's more like test mode

WouterJD, 2020-12-10:
PS -D because a and A already in use

switchabl, 2020-12-10:
Yikes! We should really have a configuration file. I had started some work on that for the remote control feature (for mapping the buttons). Then we could have some (optional) parameters for advanced users/testing/debugging without creating 1000 new command-line switches. And maybe some day a "Settings" dialog for the most important things so normal users don't need to worry about command-line. Well, I will get back to that at some point.

WouterJD, 2020-12-10:
-D implemented in version 3.9 which is to be released this week

PS. configuration file for somewhere in future. We still have a lot of alphabet left to enjoy :-)

closed 2020-12-10 by WouterJD