WouterJD/FortiusANT #164: Enhancement - ignore head unit buttons when GUI is minimized or out of focus

BrRoBo, 2020-12-11:
I've accidentally killed my ride in Zwift a couple times by hitting the Cancel button on the head unit.   

Got me thinking, why not ignore the buttons when the GUI is minimized or when out of focus?

WouterJD, 2020-12-14:
If you run FortiusANT, hidden behind Zwift and want to control with the head unit on top of your bike that would not be possible.
I would say: it happens once but not again...

WouterJD, 2020-12-14:
When zwifting, head unit can be used for virtual gear (up/down buttons).
So that's why head unit remains active.

BrRoBo, 2020-12-14:
Yes, but when the GUI is in the background out of focus, why would you want it to respond to the "Cancel" button and exit the "Start" (running) mode?  It is the GUI states that I am looking to stop changing.  If in Zwift, only Zwift should respond to the head unit, not the GUI.

switchabl, 2020-12-14:
I have a slightly provocative question here: do we need a button for "Stop" at all? I've only ever pressed that by accident, resulting in the same frustrating experience @BrRoBo had.

BrRoBo, 2020-12-14:
> 
> 
> I have a slightly provocative question here: do we need a button for "Stop" at all? I've only ever pressed that by accident, resulting in the same frustrating experience @BrRoBo had.

I do find the Stop button is useful for allow us to go back and forth between the calibrated/run mode and back to the Runoff mode.  It can take a good 15 minutes of warmup riding before our trainer results stabilize (due to cold garage) and report good watts per effort. 

An application should not respond to GUI events unless it is in focus.  That is a industry standard approach to any GUI application.

WouterJD, 2020-12-15:
requirements
- Be able to control FortiusANT from head-unit
- Ignore button when pressed inadvertently; but do attend when pressed intentionally
- Do cancel when we are in calibrate mode
- Igore it when GUI is hidden
- Do not ignore when no GUI used (console mode)

Perhaps just provoking as well.

It works as designed and I leave it the way it is.

closed 2020-12-15 by WouterJD

WouterJD, 2020-12-16:
Implemented: main window cannot be closed from the head-unit; if you do this by accident you have to get off your bike to restart the program