WouterJD/FortiusANT #165: Extend new power curve to T1902/T1901 combination

switchabl, 2020-12-11:
Not going to start on this right now, just to keep it on the radar and collect some thoughts. Probably better to wait for some more feedback now that it is released anyway. But if there are no unexpected issues, we will probably want to do this at some point.

Plan:
- Try to load the T1902 (solid green) firmware on my old T1942 (solid blue); should be possible according to @totalreverse. Failing that get my hands on an actual T1902.
- Measure the brake control output from the T1902 for different resistance settings
- Compare with the T1932 (that data I already have) to find the relationship between T1932 and T1902 resistance values
- Check that the results are consistent with the data collected for the original T1902/T1901 curve
- Port the code to the legacy USB class, with the addition of a mapping between resistance values.
- Check against some test ride with a power meter to make sure it actually works

So except for the last step this should be possible without anyone even having to get on a bike. The tricky part for the power curve is the brake and that part we have already figured out.

totalreverse, 2020-12-11:
The T1902 and T1942 devices have a different socket pin-out. You can load the firmware, but without remapping the control of the pins, the firmware cannot control the eddy current brake.

There is an (highly) experimental firmware to control the eddy current brake in my github repo (https://github.com/totalreverse/ezTraCon). Connected to a eddy current brake, the firmware emulates the T1942 protocol. 
With some effort and a little C knowledge one could implement the ancient T1902 protocol.

switchabl, 2020-12-11:
@totalreverse Thanks. I don't have a T1901 brake right now anyway, so I was just going to open it and attach the oscilloscope directly to the C.7 pin (and probably connect a function generator set to to simulate the brake powerline signal and maybe the wheel sensor). I gathered the pin is probably just unconnected so this should work? I only need the mapping resistance value -> duty cycle.

totalreverse, 2020-12-12:
@switchabl You want to analyze, how the original Tacx software tries to control the resistance (by changing the duty cycles) if you change 
- target power (in the software) or 
- the (simulated) wheel speed?

My personal opinion is, that the best solution is a power model based on a "fit" with real world data gathered with a power meter. 

To analyze the original Tacx software, I personally would adapt the firmware to send synthetic data.  The "solid" head units can send any frame you want (and even can emulate every USB device you want). 

BTW: You also did a great work with your deep dive into the Tacx legacy ANT protocol. It's almost complete.

WouterJD, 2020-12-14:
Using the following inputs:
- Measurements as done by @yegorvin, see manual
- function Resistance2MB() as implemented
- FortiusANT - Powercurve.xlsm as available

I created below function (in Excel), which has the same structure, just ScaleFactor is different (0.355 instead of 0.0149)
```
Public Function Resistance2PowerMBLegacy(Resistance, SpeedKmh) As Double
    Dim ScaleFactor         As Double
    Dim CriticalSpeed       As Double
    Dim RollingResistance   As Double
    
    ScaleFactor = 0.0149            ' N
    CriticalSpeed = 4.85            ' m/s
    RollingResistance = 15          ' N
    
    Dim Speed                 As Double
    
    ScaleFactor = ScaleFactor * 1.55                ' Experimental
    ScaleFactor = ScaleFactor * 3070 / 200          ' R=3070 gives 370Watt @30km/hr <==> R=200 for @yegorvin
    
    Speed = SpeedKmh / 3.6
    Resistance2PowerMBLegacy = Speed * (ScaleFactor * Resistance * Speed / (Speed + CriticalSpeed) + RollingResistance) * 0.68
End Function

```
This provides the following graph, comparing @yegorvin's measurements, Current implementation (1902/Legacy) and above formula:
[image]

So @switchabl you formula can be used, with ScaleFactor = 0.355 providing same accuracy as the current formula (based upon one set of data only), providing consistency over both implementations.

switchabl, 2021-01-17:
@totalreverse Have you ever opened a T1902 and taken a photo of the PCB? I have succesfully loaded the T1902 firmware on the T1942 and switched the PWM pin (it was really just 3 bytes that needed to be patched). There is a relatively large resistor on the power line input, so it will probably be fine (together with the internal diodes of the EZ-USB), but I do wonder if the T1902 has an additional protection diode. There are some unpopulated pads in my T1942.

I've also thought about writing a short code for reprogramming the VID in the EEPROM.

totalreverse, 2021-01-17:
I opened the T1902 head unit three years ago and I probably took pictures, but I can't find them right now. 

Reprogramming the EEPROM makes it difficult to identify the head unit before loading a firmware. 
The ezUSB can renumber the VID/PID in software. Maybe you can pick up the code from the T1942 firmware to renumber the device or you have a look the the ezTraCon firmware for a C implementation. 

In the t1942 firmware, the USB jump table code starts at address 0x371 and you find the USB "Get Descriptor" code from address 0x394 to 0x4b1. The raw USB Descriptor data (T1942) is from 0x10ba to 0x111c.

Or you pick up the ezTraCon firmware and implement the old legacy T1902 protocol - shouldn't be too difficult.

switchabl, 2021-01-19:
@totalreverse Thanks!

> The ezUSB can renumber the VID/PID in software. Maybe you can pick up the code from the T1942 firmware to renumber the device or you have a look the the ezTraCon firmware for a C implementation.
>
> In [...]

The problem is the T1902 firmware does not re-enumerate, it keeps the built-in endpoints/descriptors (with the PID/VID from the EEPROM). So it would be a rather big change (or a rewrite based on your firmware, but for my purposes I would like 100% identical behaviour, so this is not ideal). But it is mostly a matter of convenience anyway. If I load the patched firmware, I can turn a T1942 into a perfectly fine T1902 with PID e6be. This is ok for testing because I can just change the PID in FortiusANT.

Which is really the whole purpose: it is currently quite scary to make bigger changes to the FortiusANT core because no one has all the supported device combinations available for testing. Now I have at least:
- T1941 Fortius motor brake (230V)
- T1901 Flow magnetic brake (230V)
- T2021 Genius brake (230V)
- T1932 new head-unit
- T1942 old Fortius head-unit
- improvised T1902 head-unit
- T2022 Genius head unit
- (currently trying to buy a T1981 Bushido brake + T1982 head unit)

I have also successfully used the improvised T1902 to correlate the resistance values with the T1932. So I should indeed be able to transfer the power curve from the T1932.

> I am curious what's the third byte you have to patch to switched the PWM pin ...

If I remember correctly, the 3 relevant places were:
- write to `OEC` to enable pin (in the initialization)
- apply OR bitmask to `OUTC` for turning pin on
- apply AND bitmask to `OUTC` for turning pin off (both in some timer interrupt)

This is the diff (relative to old firmware version at https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/tacximagic_1902_firmware.hex)
```
75c75
< :100EE5004480F0901042E0F5CCA3E0F5CDC2CFD21E
---
> :100EE5004402F0901042E0F5CCA3E0F5CDC2CFD29C
79c79
< :0C0F560098E0547FF0D082D083D0E032CD
---
> :0C0F560098E054FDF0D082D083D0E0324F
187c187
< :100B8200907F9E7480F0907F967430F0E4907F970F
---
> :100B8200907F9E7402F0907F967430F0E4907F978D
```

> Found a picture - not sure if this was a T1902 head unit ..

It looks like it is missing the connection between pin 14 and 16 on the MCU (PC0 and PC2), so I think it is.

Here is my T1942 for comparison. It looks very similar (if you ignore the silkscreen), but it is missing 3 components. One of them is likely a Zener diode (the one with the blue band). So it seems the T1902 MAY indeed have additional protection circuitry. I need to check what exactly the unpopulated pads are connected to.
[image]

WouterJD, 2021-01-20:
I'm proud on you all!

WouterJD, 2021-11-22:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2021-11-22 by WouterJD

switchabl, 2021-11-22:
This will still happen. I just don't have the time right now, so I am focusing on smaller things, where I don't have to switch trainers a lot etc. But I'm fine with closing this in the meantime. It will probably just return as a PR at some point.

WouterJD, 2021-11-22:
Yeah good. I check all open issues now/then to see where assistence is required. Therefore try to keep list short, but do not intend to close too soon😉