WouterJD/FortiusANT #167: FortiusANT license?

switchabl, 2020-12-14:
I just realized that there is no mention anywhere how FortiusANT is licensed. Now, from what I gather you consider this open source and are fine with other people making their own versions. And I don't anticipate anyone sueing anyone. But it does leave everyone sharing it and modifying it on somehwat shaky ground and it should be easy to fix.

1) Decide on a license. Probably the most popular options:
- MIT license: do what you want
https://opensource.org/licenses/MIT
- BSD license: do what you want, but you must credit the original author(s) in the software and the documentation
https://opensource.org/licenses/BSD-2-Clause
- GPL license: you can share and modify it, but if you do you have to share the source code as well
https://opensource.org/licenses/GPL-3.0
2) Ask everyone who contributed (non-trivial) code if they are ok with it
3) Add LICENSE file to Github with the license text

Because of step 2) it should maybe be done sooner rather than later, because it becomes more work as more people contribute.

WouterJD, 2020-12-15:
I agree and was thinking about this.

This project is my first project on github, open source and python and therefore I'm open for suggestions.
Like many projects (I suppose) I created it to make my own trainer work and published "let it be helpful to someone else". It has grown beyond that; helping people with all questions takes time. Some formal steps should be set to manage it.

What I want is to keep one FortiusANT and avoid that many versions are going to spread.
_This is a reason I also accept changes when the usefulness for the greater audience is not always clear._

As you have seen I like to know who I communicate with but (unless I missed your introduction) I feel like talking to @switchabl as github ID. I appreciate your energy to improve FortiusANT; thanks for that. I think focus must be on supporting old Tacx trainers and making that as usable and accessible for an as-large-as-possible user-community. And funny options allowed, as second prio.

Suggestions to describe this focus and related license issues welcome.
Are there other github specific options to be used and/or activated?

marcoveeneman, 2020-12-15:
Yes, picking a license would be a good idea, i don't mind which one to be honest.

Regarding other GitHub specific options, yes, please start using software releases instead of committing the binary files (.exe,.pdf etc) to the repository, this is considered bad practice.

BikeBeppe64, 2020-12-15:
ready to participate with a donation for future registrations to be paid

switchabl, 2020-12-16:
> What I want is to keep one FortiusANT and avoid that many versions are going to spread.
> This is a reason I also accept changes when the usefulness for the greater audience is not always clear.
>I think focus must be on supporting old Tacx trainers and making that as usable and accessible for an as-large-as-possible user-community. And funny options allowed, as second prio.
> 
> Suggestions to describe this focus and related license issues welcome.

I feel this is more about how you manage the project than about a license, and for what it's worth I think you have been doing a great job in that respect. If people see that there is an active maintainer and that there is a good chance to have their work merged into the upstream repo, then they will usually try do that. It is just a lot less work in the long run than having your own fork. Of course, if it is really niche/far away from core functionality than it may make sense to have a fork instead.

Maybe this points slightly towards GPL because that ensures if someone makes changes, they have to share the code and you can always integrate that even if they are not actively pushing for that. And it more or less precludes them from making e.g. just a new GUI and then not sharing but selling it instead (if that is something that bothers you).

Something that is common is to have a file "CONTRIBUTING" in the root directory of the repo. There you could state that focus, explain that contributions are welcome and say how you expect people to proceed if they want to contribute.

Probably the most important reason to have a proper license is this: if one day you cannot/do not want to maintain this anymore for whatever reason, it is clear that someone else can just adopt the project under the existing license and people can continue using their trainers. Some projects have died because the original maintainer stopped and there was no proper open-source license so it was not clear it was ok to continue. Most approved open-source licenses are fine for that, it is just important to have one.

> Are there other github specific options to be used and/or activated?

Releases mostly, like @marcoveeneman said:
https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/managing-releases-in-a-repository

Having big binary files in the Git repo is not ideal, particularly if they are updated regularly, because it will make it grow bigger and bigger over time (when you `git clone` you download all released versions, not just the current one). This applies mostly to the .exe files and the manual.

Usually they are just attached to the release but not tracked in the repo, like so:
https://github.com/switchabl/FortiusANT/releases/tag/v4.0
(I originally only had the driver installers there, but extended it a bit to show how a full release might look like)

Apart from that, having people work on their own forked copy and then having one project maintainer integrate that like you do now is fairly common and works well for smaller projects. You may save time by relying more on pull requests more instead of integrating by hand.

> As you have seen I like to know who I communicate with but (unless I missed your introduction) I feel like talking to @switchabl as github ID. I appreciate your energy to improve FortiusANT; thanks for that.

Apologies, I am not really comfortable with sharing personal information online (not using social media either). I did originally post this:
>Greetings from Bavaria, Germany. I've been using my own hack-ish solution to connect my Fortius to Zwift for a couple of years (essentially hooked up part of the ANT+ simulator to the Tacx dll, of course this could never be published because of the license issues) but this is now a better and cleaner solution. I would like to thank everyone who has worked on it!

If you would like to have a more private chat, feel free to drop me an e-mail (switchabl _at_ mailbox _dot_ org). I hope you can understand.

PR #169 (2020-12-21): Add license
This PR applies the GPLv3 license to FortiusANT, as discussed with @WouterJD.

Following GitHub-users have contributed to the existing code; please approve this PR ASAP to agree with the license change:
- @mattipee 
- @fritz-hh
- @cyclingflow
- @switchabl

Fixes #167
closed 2020-12-21 by WouterJD