WouterJD/FortiusANT #168: pre-packaged driver installers

WouterJD, 2020-12-16:
I've made some pre-packaged driver installers if anyone wants to try. Let me know if it works.
- T1902 head-unit (solid green)
https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T1902.exe
- T1942 head-unit (solid blue)
https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T1942.exe
- T1904 head-unit (white/green)
https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T1904.exe
- T1932 head-unit (white/blue)
https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T1932.exe

Works fine for me with T1932 and T1942, could not test the other two. I have no way to sign the installers, so there may be some security warnings (the driver itself is signed though). There should be no need to change anything in the device manager, neither after loading the firmware in FortiusANT, nor when you plug it into a different USB port.

_Originally posted by @switchabl in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-745655618_

WouterJD, 2020-12-16:
@switchable; could you please suggest how to maintain this? What documentation and/or instruction is required?
@mickyduck55; you could refer to this issue and/or wait untill a final solution is published.

Thanks all for improving FortiusANT

Mickyduck55, 2020-12-16:
I have added a link to this thread so it can be read in context....

switchabl, 2020-12-16:
@WouterJD There probably is no maintenance, it is unlikely that there will be a new version. If anyone wants to re-produce: those were created using the InfWizard from libusbK (that you suggest in the manual), so nothing fancy.
1) Choose libusb0 package
2) Skip device selection
3) Enter data manually 
- VID: 0x3561
- PID: 0x1902, 0x1904, 0x1932, 0x1942, 0xe6be
- Manufacturer: Tacx
- Description: "Tacx T1902" etc, "Tacx T1942 (no firmware)" for 0xe6be
4) Select "Client installer"
5) Name: "T1902" etc, "T1942_no_firmware" for 0xe6be

T1902, T1904, T1932 you are done, just take the "InstallDriver.exe" from the output dir (contains everything).
For T1942:
1) take the T1942_no_firmware.ini from the T1942_no_firmware dir, put it into the T1942 dir
2) download old 7z 9.20 (https://sourceforge.net/projects/sevenzip/files/7-Zip/9.20/7za920.zip/download)
3) put 7za.exe into T1942 dir, rename to 7z.exe
4) run "re-pack-files.cmd" in T1942 dir
5) InstallDriver.exe now contains both T1942 "personalities"

For the user, it should only be "run appropriate installer for your head-unit", confirm install ("yes"/"next"). It will probably trigger security warnings for some users (unless we can get someone to sign them with an EV certificate). We will need to wait for feedback on that.

switchabl, 2020-12-16:
Can anyone think of a reason not to have an all-in-one package and just install the drivers for all supported interfaces everytime? I suppose in the rare case that you want to use libusb for a T1902 and the Tacx drivers for a T1932 or something you are probably an advanced user and are fine with using Zadig (or switching drivers in the device manager).

Mickyduck55, 2020-12-17:
This has been asked on another forum after posting a link to this thread,
could anyone supply the answer to the final  I have copied the entire post
for clarity

That is an interesting link. There are several installer packages shown,
including for Imagic 1902 and 1904 handlebar controllers. The T1902 is all
yellow-green and has always been described by Tacx as only compatible with
a 32-bit OS. The T1904 is white with a yellow ring and was both 32-bit and
64-bit compatible. On the link above, the discussion involved someone using
an Imagic trainer with a T1932 controller (white with a blue ring) which is
also both 32-bit and 64-bit compatible.

"So, I remain curious whether there is anyone who has actually used
FortiusANT with a T1902 on a 64-bit OS. T1902 controllers are getting to be
a fairly scarce item"

WouterJD, 2020-12-17:
@Mickyduck55 Until now I have not heard anybody complain about 32/64 bit OS issues.

Mickyduck55, 2020-12-17:
Thank you.. I will get back to you re my system etc when my dongles arrive
and I get this working ..
When all is well if you don't mind I will post a link to your page on my
site Mick Drake <http://mickdrake.com>
I will add a new page specifically for FortiusAnt

WouterJD, 2020-12-17:
@Mickyduck55 nice site and sure libk to FortiusANT👍

Mickyduck55, 2020-12-17:
It is a pity not more people know about your recent devellopments

I have known about your FortiusANT for some time but always considered it
complicated but I saw a link to your new manual and was very impressed.

The auto drive install is a good step as well ... getting the driver
installed and working on a windows 10 PC can be tricky with all the driver
signing security issues.  I still use TTS and Golden Cheetah, the latter
being unreliable with the current USB interface, looking forward to trying
your solution.

I see a lot of messages re TTS and Fortius trainers being obsolete (on the
Lunicus site and GC Google group page) .. the good news is that if you want
one here in the UK they can be bought for about £50 as most people believe
them useless without TTS.

If you have had a quick look at my site you do know all about me now :0
just need to get the ANT dongles and I hope to be sorted.  Thanks again

WouterJD, 2020-12-17:
> I see a lot of messages re TTS and Fortius trainers being obsolete

Feel free to link those sites to FortiusANT.
As you know now, the trainers are not obsolete at all; the Tacx software is :-)

Mickyduck55, 2020-12-17:
I will .. as soon as my dongles arrive and I have it working I will be
shouting very loud...  As I said my TTS is still working and I have all the
files that I need backed up... I do like Golden Cheetah though

switchabl, 2020-12-17:
> 
> "So, I remain curious whether there is anyone who has actually used FortiusANT with a T1902 on a 64-bit OS. T1902 controllers are getting to be a fairly scarce item"

This is again a case of the Tacx software/drivers being obsolete, rather than the device itself. They just never bothered making a 64-bit compatible drivers. Since we are using the libusb driver, this is not an issue. I don't have a T1902, but I have just used a T1942 running the T1902 firmware (which is basically the same with a blue case) for development purposes on a 64-bit Windows 10 machine with no issues at all.

I think it would be fairly easy to make FortiusANT a lot more widely known if you were to contact sites like dcrainmaker, gplama and zwiftinsider and have them write an article/make a video about it. But personally I would wait until the user-experience is more suitable for non-technical users (one complete installer including FortiusANT and drivers, no need to use command-line options, simple setup/pairing screen and so on), if that ever happens. Otherwise you will have a support nightmare on your hands. And probably double-check there aren't any legal/copyright/trademark problems that Garmin/Tacx could use to shut the project down if it gets on their radar.

Mickyduck55, 2020-12-17:
OMG really hope the last sentence is not valid until I get this all
working... I'm not sure they would really bother (Although obviously don't
know) I have seen TTS serial number sharing, License files being listed on
public boards nothing ever seems to get taken down.
I think your point re user-experience is very valid. I have been put off
attempting this until I read the recent manual and it still remains to be
seen if this is as trouble free as I am hoping.. I am awaiting my second
ANT dongle before I can try installing for use.  I have installed just one
dongle and it found my fortius without any messing about so i need the
second one to connect to Golden Cheetah.

switchabl, 2020-12-17:
@Mickyduck55 Don't worry, I am not expecting anything to happen in that regard. And they normally shouldn't have any grounds for it (there is nothing wrong with writing your own compatible software, Golden Cheetah does the same). My point was just that if you were going for much wider publicity (which is probably premature anyway), you have to double-check you are not making any unforced errors there.

WouterJD, 2020-12-17:
> * T1932 head-unit (white/blue)
>   https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T1932.exe

@switchabl I reinstalled my laptop for FortiusANT today.
Works like a charm!

switchabl, 2020-12-17:
@WouterJD That is good to hear. I made an all-in-one package now, suitable for all head-units:
https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T19xx.exe

This is probably the one people should use now (unless there is a somehow a specific reason not to). Procedure to build is similar to T1942 above, but instead of adding a second .inf file, you should replace it completely with this hand-made one https://gist.github.com/switchabl/a30cb9932f00fba689f0a5140e3c591b and then run re-pack.

ericchristoffersen, 2020-12-18:
Hi Guys!

The world contains a disaster of old obsolete tacx trainers. I don't own a tacx trainer so can't debug or investigate actual issues, but I wouldn't mind if these things became un-obsolete. I've been trying to get Mick up and running in golden cheetah slope mode and from what I can tell the stuff in golden cheetah isn't working, it appears from Mick's data that the tacx interface's built in 'slope mode' doesn't work like golden cheetah (or fec) expects.

Micky pointed me at your manual and I was pleasantly surprised. You guys noticed some serious limitations of the fortius and have a bunch of cool mitigations.

It seems like you are always running the devices in ergo mode, driving the brake yourselves, so I tried that in golden cheetah too.

https://github.com/GoldenCheetah/GoldenCheetah/pull/3725

The change sends a 'resistanceWatts' hint whenever a controller is given a gradient. Since its fully parameterized by the physics it should be accurate estimate of watts needed to maintain current (virtual riding) state.

Because I don't have a tacx trainer I can't really test any of this, and there are too many devices to count - I'd like to know how to best adapt your technology into golden cheetah. Of course it would be possible to ask people to get ant dongles and go the fec route, but seems absurd when gc could just set the load directly.

You guys appear to be the current public authority on how to drive all these janky old trainers. What do you think is the best way for gc to utilize your stuff? Is it possible you will ever create a c++ interface to the fortius devices that avoids the ANT stuff?

Thanks!

WouterJD, 2020-12-18:
@ericchristoffersen thanks for the inspiring post; to be further discussed in separate issue👍🚴‍♀️🚴

Mickyduck55, 2020-12-18:
Hi Wouter
I started this journey with Eric only a few days ago and then came across
your solution... I hope you are able to work together on this it would be
so cool

WouterJD, 2020-12-24:
> This is probably the one people should use now (unless there is a somehow a specific reason not to).

Is this the final? I will then work on it to integrate

switchabl, 2020-12-24:
@WouterJD I haven't really had any feedback from people one way or the other (but I have tried it successfully myself a few times). But unless we get reports of people having problems with it, yes, I think this should be final.

Mickyduck55, 2020-12-24:
I used it without issue and there is a post on the GC group from a user who
installed his fortius 1932 successfully after struggling with previous
installs

"With the driver installers Mick linked
<https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-745655618>,
GC now detects my Fortius T1932. Great! Thanks!"

Mickyduck55, 2020-12-24:
@WouterJD/FortiusANT <FortiusANT@noreply.github.com>
I used it without issue and there is a post on the GC group from a user who
installed his fortius 1932 successfully after struggling with previous
installs

"With the driver installers Mick linked
<https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-745655618>,
GC now detects my Fortius T1932. Great! Thanks!"

switchabl, 2021-01-19:
@WouterJD The all-in-one driver package now has a new home at https://github.com/switchabl/t19xx_usb, along with some helpful info (hopefully), in case you want to link to it/mention it in the manual.

WouterJD, 2021-01-20:
Yes - definitely next step
Good idea to define as separate project👍

Mickyduck55, 2021-01-21:
@switchabl @WouterJD 
I have already posted a new link .. thank you

WouterJD, 2021-02-01:
It's added to manual in version 5.2 as will be released shortly.

mattipee, 2021-02-01:
> @WouterJD The all-in-one driver package now has a new home at https://github.com/switchabl/t19xx_usb, along with some helpful info (hopefully), in case you want to link to it/mention it in the manual.
> 
> @Mickyduck55 @mattipee @ericchristoffersen This may be interesting for the GoldenCheetah people as well.

Thankyou, @switchabl ! That made it easy. :+1:

Mickyduck55, 2021-02-01:
@switchabl I have already linked to this everywhere I post ... thank you so
much it was a real pain to install before this

closed 2021-02-04 by WouterJD