WouterJD/FortiusANT #173: Version 4.0 Communicates Much Higher Power vs. 3.8

clearwave1, 2020-12-20:
Details:
- I have both Windows executables on the same machine
- Fortius with T1942 Head Unit
- Version 3.8
     - Start with no parameters
     - It asks me to start pedaling
     - It goes through the calibration sequence
     - I then ride in a 52 x 21 @ 100rpm which is about 35 km/hr and it shows about 105 watts
- Version 4.0
     - Start with no parameters
     - It has automatically started and does not ask me to start pedaling or calibrate
     - I then ride in a 52 x 21 @ 100rpm which is about 35 km/hr and it shows about 305 watts

I'm wondering whether the lack of the calibration when it starts is the issue.

I do not have a power meter so I have nothing to measure against. However, I also have a non smart Fluid 2 trainer which Zwift has the torque curve for and the FortiusAnt v3.8 felt closer to this than 4.0 does.

Let me know if there are things I can try to debug this.

Thanks for all your hard work.

WouterJD, 2020-12-20:
Do you use calibration, if not why?
Run with -d127 which produces a logfile; attach to response and I will check.

clearwave1, 2020-12-20:
I mentioned in my initial report that 4.0 does not let me calibrate. 

I have attached two sets of log files - one from 3.8 and one from 4.0 (named in the file name). Github doesn't seem to support .json files so I added a ".log" on the end of these files.

[FortiusANT-3.8.2020-12-20 22-51-55.log](https://github.com/WouterJD/FortiusANT/files/5721643/FortiusANT-3.8.2020-12-20.22-51-55.log)
[FortiusANT-4.0.2020-12-20 17-45-29.log](https://github.com/WouterJD/FortiusANT/files/5721644/FortiusANT-4.0.2020-12-20.17-45-29.log)
[FortiusANT-4.0.2020-12-20 17-45-29.json.log](https://github.com/WouterJD/FortiusANT/files/5721646/FortiusANT-4.0.2020-12-20.17-45-29.json.log)
[FortiusANT-3.8.2020-12-20 22-51-55.json.log](https://github.com/WouterJD/FortiusANT/files/5721647/FortiusANT-3.8.2020-12-20.22-51-55.json.log)

WouterJD, 2020-12-21:
`17:45:31,134: Motor Brake Unit Firmware=4103 Serial=  127 year=2005 type=T1946 Version2=3086 MotorBrake=False`

Version 4.0 detects that you use a magnetic brake, version 3.8 did not and treated the magnetic as a motor-brake.
The power curve for magnetic brake was incorrect; at least for most users. Version 4.0 should be better.
Please read the manual to understand this and how to act and/or refer to issues #143 and #153.

switchabl, 2020-12-21:
@WouterJD But T1942 is old solid blue unit and does not support magnetic brake I think? So this sounds wrong.

WouterJD, 2020-12-21:
@clearwave1 did not tell what configuration he is using, so I cannot judge.
Why would 1942 not support magnetic brake?
In anyway, the controller does not report the Motor brake. See FortiusANT-4.0.2020-12-20 17-45-29.log

clearwave1, 2020-12-21:
I definitely have a motor brake. It is model T1946.50.

Overall, I have one of the original complete Fortius packages with the steering unit and motor brake. the motor brake says it is a 300W model - whatever that means.

switchabl, 2020-12-21:
> 
> Why would 1942 not support magnetic brake?

According to @totalreverse (and my own experience), T1942 firmware mostly passes commands back and forth to the brake using a serial protocol, but the magnetic brake has no serial protocol, it is completely "analog". The T1932 is smarter and can control the magnetic brake itself, while emulating the motor brake commands so the protocol is similar.

totalreverse, 2020-12-21:
Looks like you have to add the T1946 US model (60 Hz / 110V) to your list of motor brakes?

WouterJD, 2020-12-21:
I have implemented NewUsb as if all equal; now it appears that only T1932 supports magnetic brake. Sorry for that, I will make a patch.

In the meantime you can use 3.8

switchabl, 2020-12-21:
@WouterJD Well, we could just assume if head unit = T1942, it is always motor brake. But that's not really necessary, I think it is just this line that is the problem:

```
if self.MotorBrakeUnitType in (41, 42):
    self.MotorBrake = True
```

As @totalreverse said, the possible motor brakes are T1941 (230V) and T1946 (110V), so I think it should be `(41, 46)`.

WouterJD, 2020-12-21:
@switchable, I have checked the code and what was returned by the T1942 headunit (logfile). You're right.
I will adjust `if self.MotorBrakeUnitType in (41, 42):` into `(41, 46)` in a next version.

I don't remember where the 42 comes from (typo, mix-up with headunit?).

If it **is certain** that only T1932 can control a magnetic brake, I could add `if self.headunit != T1932: self.MotorBrake = True`, so if we miss the MotorBrake message for whatever reason, we don't switch to Magnetic brake.
_Please suggest._

@clearwave1 I suppose the 300Watt is the powerback, not indicating the resistance of the motorbrake (ref [TotalReverse](https://github.com/totalreverse/ttyT1941/wiki) search for T1946 and T1949.50)

switchabl, 2020-12-21:
@WouterJD I'm quite sure T1942 can't do magnetic brakes (not without firmware hacking at least). But I have thought about it a bit: normally the check should never be necessary (unless we missed something). And if it had been there, we would not have found this error; not until someone tried to use a _T1932_ with the T1946 brake AND realized it is wrong. So it may be better not to have it.

I can't check what is returned with a magnetic brake right now, but I think it was all 0s or something? If that is so, it could be safer to check for 0s and assume anything else is motor brake.

WouterJD, 2020-12-24:
Release 4.1 published; should solved the problem

TriScott, 2020-12-24:
@WouterJD The 4.1.d build seems to be bad:

Traceback (most recent call last):
  File "FortiusAnt.py", line 38, in <module>
  File "c:\users\wouter dubbeldam\appdata\local\programs\python\python38-32\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "FortiusAntBody.py", line 210, in <module>
  File "c:\users\wouter dubbeldam\appdata\local\programs\python\python38-32\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "bleDongle.py", line 15, in <module>
ModuleNotFoundError: No module named 'requests'
[12340] Failed to execute script FortiusAnt

TacxBiker, 2020-12-24:
@WouterJD 

4.1 doesn't calibrate like in 3.8
And the output power seems to high.
for the record, using a T1932 headunit and a T1941 brake

WouterJD, 2020-12-24:
Please attach logfile

TacxBiker, 2020-12-24:
[FortiusAntGUI.2020-12-24 15-23-51.log](https://github.com/WouterJD/FortiusANT/files/5739868/FortiusAntGUI.2020-12-24.15-23-51.log)

TacxBiker, 2020-12-24:
[FortiusAntGUI.2020-12-24 15-23-51.log](https://github.com/WouterJD/FortiusANT/files/5739870/FortiusAntGUI.2020-12-24.15-23-51.log)

TacxBiker, 2020-12-24:
[FortiusANT.2020-12-24 15-23-50.log](https://github.com/WouterJD/FortiusANT/files/5739872/FortiusANT.2020-12-24.15-23-50.log)

TacxBiker, 2020-12-24:
do you also need the .json log?

WouterJD, 2020-12-24:
No thanks

TacxBiker, 2020-12-24:
Ha, no problem. Have a good one!

WouterJD, 2020-12-24:
```
15:23:52,260: clsTacxNewUsbTrainer._ReceiveFromTrainer_MotorBrake()
15:23:52,261: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 1, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)

```

The issue is clear: the T1932 head unit returns a short buffer; 64 bytes are expected and only 24 returned.

The solution is not clear yet:
- either retry to see whether a longer buffer will be returned; this helped in other occasions
- define a command-line parameter what to do; e.g. -t T1941 or T1946 to define that we're dealing with a motorbrake or magnetic brake
- assume that it is a motor-brake

At this moment, a short message was not an expected situation and causes FortiusANT to react as if a motor-brake is attached.

I wanted to get the analysis out-of-my head, but will create a test-solution after the weekend.

WouterJD, 2020-12-24:
If you want to proceed during the weekend:

edit `usbTrainer.py line 237` replace `False` by `True`; then always MotorBrake is assumed.
We will work on a proper fix later...

TacxBiker, 2020-12-25:
Hi Wouter, that worked, thanks

if you need a test, let me know

WouterJD, 2020-12-27:
Short message of 24 bytes was received; retry implemented.
Branch: 4.2-Quality-upgrade; python only.
Please test (I will do integration test tomorrow)

TacxBiker, 2020-12-28:
I'll test this tonight

TacxBiker, 2020-12-28:
@WouterJD 
I uses the 4.2 branch. It did ask for the first time to calibrate. Only one time. it didn't matter whether I  stopped and started the program,
Only unplug the USB cable and replug gave a new calibration.
On the occasion the calibration didn't start the power was not right, to high. 
So stop the program, unplug the USB cable, plug it in again, start the program and calibrate. All things good then.

Another thing, in the 4.2 brach the red light of the head unit lighted up sometimes, rather mutch. It seem something to do with an error:
[image]

evertime the error message passed the light flashed.

Unfortunately I didn't use the debug mode..

Another question tough, in the hypothetically event I can't produce the asked power there isn't a way to reduce the power temporary is there?

All in all I had a good training for more then 1,5 hour, bluetooth worked perfectly. 
The Sufferfest app is using its onw calculated speed, It looks like the speed isn't send out on Bluetooth. There is no speed in the sufferfest app when selected the speed direct from the trainer.

WouterJD, 2020-12-28:
Well, at least we now touch the subject.
Apparently the head unit does not return enough data, which is caused by either loose contacts OR too fast polling of the headunit.

Since you did not provide the used command-line nor the trace-file, further analysis is not possible.
At least do NOT use -A for pedalstroke analysis, so that the trainer is polled once/250ms and provide logfile (start with -d127)

TacxBiker, 2020-12-29:
[FortiusAntGUI.2020-12-29 08-00-00.log](https://github.com/WouterJD/FortiusANT/files/5749855/FortiusAntGUI.2020-12-29.08-00-00.log)

[FortiusANT.2020-12-29 07-59-58.log](https://github.com/WouterJD/FortiusANT/files/5749858/FortiusANT.2020-12-29.07-59-58.log)

the log file of a start of the program, calibration and a minute training.

WouterJD, 2020-12-29:
```
08:00:01,574: clsTacxUsbTrainer.SendToTrainer(T=10, M=1, P=100, G=0, R=0, W=85, PE=0, S=0.0, C=0)
08:00:01,574: Trainer send data="02 00 00 00" (len=4)
08:00:01,574:                   tacx mode=10 target=0 pe=0 weight=0 cal=0

08:00:01,678: clsTacxNewUsbTrainer._ReceiveFromTrainer_MotorBrake()
08:00:01,678: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:01,678: Retry because short buffer received
08:00:01,782: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0, 3, 12, 0, 0, 4, 16, 0, 0, 53, 92, 124, 24, 28, 2, 0, 0, 17, 165, 0, 0, 0, 0, 0, 0])" (len=48)
08:00:01,782: Motor Brake Unit Firmware=4100 Serial= 3253 year=2008 type=T1941 Version2=540 MotorBrake=True

```

Here you see that the motorbrake message is sent/received;
- the first receive fails (len=24)
- the retry provides the correct data; type=T1941 and MotorBrake=True

This resolves the topic of this issue: since the correct brake is detected, the correct powercurve is selected.

WouterJD, 2020-12-29:
```
08:00:03,834: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:03,834: Retry because short buffer received
08:00:03,938: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0, 3, 19, 2, 0, 0, 0, 0, 0, 0, 0, 48, 95, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 101])" (len=48)

```
the short buffer error occurs more often, which is recovered in one retry here

WouterJD, 2020-12-29:
```
08:00:04,090: clsTacxNewUsbTrainer._ReceiveFromTrainer()
08:00:04,090: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:04,090: Retry because short buffer received
08:00:04,194: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:04,194: Retry because short buffer received
08:00:04,311: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:04,311: Retry because short buffer received
08:00:04,415: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:04,415: Retry because short buffer received
08:00:04,520: Trainer recv data="array('B', [24, 21, 24, 0, 9, 1, 0, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13, 10, 0, 0, 0, 0])" (len=24)
08:00:04,520: Tacx returns insufficient data, len=24
08:00:04,520: To resolve, check all cabling for loose contacts.

```

but sometimes even four retries fail and then cause the error "To resolve, check all cabling for loose contacts." is given.
Untill now, I have no other remedy for the short buffer message.

TacxBiker, 2020-12-29:
I don't think its a large problem, everything else seems to work fine

TacxBiker, 2020-12-29:
still one thing, calibration starts only once. Calibiration should start every time the programs starts. but perhaps thats a different thing

WouterJD, 2020-12-29:
Calibration is done every time when Start is pressed.

TriScott, 2020-12-29:
The red led on the 1932 is supposed to correspond with heartrate from a wireless monitor strap, but I notice mine flashes randomly in recent years even with TTS (i.e. not specific to FortiusANT).

Regarding the short 24 byte messages:  these also come from a 1932 head unit continuously when the powerback device is powered off, suggesting they are at least sometimes correlated with a missed message from the motor brake firmware.

TacxBiker, 2020-12-29:
Tried the calibration, that works everytime when pressed start.

WouterJD, 2020-12-29:
> the powerback device is powered off

Check all cabling, not only signal. @TriScott no you say it, I recall that a semi-loose 220V cable caused similar issues, perhaps causing some noise or so.

TriScott, 2021-01-09:
There is still something fishy with the motor brake detection with 1932 Fortius headunits.  I have stuck with FortiusANT 3.8 as it has been working well for me, but FortiusANT 5.0 still detects my Fortius as a magnetic brake and does not provide the "start pedaling" calibration prompt.

I do not think this is a "loose cabling" issue, but rather some sort of fickleness regarding the 1932 device.  My setup works very well and very reliably using FortiusANT 3.8 + Zwift for long rides (and with TTS), yet it 100% of the time returns 24 bytes during FortiusANT initialization ("Tacx returns insufficient data") and fails to be identified as a motor brake.

TriScott, 2021-01-09:
@jujroy Very likely you are experiencing this issue (as you described in your intro comment).  If you don't go through the calibration procedure every time you start, your motor brake is being treated as a magnetic resistance brake and the power numbers are way off.

Until the 1932 initialization is resolved once and for all, you are best off continuing to use FortiusANT 3.8.

WouterJD, 2021-01-09:
HI all, sorry for this inconvenience

3.8 works well for T1932 + motorbrake
It does not work well for magnetic brake

4.0 implemented the magnetic brake AND detected which way to go, asking the T1932 headunit for type-of-brake.
BUT the headunit does not always respond well; error-recovery was implemented. And still this is not enough.

This results in uncertainty what type of brake is connected when the motor-brake message is NOT returned; which results in the conclusion that it is a magnetic brake.

Therefore **-t Motorbrake** is implemented in version 5.1, which should solve the problem

(Alternative could have been to define Motorbrake as the default and implement -t Magneticbrake to overrule, but I did not do that)

So please test 5.1 from branch https://github.com/WouterJD/FortiusANT/tree/4.2-Quality-upgrade (which merges the working 4.2 with the 5.0 version.

WouterJD, 2021-01-09:
Please comment the latest version in pull request https://github.com/WouterJD/FortiusANT/pull/198

jujroy, 2021-01-09:
Hello Wouter,  thank you very much for your assistance and prompt reply on this issue.  I would love to add myself to the map.  I went to issue 14 and I am unable to figure out the way to do it :(  Keep going  around in circles.   Help please.  I will be out skiing today but will try to test 5.1 ASAP.  Many thanks

WouterJD, 2021-01-09:
Just added a comment un der #14. Sorry that the map is not clickable.

> I will be out skiing today
Lucky you; we're in lock-down in the Netherlands and skiiing in Austra/Italy impossible,

DoubleH63, 2021-01-10:
Hello Wouter

I am from Montreal, Canada.

First of all, thanks for giving us the possibility to use our old TACX Fortius on Zwift.

I found FortiusANT on January 1st. I had a few issued installing LibUSB since it is an unsigned driver. I had to follow the instructions for installing an unsigned driver on Win10.

I received my 2 dongles Wednesday and I was able to do a quick test on Zwift Wednesday night and it works fine.

I did a run Friday on the Richmond UCI: 17.1km in 24m40s, average speed=41,6km/h
I did a run this afternoon on the Watopia road: 30.2km in 45m45s,  average speed=39.6km/h

I was asked by a Zwifer if I was using a Diesel Engine or an Electric bike ;)

I see that the resistance is not exactly like it was with my TACX TTS 2.07. My average should be around 30km/h.

In the log file with the debug option, I have this:
Connected to Tacx Trainer T1932
Motor Brake Unit Firmware=4103 Serial=  879 year=2008 type=T1949 Version2=1300 MotorBrake=False

My problem is similar to the one reported by jujroy.

I will try version 5.1 to see if it solved the problem with -t Motorbrake

Many thanks

WouterJD, 2021-01-10:
@DoubleH63 that willl work; type=T1949 is not expected to be a motor brake and therefore MotorBrake=False.
I will study how to rework this brake-detection part

DoubleH63, 2021-01-10:
Hello Wouter,

I just tested version 5.1 on the same road than yesterday,

Result: 31.8km in 46m35s for an average speed of 41,0km/h

Doesn't seem to use the MotorBrake.

Here is the information I have when I started:

1. FortiusANT.exe -a -g -H0 -A -t Motorbrake
2. Connected to Tacx Trainer T1932
3. Motor Brake Unit Firmware=4103 Serial= 879 year=2008 type=T1949 Version2=1300 MotorBrake=False

The GUI shows: Fortius Antifier v5.1 test II

Is it help ?

WouterJD, 2021-01-10:
Yes it helps; thanks.
Working hard to get the version detection more solid

jujroy, 2021-01-11:
> 
> 
> Hello Wouter,
> 
> I just tested version 5.1 on the same road than yesterday,
> 
> Result: 31.8km in 46m35s for an average speed of 41,0km/h
> 
> Doesn't seem to use the MotorBrake as it should,
> 
> Here is the information I have in the DOS window when I started:
> 
>     1. FortiusANT.exe -a -g -H0 -A -t Motorbrake
> 
>     2. Connected to Tacx Trainer T1932
> 
>     3. Motor Brake Unit Firmware=4103 Serial= 879 year=2008 type=T1949 Version2=1300 MotorBrake=False
> 
> 
> The GUI shows: Fortius Antifier v5.1 test II
> 
> Is it help ?

Good evening.

I found out, by fluke, a sequence to make my system work consistently.
-I turn OFF the power switch on the Power Pack;
-I launch the FortiusAnt software and wait until the GUI gives me the Start Pedalling Prompt
-I then power up the Power Pack, start pedalling, go thru the calibration process and guess what...I get the proper power output.

I tried this sequence 5 times and always got the right power output.
Tomorrow I will Zwift with V5.1,

Hope this works for you. 

Cheers

DoubleH63, 2021-01-11:
> Good evening.
>
> I found out, by fluke, a sequence to make my system work consistently.
> -I turn OFF the power switch on the Power Pack;
> -I launch the FortiusAnt software and wait until the GUI gives [...]

@jujroy I will try your sequence also tomorrow.
Regards

WouterJD, 2021-01-11:
Version 5.1 Test III is available with major update on Motor/Magnetic brake determination.
(for details see [wiki](https://github.com/WouterJD/FortiusANT/wiki#version-51-test-iii-available-for-evaluation-jan-11-2021))

Please test

jujroy, 2021-01-11:
@WouterJD 
Hello Wouter,  I proceeded with 5 tests all with the Power Pack switch ON.
GUI displayed version 5.1 III with virtuel cassette / chainring 50/19 ratio.  I did not change the ratio at all.
All 5 times I got the Start Pedalling prompt and got the right amount of power similar to V3.8 without even going to the Parameters where is used to select **Motor Brake**.

All my compliments.

QUESTION:  in Zwift yesterday and this morning in the device coupling box I noticed 2 options for the Power source: TACX FE/C 57591 and also GARMIN PWR 57596.  I had the first one selected.  Also in the Cadence box I had 4 options : 57591, 57596 and Speed/Cadence 57595 (My selection) and Speed/Cadence 62509 which I suspect is my ANT+ Bontrager Duo trap.  Can you confirm that my selections were the right ones.  Many thanks.  Jean

jujroy, 2021-01-11:
@WouterJD ,

forgot to mentioned that I zwifted earlier this AM with V5.1II.  Results were amtching V3.8

TacxBiker, 2021-01-11:
with this new version motor brake detection seems to work every time.
after calibration the power feels the same as in 3.8 or previous versions with motor brake set True
Now I don't need to either enter trainer type or edit the usbtrainer.py

DoubleH63, 2021-01-12:
Hello @WouterJD,

I did test version 5.1 III and it did not work for me.

I did the calibration and it ran for almost one minute. Then I went to Zwift.

I was doing 110 RPM on 52/13 gears on a 4% hill with 164 Watts... and going 13km/h. The display on FortiusANT GUI was showing 65km/h.

The Power is not transfered to the motor brake.

Also, when I do the Runoff test the wheel stops from 40 to 0 in 4.4 seconds. It is supposed to take around 7 seconds. I did remove pressure on the wheel and it does not change the result.

Do you suggest a parameter adjustment?

Regards

WouterJD, 2021-01-12:
> with this new version motor brake detection seems to work every time.
> after calibration the power feels the same as in 3.8 or previous versions with motor brake set True
> Now I don't need to either enter trainer type or edit the usbtrainer.py

@TacxBiker Thanks

WouterJD, 2021-01-12:
> I did test version 5.1 III and it did not work for me.

@DoubleH63 that's a real pity.
Please rerun with -d127 (create logfile, which is also possible using the settings dialogue) and attach the logfile and json file to your next post. Sorry that it takes so much effort to get your config working reliably.

WouterJD, 2021-01-12:
> Can you confirm that my selections were the right ones. Many thanks. Jean

@jujroy FortiusAnt broadcasts the following devices
57591    FE-C Controllable Fitness equipment
57592    HRM Heartrate monitor (optional, according -H parameter)
57595    SCS Speed/cadense sensor
57596    PWR Power monitor

**In fact, only the FE-C (57591) is required**. SCS and PWR are implemented so you can attach a Garmin Edge on your bike, it should not be required by Zwift. If you use an ANT+ heart rate monitor, Zwift will connect to that and the FortiusAnt HRM is not required.

WouterJD, 2021-01-12:
> Results were matching V3.8

@jujroy thanks for confirmation

[9 more replies]

closed 2021-01-24 by WouterJD