WouterJD/FortiusANT #176: Installing Antbridge

AndrewB53, 2020-12-21:
Hi, I am currently setting up FortiusAnt on my kubuntu system and I am confused by a line in the user manual relating to installing the Antbridge packages. The line is "LOC=/lib/modules/uname -r/kernel/drivers/usb/serial/", however the location doesn't exist in my system. Do I need to create it?  I'm hoping that I can use my trainer (head type 1932) with Golden Cheetah, is Antbridge required to do this?
This looks a graet project.

WouterJD, 2020-12-21:
Hi @AndrewB53 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

switchabl, 2020-12-21:
I don't use (K)ubuntu personally (usually Fedora when I'm on Linux), but I think the instructions for Ubuntu are not helpful for FortiusANT. It may result in a working installation, but they contain a lot of things that are not required (like openant and antfs-cli) and some very bad advice (deleting the kernel modules) that may cause serious problems.

I am sure they were provided with the best intentions but I a would in fact suggest they are removed (and I think @WouterJD knows by now that I would not suggest that lightly).

I will set up a Ubuntu VM tomorrow and suggest a new installation procedure.

WouterJD, 2020-12-21:
Thanks; I will gladly adopt

AndrewB53, 2020-12-22:
Hello both, thank you for the support, I cannot offer much in the way of technical knowledge or coding but I'll certainly give any feedback I can. I have a dual boot machine with Debian and kubuntu (low latency), essentially the same, but I'll insatll on the Debian os when the kubuntu side is up and running.

switchabl, 2020-12-23:
Here are some preliminary instructions:
1) Install prerequisites
```
sudo apt install git python3-venv libsdl2-2.0-0
```
2) Download latest FortiusANT from GitHub:
```
git clone https://github.com/WouterJD/FortiusANT.git
cd FortiusANT
```
3) Set up Python environment
```
python3 -m venv venv
source venv/bin/activate
pip install pyusb numpy
pip install -f http://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04 wxpython
deactivate
```
4a) Support files (@WouterJD these should be included in FortiusANT in the future, so this step will become unnecessary)
```
mkdir supportfiles/udev
```
- create supportfiles/udev/65-antusb.rules
```
# USB ANT dongles
SUBSYSTEM=="usb", ATTR{idVendor}=="0fcf", ATTR{idProduct}=="100[8,9]", MODE="0664", GROUP="plugdev"
```
- create supportfiles/udev/66-tacx-trainer.rules
```
# Tacx T1902 "solid green"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1902", MODE="0664", GROUP="plugdev"'

# Tacx T1904 "white/green"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1904", MODE="0664", GROUP="plugdev"'

# Tacx T1932 "white/blue"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1932", MODE="0664", GROUP="plugdev"

# Tacx T1942 "solid blue" (no firmware)
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="e6be", MODE="0664", GROUP="plugdev"

# Tacx T1942 "solid blue"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1942", MODE="0664", GROUP="plugdev"
```
4) Allow access to Trainer/ANT USB devices without root privileges
```
sudo install -m 644 supportfiles/udev/*.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
```
5) run FortiusAnt (without root)
```
./venv/bin/python pythoncode/FortiusAnt.py
```

@AndrewB53 Let me know if that works. I had some problems with crashing/freezing. Maybe it is just my VM, but maybe there are some stability problems we need to work on.

I would like to get rid of step 3 as well, but there seems to be a compatibility problem with the `python3-wxgtk4.0` included in Ubuntu so it seems to be necessary for now.

AndrewB53, 2020-12-23:
@switchabl I will try it later today if time allows.

AndrewB53, 2020-12-23:
@switchabl I have problems with my kubuntu installation so I started to set up FortiusANT on my Debian partition. However at step 3 when I try to install wxpython I get a build error, see text file for details.
[wxpyhon_error_text.txt](https://github.com/WouterJD/FortiusANT/files/5734826/wxpyhon_error_text.txt)
I will repair my kubuntu installation and try again.

>

switchabl, 2020-12-23:
@AndrewB53 Yes, sorry, this will not work on Debian without some changes. At least the wxpython command needs to be changed to: (assuming Debian buster)
```
pip install -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/debian-10 wxpython
```
But there is a good chance that you would need to install some additional packages as well.

AndrewB53, 2020-12-23:
@switchabl The installation of wxpython went okay, but I get an error at step 5 , "no such directory".  I guess it means the  python folder in the Italian environment.

switchabl, 2020-12-23:
@AndrewB53 It probably means you were a) not in the FortiusANT directory when you created the venv in step 3 or b) are not in the FortiusANT directory now. If a) find the "venv" directory and move it to the FortiusANT directory. The language setting should make no difference.

AndrewB53, 2020-12-23:
@switchabl I repaired my kubuntu partition and installed the FortiusANT without problem. The GUI runs okay and it sees my Ant+ dongle, which arrived today, however although not branded Cycplus it revealed itself to be that type when identified by the app. The heart rate kept disappearing, more disappeared than not. I had one crash whilst running the app. The trainer was identified okay 1932. I'm not sure where the "Italian" came from, it's not what I intended to type! I think it was meant to be "virtual"!

TriScott, 2020-12-24:
Running FortiusANT on Linux is an exercise in misery.  Getting it to run requires passing through dependency hell, and, once it is running, it is far less stable than when running on Windows.  I gave up.

I don't mean to sound critical -- this is an excellent project that has made this approach to modernizing Fortius trainers viable for many people.  But Linux is effectively unsupported as the core developers are not using that platform, and the FortiusANT project might as well save people trouble by saying as much.  Linux and MacOS support is reliant solely on Python's cross-platform portable nature and clearly that is imperfect.

It would be nice to simplify things a bit with a command line only version, decoupling some of the unnecessary and painful dependencies like wxPython and Pygame, but I won't formally request that until I am prepared to offer my services in support :)

mattipee, 2020-12-24:
> It would be nice to simplify things a bit with a command line only version, decoupling some of the unnecessary and painful dependencies like wxPython and Pygame, but I won't formally request that until I am prepared to offer my services in support :)

@WouterJD , @TriScott , as for wx, if the mode constants get modes out of FortiusAntGui, then I've had it running without wx. Maybe half a dozen references. That would be the first quick win.

switchabl, 2020-12-24:
> 
> But Linux is effectively unsupported as the core developers are not using that platform, and the FortiusANT project might as well save people trouble by saying as much.

I mostly agree with this assessment. I use Linux a lot myself, but not for FortiusANT. I normally run that on Windows along with Zwift. At the moment, it is hard to recommend the Linux version to anyone who is not an advanced user and can fix problems themselves. I had it running fine on Fedora 31, but I still haven't managed to get it to work properly on Ubuntu 20.04.

Some thoughts on how to improve the situation:
- Include the udev rules in the repo
- Replace old (bad) instructions with those above (in particular, don't suggest installing unnecessary libraries, messing with the global python installation, removing kernel modules or running FortiusANT as root)
- Dependency hell: It is maybe not quite as bad as it seems. I don't _think_ `pynput`, `keyboard` and `pygame` are actually required (I've opened #183 to clear this up). `pyinstaller` and `wheel` are more related to building/packaging and should not be needed for end-users. `pyusb` and `numpy` from the Ubuntu repo seem to work fine. So if wx becomes optional, installing FortiusANT cli could be as simple as
```
sudo apt install git python3-usb python3-numpy
git clone https://github.com/WouterJD/FortiusANT.git
cd FortiusANT
sudo install -m 644 supportfiles/udev/*.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
```
- The wxpython in the Ubuntu repos (`python3-wxgtk4.0`) crashes immediately for me. I don't really know why yet. If this could be fixed, this should be all the dependencies, no venv required:
```
sudo apt install python3-usb python3-numpy python3-wxgtk4.0
```
- I have been thinking about running the Linux version myself for a while with the hope of fixing some stability issues.
- Maybe have pre-built packages eventually, at least for Ubuntu. I think mostly we need to start using setuptools and make sure we don't try writing log files to the executable directory.

WouterJD, 2020-12-24:
Change UseGuid in constants to False and you're done

AndrewB53, 2020-12-24:
Since yesterday I repaired my kubuntu partition and installed FortiusAnt using @switchabl 's method.  It was running okay apart from  the HRM , as previously described. I also installed the Windows version on a desktop  PC and got that working too, but with the same HRM problem. I found that the  resistance appeared to vary as I was pedalling and I am not sure of the reason for that. I also had a problem with the bike it wouldn't run on the big ring so I couldn't  get to 40 kph. The bike had previously  been on an old Elite trainer and was okay, so I was unsure if the resistance unit was the problem.

AndrewB53, 2021-01-21:
**Progress Update**
Following @switchabl 's instructions and eventually getting the dependencies I have succesfully installed FortiuaANT onto my Kubuntu laptop. After swapping my Anself (CYCPLUS) dongle for pair of Dynastream dongles from Wiggle i connected to Golden Cheetah and was able to use the i-Flow. However I did find that FortiusAnt had a tendency to crash. I'm not sure why but it could be the laptop is not up to the job of running both apps at the same time. So I am now running FortiusAnt on a Win 10 desktop machine and Golden Cheetah on the laptop. So far it is working quite well, although today the speed connection wasn't very good so my "ride" was only 0.5 km. However, this is further than I really moved. Using Golden Cheetah it doesn't really matter as the power is the most important metric. But it would be an issue with Zwift and similar platforms. So I guess that's now closed, thank you for the support.

closed 2021-01-21 by AndrewB53

switchabl, 2021-01-21:
@AndrewB53 Thanks for the update. I have also noticed that FortiusANT tends to have stability problems on Linux (I have one Fedora laptop where it works fine, but on another with Ubuntu it was barely usable). So I am aware of this and I have a hunch that it has something to do with wx and multiprocessing, but I have not had time to follow up on it. For the moment, my focus is on helping to improve core functionality (in particular calibration support and power accuracy across all different trainers) and ease-of-use on Windows because I think this will benefit the largest number of users. Hope you understand.

I think almost all CTPs (in particular Zwift) basically ignore the wheel speed and do their own calculation based on power. Your issue _may_ have something to with a bad radio connection/interference (in this case putting the dongles on extension cables may help). If the problem persists, you could open a new issue so we can investigate properly. Even if it doesn't bother you personally, fixing it may help other users later on.

AndrewB53, 2021-01-21:
@switchabl I was looking at the speed issue in the Golden Cheetah forum and there was a link to the Github wiki which said not to connect the speed+cadence and power channels but only to use the Fe-C. So I will try that next time I do a session. The dongles are about 90cm apart, but I could try and get them closer.  Also today i did youse my mobile phone with bluetooth headphones so there could have been some interference from those items. I'll post a follow up on that one.