WouterJD/FortiusANT #178: Unstable watt valus in Zwift...

oltasvno, 2020-12-22:
Took my first trip on the Zwift today and am not happy!
Unstable watt values, tried to hold 220W over different time intervals and watts oscillate from 0w to 270w, I could not hold exactly 220w ...........frustrating.
Somthing i missed?
Any tips?

Tacx Fortius
Headunit : Blue Box T1942
Clean installation
1 PC with 2 Ant+ dongles

clearwave1, 2020-12-22:
I have the same hardware setup as you and had the same problem starting with version 4.0.

See the thread: [https://github.com/WouterJD/FortiusANT/issues/173](url)

Try version 3.8 until this gets fixed.

oltasvno, 2020-12-22:
Thanks for reply!
@clearwave1 Can I have both 4.0 and 3.8 installed at the same time?
I also have a kind of windows sound (beep bloop) every 10 seconds after I start up FortiusAnt, do you have it?

clearwave1, 2020-12-22:
Yes. I use the Windows executable and they just run (e.g. they are not installed in Windows). I have both .EXE files in the same folder and just click on the version I want to run. I'm back to using the 3.8 version until 4.0 is fixed.

WouterJD, 2020-12-23:
That's absolutely OK.
FortiusANT does not initiate sound, so I have no explanation for that

oltasvno, 2020-12-23:
@WouterJD Appreciate all the help!
Can't find version 3.8 for download ???

oltasvno, 2020-12-24:
Merry Christmas everyone!
Tried last night the new software that was released 21.12.20.
Now my roll became completely impossible to cycling on ..... I can not even get the roll up to 40 km / h to run runoff.
The roller just chops and releases and makes it impossible to ride ...
I have checked the air pressure on the tire to 100 PSI.
On the previous software, the runoff matched well with rolling the rear wheel for approx. 7 sec. after I stopped pedaling.
PS! The roll worked as it should when I put it away a few years ago on Tacx software.
Anyone else experiencing the same as me?

WouterJD, 2020-12-24:
Release 4.1 published; should solved the problem

closed 2020-12-24 by WouterJD

oltasvno, 2020-12-24:
@WouterJD As I said in the previous post, I experienced this with the latest version of software 4.1.
Did the latest version works well for you  @clearwave1 ?

clearwave1, 2020-12-24:
Here is version 3.8 in case it helps you.

[FortiusANT 3.8.zip](https://github.com/WouterJD/FortiusANT/files/5740579/FortiusANT.3.8.zip)

oltasvno, 2020-12-24:
@clearwave1 Thank you and happy christmas!