WouterJD/FortiusANT #180: How to get TTS4 and FortiusAnt to coexist?

MustangMendonca, 2020-12-23:
Forgive me if I have put this in the wrong place,

Hardware setup:
Windows 10 Pro desktop system successfully running TTS4 for years. T1930 Tacx Fortius package, T1941 Motor Brake and T1932 Head Unit.

Solution Sought:
Looking to add the ability to use TDA with/through FortiusAnt

Issue:
When I install the libusb-win32 drive it removes the Jungo VR-interface driver and makes TTS4 unusable as it can't find the driver. I do not intend on having TTS4 and FortiusAnt running at the same time.

I have looked everywhere and have found some reference to this being possible and I have followed the instruction in the FortiusAnt manual but it only lays out the process for uninstalling the Jungo driver. When I install the libusb-win32 drive it removes the Jungo VR-interface driver and makes TTS4 unusable as it can't find the driver. Anyone who has done this and or knows where to point me, I would be very grateful!

Thanks in advance!

WouterJD, 2020-12-23:
Welcome @MustangMendonca to the FortiusANT community
- - -
I'm always curious to know who I communicate with and where FortiusANT is used.
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

As you have found, TTS/TacxFortius coexistance is discussed in usse #99 and the manual.

switchabl, 2020-12-23:
@MustangMendonca I guess it is not spelled out explicitly how to switch drivers anywhere.
- First, _don't_ uninstall the Jungo driver. This should almost never be necessary. If you already have, re-install by running TacxDriversSetup.exe from the TTS install folder (normally C:\Program Files (x86)\Tacx\TacxTrainersoftware4\TacxDrivers).
- Just install the libusb-win32 driver on top of the existing one (either manually as described in the manual or with the new all-in-one installer #168: https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T19xx.exe)
- To switch back, got to the Device Manager, find the libusb-wine32 Tacx T1932 device, right-click, choose "Update Driver", choose "Browse my computer for driver software", then "Let me pick from a list of drivers on my computer". Pick the "VR-Interface" driver.
- To switch to libusb-win32 again, do the same but pick the libusb-win32 driver instead ("Tacx T1932" if you used the all-in-one package).

Alternatively you can switch drivers by running TacxDriversSetup.exe or libusb-win32_T19xx.exe when you want to use one or the other (recommended in particular for the T1942 because switching by hand is more complicated there).

A third option is to run TTS through FortiusANT. Leave the libusb drivers in place. Start FortiusANT first, then TTS. In TTS, scan for new hardware, FortiusANT should show up as a Tacx NEO.

WouterJD, 2020-12-24:
From the point of view
- I start/stop TTS
- I start/stop FortiusANT
coexistance is not possible without the described driver-switch (which could be considered "installation")

Please respond whether the third option works, I will then add to the manual.

MustangMendonca, 2020-12-24:
I will try this and report back. It won’t be until next week.

tschmidty69, 2020-12-30:
I simply installed the libusb driver as a filtering driver and have had no issues with using either program.

MustangMendonca, 2020-12-30:
I have done some testing and report the following: 
1.	Assuming you are in a current state of having the Jungo Driver installed, it is not necessary to uninstall the Jungo driver. The libusb-win32 driver can be installed on top of the existing Jungo driver (either manually as described in the manual or with the new all-in-one installer from #168: https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T19xx.exe) I only tested using the inf-wizard.exe in option 2 of the user manual. 
2.	 To switch from Fortius ANT back to TTS4, go to the Device Manager, find the libusb-win32 driver, right-click, choose "Update Driver", choose "Browse my computer for driver software", it will find the Jungo driver and installs it.
3.	To use Fortius ANT again simply switch to libusb-win32 driver by using the process in 1. Above. 
Regarding @switchabl’s third option to run TTS through Fortius ANT. I did test leaving the libusb drivers in place. Starting Fortius ANT first, then TTS4. The following was observed:
Unfortunately, Fortius ANT does not show up as a Tacx NEO as @switchabl suggested. Fortius ANT shows up as FE-C compatible device in TTS4 when Add Device is selected in device manager found under the Setup tab of TTS4. It will connect successfully thereafter but I discovered the handlebar controller does not control any features of TTS, as it is controlling the feature of Fortius ANT. The computer mouse and keyboard are required to navigate TTS4. Calibration of the motor brake must be performed in Fortius ANT as calibration does not function in TTS$ in this configuration. Of note, once the libusb device driver has been used with TTS4 any subsequent use of TTS4 requires a manual reinstall of the motor brake by using Add Device from the Device Manager. Something must have been corrupted and I will investigate to find out why this is happening.
At this time, I would not recommend switchabl’s the 3rd option as TT4 navigation functionality is lost and the ensuing motor brake issue.

MustangMendonca, 2020-12-30:
> 
> 
> I simply installed the libusb driver as a filtering driver and have had no issues with using either program.

@tschmidty69 can you kindly elaborate? I am not sure how I would do this.

switchabl, 2020-12-30:
@MustangMendonca Ah, yes, I didn't think of that, sorry. There is no way to transmit the button presses through ANT FE-C. I guess we could emulate a T2022 ANT head unit to relay the buttons if we really wanted to. But making the buttons programmable in FortiusANT would likely solve the problem as well (and would be more helpful in general because it could work with Zwift and other software as well). I plan to work on that next.

@tschmidty69 The filter driver is an interesting option though. I seem to remember it wasn't recommended by the libusb-win32 developers (presumably it caused problems with some devices), so we probably shouldn't use it by default. But I will look at making a pre-packaged installer with the filter driver as an alternative option for people who want to keep using TTS4.

MustangMendonca, 2020-12-30:
Thanks for the comments @switchabl. Also the steering unit does not function in your option 3 when tested. Forgot to mention that :). I wish I could help with the programming part of this but just not my skill set. Happy to test anything you come up with. 

Regards

tschmidty69, 2020-12-31:
@MustangMendonca In the libusb package under the bin/amd64 folder is an installer, install-filter-win.exe. WIll give a gui that allows you to select what to attach the driver to, select Fortius Unit and that was all I needed to do. Zwift has been really solid with maybe one or two momentary dropouts but not sure if they are related

@switchabl I'd be curious what they say the issues may be.

switchabl, 2020-12-31:
@tschmidty69 There doesn't seem to be much detail in their wiki:
https://sourceforge.net/p/libusb-win32/wiki/Home/
> However, the device driver mode is the preferred way to use libusb-win32 followed by using device filter driver mode. The class filter driver is not recommended to be used.

I seem to remember there being a somewhat starker warning along the lines of: can cause BSODs, mess with power management, don't distribute to end-users. But it looks like the situation may have improved.

In any case my more immediate problem is that the filter installer seem designed to work with currently(/previously?) attached devices, which is not ideal for making an installer package. I think the same can probably be achieved with a suitable .inf file, but I will need to experiment a bit with that.

* Anyone who just wants to use FortiusANT should use the all-in-one device driver package (#168). I guess this will become the officially recommended option soon if there aren't any problems reported.

* Advanced users who want to keep using TTS4 seamlessly can try to install the filter driver on top of the Tacx/Jungo driver manually instead (https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/libusb-win32-devel-filter-1.2.6.0.exe/download). I will try to make a one-click package for that, but it may take a while.

The third option of running TTS4 through FortiusANT should still work, but as @MustangMendonca noted it is unfortunately somewhat limited as button navigation and steering are not transmitted. Fixing this by emulating legacy ANT remote control and steering devices is theoretically possible if someone really wants to do it. But I don't know if it is worth the effort.

tschmidty69, 2021-01-03:
Yeah, the filter driver is super easy to install, start gui, select Fortius, click ok. so not sure you should worry too much about a one click installer and the all-in-one driver would be the best option for users coming into it that don't have TTS4 already. LOL, if you have TTS4, you have spent some time troubleshooting anyway so using the filter driver will be easy. I think a better use of time would be to make the windows  executable package friendlier so you don't need to start it with command line options, some more options in the gui that can be saved to a config file, etc. 

FWIW, I have bounced back and forth from TTS4 and Zwift and had zero problems with either driver in game and no stability problems either.

switchabl, 2021-01-03:
> Yeah, the filter driver is super easy to install, start gui, select Fortius, click ok. so not sure you should worry too much about a one click installer

Well, the same could probably be said about Zadig/InfWizard for the device driver. But still it turned out to be a major roadblock for users. The old Fortius head unit (T1942) is particularly confusing because it changes its USB VID after loading the firmware so you have to repeat the procedure.

> I think a better use of time would be to make the windows executable package friendlier so you don't need to start it with command line options, some more options in the gui that can be saved to a config file, etc.

Agreed, that would be a good thing. I had started some work on configuration file support (with the idea of adding a settings GUI later on) already a while ago. Now that device support is getting nearly complete (a lot of work was done on getting Flow, Bushido and Genius brakes working properly by myself and others lately), I plan to revisit that. Wouter and I have also talked about automatically detecting the trainer type. Once the `-t` option is no longer required, I think many users won't need to worry about command line options anyway.

> FWIW, I have bounced back and forth from TTS4 and Zwift and had zero problems with either driver in game and no stability problems either.

That sounds good. Ha, I'm not sure I ever had zero problems with TTS4 with any driver. I didn't have a chance to try myself yet (don't have my Fortius here at the moment), but I will.

cyclingflow, 2021-01-04:
A little comment on co-existing TTS and FortiusAnt.

If you can install TTS software in a virtual box, it will do the "driver switching" for you:
the VM will grab the device, remove it from he device-list on the host and make it available with the tacx driver in the VM.
After closing the VM, the device will re-appear again in the host, with the lib-usb drivers.

Probably not a viable option for many, because it will bring you other problems, such as getting the display-driver inside the VM right. Installation is a hassle. Not sure it would run smoothly. For me, when installing TTS3 and using it to just check out the configuration/setup of my units, it works fine.

tschmidty69, 2021-01-04:
@switchabl I don't have a ton of time but can start to take a stab at adding a config file and gui support. That is just the kind of "busy work" that it would be good to pawn off to someone else. Without getting into the code I would probably look at a .ini file and some basic drop down/tickboxes for options. If you or @WouterJD had any other ideas let me know.

switchabl, 2021-01-05:
@tschmidty69 Sounds good, we should probably create a new issue for that. I was looking at a json file because I wanted a bit more flexibility for things like custom keymaps (for head unit buttons and other remote controls). But I don't think there has been any serious discussion about it yet.

WouterJD, 2021-01-05:
I was just about starting off the GUI-for-parameters because too many requests for it.
If I create it, it will be complete (all the settings) and indeed using a json file, since that's the todays path to go.

WouterJD, 2021-01-18:
Implemented in version 5.1 Test V

closed 2021-01-18 by WouterJD

switchabl, 2021-01-20:
@WouterJD @tschmidty69 I didn't get the all-in-one installer working with the filter driver (not quite sure what the problem is). But installing it manually with `install-filter-win.exe` works well for me. Switching between FortiusANT and TTS is seamless.