WouterJD/FortiusANT #181: Cadence indication unstable

JSlewe, 2020-12-23:
Hi there, I am using the fortius Ant software for two weeks now, and I am also running Zwift.
I have an older Tacx Fortius T1932 and I use 2 ANT dongles on one laptop.

The (little) problem I have is that every 2 or 3 minutes the cadence indicator is all over the place it goes from -11234 to 20445 within a second and after 2 seconds it is stable again.  This is on Zwift, haven't seen it on the Fortius Antifier display.

It doesn't seem to influence the ride on Zwift only my leggs speed up very much.
I checked the sensor with a multimeter but it looks fine.
Any idea what is wrong here, are any ather people with the same problem??

Best regards, Jurgen

rcerrie, 2020-12-24:
> Hi there, I am using the fortius Ant software for two weeks now, and I am also running Zwift.
> I have an older Tacx Fortius T1932 and I use 2 ANT dongles on one laptop.
>
> The (little) problem I have [...]

I have the exact same problem. Cadence goes ballistic for a short while ( my avatar is spinning his legs for that period) but speed stays constant.

WouterJD, 2020-12-24:
Hi @rcerrie, @JSlewe 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14, something jurgen already did
- - -
This is very strange,  especially since Cadence is an unsigned char - so I would expect a value between 0 and 255.
Please do a ride and specify -d127 on the command line. This creates a logfile. Add the logfile to the next message on this issue and I will look into it.

JSlewe, 2020-12-24:
Hi @WouterJD I tried to upload the logfile but github says it is too big, therefore I put it in a ZIP file.
[FortiusANT.2020-12-24 17-00-28.zip](https://github.com/WouterJD/FortiusANT/files/5740639/FortiusANT.2020-12-24.17-00-28.zip).
I also noticed that a red led is blinking on the head unit when I produce more power (>260W), I suspect this also is the moment that the cadence goes out of control, but not always.

WouterJD, 2020-12-27:
[#181 Cadence indication unstable.zip](https://github.com/WouterJD/FortiusANT/files/5744658/181.Cadence.indication.unstable.zip)

If I take all lines "ReceiveFromTrainer() = " which is received from the trainer
and the lines "p=25(Trainer Data)" which are sent to Zwift

then I see that in all cases the Cadence sent = cadence received.

  |  ReceiveFromTrainer() |  p=25(Trainer   Data))
-- | -- | --
  | Cadence | Cadence
17:12:42,126: ReceiveFromTrainer() = hr=0   Buttons=0 Cadence=78 Speed=30.8 TargetRes=1874 CurrentRes=3052   CurrentPower=219, pe=1 Connected to Tacx Trainer T1932 | 78 | .
17:12:42,377: ReceiveFromTrainer() = hr=0   Buttons=0 Cadence=78 Speed=30.9 TargetRes=0 CurrentRes=3362 CurrentPower=243,   pe=1 Connected to Tacx Trainer T1932 | 78 | .
17:12:42,464: Dongle    send     : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe4, info="00 19 55 4e   da 1c f3 00 30" (ch=0 p=25(Trainer Data)) | . | 78

JSlewe, 2020-12-27:
Hi Wouter, the ANT dongles are only 2 cm apart, but I wil do some test placing one dongle a little further on an extention cord.
I also looked at the log file and couldn't find any strange measurements, do you know if there are any log files from Zwift?
Another question, what does the pedal stroke analyses do?

WouterJD, 2020-12-27:
If you start with -g -A yiu can see the radar graph of power/pedal angle, showing whether power is regular with regard to angle. Not perfect but indicative

BrRoBo, 2020-12-27:
@WouterJD Chiming in.   I am also now seeing the same cadence issue with my Fortius T1942.  This just started occurring midway through last week.  I'm **not sure when I picked up the latest Zwift update, but I know they had one recently published.**  

That said, at the beginning of last week I moved from a fairly weak laptop to a fresh Windows 10 install on a slightly better ASUS desktop with a i5 CPU and 16GB ram.  Zwift is using ~34% CPU and 94% GPU (it's just an integrated Intel HD that borrows 2GB from the system).  For a couple days everything appeared to be 100% smooth, then got the quirky cadence issue later in the week.  

I've only been running sessions Zwift over the last week.  Tomorrow for comparison I'll do a session with the Tacx Desktop App to see if anything weird shows up there.

BrRoBo, 2020-12-29:
My cadence issue in Zwift seems resolved as of today.  Can't pinpoint the specific resolution as I did a number of things.

1. grabbed the latest FortiusAnt.exe
2. completely uninstalled/reinstalled zwift to ensure getting the latest zwiftlauncher, etc.
3. defragged the OS hard drive even though it was a recent install.
4. today zwift also seems to have pushed another update, so not sure if that had the applicable fix or just a coincidence.

Rode a few sessions in Zwift with the longest ride being around an hour.  Did both solo rides and big group, pacer rides.  Zero issues with the previous cadence oddity!  My FPS has remained the same throughout, typically 25-45 fps with a 34 average.

I know this isn't a Zwift support thread, but since the other people here were reporting the same usage problem I wanted to verify too.  Even after upping the graphics resolution and features, everything worked smoothly.  Solved for me.

BrRoBo, 2020-12-29:
Oh, forgot to add... before I did any of the above changes I tested Zwift while running FortiusAnt in Simulation mode.  The cadence issue DID NOT reproduce for these tests regardless of how long or busy (number of pacer group riders) there were in Zwift.  I thought this was very strange, not being able to repro the issue but perhaps the Simulation's data stream might be different than with my "real" Fortius T1942.

WouterJD, 2020-12-30:
@BrRoBo when running simulation mode, the ANT-side does not know that - it's just another trainer. I do not expect the data stream to be different

BrRoBo, 2020-12-30:
Yeah, that is what I thought, so it was odd I could not repro the cadence issue when observing any Zwift sessions regardless of how busy those were.  Just a bit of a head scratcher as the issue was occurring frequently when riding the trainer for real and no other diffs AFAIK.  Anyway... rode Zwift quite a lot today in multiple sessions and had zero cadence errors.

BrRoBo, 2020-12-30:
Question about using the pedal analysis (-A) option since you mentioned it above.  I read your comments elsewhere that it increases the head unit poling frequency.  

Does enabling pedal analysis have any benefit for any of the reported data accuracy or any detriment on overhead (performance)?

WouterJD, 2020-12-30:
It does NOT increase any data accuracy but only displays the power/angle radial graph.

The graphical user interface (whPython library) is the biggest consumer I think. The GUI is optional (-g flag)

JSlewe, 2021-01-01:
Hi there, for me it is not solved. I put one dongle a bit further away with an USB extention cable, and there was no change.
Next I also reinstalled Swift on my windows laptop, thew version I had was 1.0.51. The new downloaded version is the same.
So also no result there. The laptop I use is olny a few month old, therefore I don't think it has anything to do with the windows installation. At the moment I use FortiusAnt v3.8, I could try the newest one?

BrRoBo, 2021-01-04:
@JSlewe I don't know where you get your version 1.0.51 from, but the current latest build is 4.1.1 from here: https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe

BrRoBo, 2021-01-04:
> 
> 
> It does NOT increase any data accuracy but only displays the power/angle radial graph.
> 
> The graphical user interface (whPython library) is the biggest consumer I think. The GUI is optional (-g flag)

Perhaps a placebo effect, but I swear running FortiusAnt without the GUI results in noticeably snappier resistance changes on my Fortius in Zwift.  One tougher sections for trainers is the Titans Grove especially going clockwise as there are many repeated roller hills of varying pitches +/-7% or so at the most.  Riding through there solo today at a good clip, the Fortius seemed to be completely in harmony with what Zwift was showing and what I felt under me.

I can't think of any way to quantify this feedback, just my perception of better performance without the GUI using up 2% CPU (without the GUI, the FortiusAnt process is hardly registering CPU usage, at least in the Perf Mon list).

Cheers.

WouterJD, 2021-01-04:
@BrRoBo run two test, create the logfiles, JSON, TCX and compare.
It's quite some work, but should explain the situation.
I have no theoretical explanation

closed 2021-01-09 by WouterJD

JSlewe, 2021-01-11:
> @JSlewe I don't know where you get your version 1.0.51 from, but the current latest build is 4.1.1 from here: https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe

I was talking about the Zwift version 1.0.51
At the moment I also tried the lastestFortiusAnt version 5.0 , but still the same cadence issues.

WouterJD, 2021-01-12:
Hi all, please consider to use the latest version, see #198 
I do not expect different behaviour though

JSlewe, 2021-01-13:
> Hi all, please consider to use the latest version, see #198
> I do not expect different behaviour though

@WouterJD Do you also have a windows executable for this version?

WouterJD, 2021-01-14:
Yes: https://github.com/WouterJD/FortiusANT/blob/5.1%2B4.2-Quality-upgrade/WindowsExecutable/FortiusANT.exe