WouterJD/FortiusANT #183: Unnecessary dependencies?

switchabl, 2020-12-24:
While working on new installation instructions for Linux (#176), I was looking at the Python packages required for running FortiusANT. The `requirements.txt` file currently looks like this:
```
keyboard
numpy
pyinstaller
pyusb
pygame
pynput
requests
wxPython
wheel
```

But I haven't been able to find any place in the code where `pygame`, `pynput` or `keyboard` are actually used. Am I missing something or could these be removed?

WouterJD, 2020-12-24:
Please check in your branch and remove when obsolete.
Perhaps it is remaining from first version where wx was not used

switchabl, 2020-12-27:
What I have found:
- everything seems to run fine without these
- I could not find any place in the code where they are used
- I have extracted the latest FortiusANT.exe and they are not included (so PyInstaller apparently decided they are not necessary)
Conclusion: Since you are not aware of any place they are used (anymore) either, they are obsolete. I will remove them in the Genius branch before merging.

WouterJD, 2020-12-27:
👍

WouterJD, 2021-01-04:
Implemented in release 5.0; can be closed.

closed 2021-01-04 by WouterJD