WouterJD/FortiusANT #186: Question regarding Rundown -R usage

BrRoBo, 2020-12-27:
Trying to use the new -R option to reduce the required Rundown wattage for my wife.  On the Fortius trainer a 100w initial start is simply too hard for her (I can just about breath on the pedals and make them turn, but she needs a lot less to get going.  FWIW, once she gets going she can smoke a lot of other women because of her very light weight - 93 lbs).

So, I've tried many format alternatives to using this setting of 50W and just can't seem to find the correct one.  Here's one example...

--
G:\>FortiusANT.exe -d 127 -R power=50
15:47:47,483: Command line error; -R incorrect Runoff definition power=50
15:47:47,488: FortiusANT started
15:47:47,489: -c False
15:47:47,490: -C 0/0
15:47:47,490: -d 127 (0b1111111)
15:47:47,490: -D None
15:47:47,490: -H None
15:47:47,490: -p 1.0
15:47:47,491: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
15:47:47,491: -t False
15:47:47,491: ------------------
--

Tips?

Thanks!

WouterJD, 2020-12-28:
-R RUNOFF	The runoff procedure can be customized: maxSpeed/dip/minSpeed/targetTime/power. 

Using the -R command-line parameter, the procedure can be customized:
•	maxSpeed = the speed that must be reached, default = 40 km/hr (allowed = 20…50)
•	dip = the speed under maxSpeed when the runoff timer starts, default = 2 km/hr (allowed = 0…5)
•	minSpeed = when the runoff-timer stops, default = 1 km/hr (allowed = 0…10)
•	targetTime = runoff target time, default = 7.2 seconds (allowed = 0…10)
•	power = the power during the runoff, default 100W (allowed = 0…500)

Refer "FortiusAnt (gui, autostart, runoff).bat" in StartUp folder and change -R as follows:

`-R ////25` should do the job, leaving the first four parameters default, changing the 5th

closed 2020-12-28 by WouterJD

BrRoBo, 2020-12-28:
Thanks Wouter.

No way I would have guessed it since the updated manual does not show 
the / format let alone needing to use 4 of them :)  Adding the example 
to the doc should make more sense of it.

Appreciate your patience too.  I have only accolades for what you have 
done and continue to do on this project!

Cheers.

WouterJD, 2020-12-28:
Thanks, in the next update :-)
The text is a copy from the manual-in-draft