WouterJD/FortiusANT #187: tacx returns unsufficient data,    Len 24

albertjer, 2020-12-29:
Hi,

installed tacxfortius ant a few days ago, but when i calibrate i get the message:

tacx returns unsufficient data,    Len 24

what can i do about it?

i have a 1932 fortius,
magene speed meter,
stages powermeter,

WouterJD, 2020-12-29:
Hi @albertjer 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used.
- - -
Check loose contacts, not only signal- but also power-cables.

BrRoBo, 2020-12-29:
I've seen this error if the libusb-win32 driver isn't properly linked to the trainer device at the time when FortiusAnt is run.  

You'll need to provide more info such the full text output from the console and running FortiusAnt with debug turned on (ie: add "-d 127" to the startup options).

cyclingflow, 2020-12-30:
(and to state the obvious as mentioned in the error message: do make sure cable between brake and head unit is correctly connected, the black cable with telephone-like connectors)

TacxBiker, 2021-01-05:
Hi, I also do have this problem. It seems that FortiusANT is working well, but it doens't seem to find the right trainer connected.
I have a Fortius (T1941 motor unit and T1932 headunit)
FortiusAND will detect the headunit nice, but no motor brake, so no calibration.

Little work around  (as supposed bij WouterJD) is edditing usb.js 
    MotorBrake              = **True**          # Fortius motorbrake, supports calibration

This enables the calibration, but still the Len 24 error on the command line.
[FortiusANT.2021-01-04 19-31-47-edit.log](https://github.com/WouterJD/FortiusANT/files/5768934/FortiusANT.2021-01-04.19-31-47-edit.log)
[FortiusANT.2021-01-04 19-29-59.log](https://github.com/WouterJD/FortiusANT/files/5768935/FortiusANT.2021-01-04.19-29-59.log)
[FortiusANT.2021-01-04 19-30-28.log](https://github.com/WouterJD/FortiusANT/files/5768936/FortiusANT.2021-01-04.19-30-28.log)

The FortiusANT.2021-01-04 19-29-59.log and FortiusANT.2021-01-04 19-30-28.log have no edited usb.js
The FortiusANT.2021-01-04 19-31-47-edit.log is made on a ride with a edited usb.js. Also the log file is cutdown to a smaler file, not the whole ride is in the logfile anymore

WouterJD, 2021-01-05:
Download "Fortius Antifier v5.1 test I" from branche "4.2-Quality-upgrade"
specify -t Motorbrake which causes that -if no message returned- Motorbrake is assumed

TacxBiker, 2021-01-06:
-t Motorbrake works well

thanks

WouterJD, 2021-01-08:
Great!

closed 2021-01-08 by WouterJD