WouterJD/FortiusANT #188: Getting i-Magic working and then working with FortiusANT

ws25iy93hk, 2020-12-29:
Hello all.  I know that I'm going to start simple and work my way into Zwift but hopefully this will help others like me work from the very start all the with through.

I have an i-Magic trainer with the steering unit that I have not used for some time (read 15 years).  I have found the Fortius software online and have it installed on my Win10 pro machine.  But I'm unable to get the trainer to be seen on the software.

I have a green head to the trainer but I don't see any numbers on the device other than the SN.  When I plug the USB cable into the head unit from the computer, I'm getting a yellow light on the head unit.  I have checking the power going into the trainer as 122v as I'm in the US.

Because it has been so long, I don't even remember what is needed to make the trainer work at this point so I'm hoping that this group can help me with the details as it is the closest thing I have found that seems to know what they are doing.

Thanks for any help in advance and sorry if I have not included information that you will need to help but please just ask.

BrRoBo, 2020-12-29:
The original Tacx Fortius software is quite obsolete and not compatible with Ant+ trainers.  If you want to run current/modern Tacx software you'll need the Tacx Desktop Application (TDA) for Windows 10.  Otherwise, Zwift or any other Ant+ trainer software like Rouvey, RGT Cycling, Trainer Road, etc.

Also, if you have installed the old Fortius software, it probably installed the old USB drivers as well.  You'll need to either uninstall all that, or review the FortiusAnt user manual to read about how to make libusb-win32 drivers coexist with the old Tacx drivers.  Recommendation would be to uninstall all the old Tacx stuff including the drivers, then proceed with the FortiusAnt configuration per the user manual instructions.

ws25iy93hk, 2020-12-29:
I can understand and know my stuff is before the stone age.  I assume you are saying that the TDA will work with my i-Magic.  As I understand it it would be from the i-Magic via the USB to the TDA for power and brake control and with an ANT+ dongle for HRM.  
How can I know what versions of the hardware I have?  I don't see anything on the motor unit or the head unit.
I have some Wahoo stuff now and just using the Android software on a phone so I do understand the ANT+ protocols.

ws25iy93hk, 2020-12-29:
Sorry, I'm still reading the manual.  I have the old solid green head unit so I must have the T1902

BrRoBo, 2020-12-29:
Yes, TDA and all the other trainer software I listed will take the place of your old Fortius software and control your i-Magic trainer in the similar manner.  

The simplest procedure is to run the FortiusAnt bridge app first to initialize the USB head unit and start broadcasting its USB data as Ant+ FE-C data (Ant+ dongle 1), then start your newer trainer software to receive the Ant+ data (will use Ant+ dongle 2) just as if it were from a real FE-C trainer like the Wahoo Kickr or Tacx Neo.

WouterJD, 2020-12-30:
@ws25iy93hk 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - 
- 
- @BrRoBo I do not have much to add; please note that BLE is implemented in the most recent version. -b activates the bluetooth interface at FortiusANT side; for BLE-installation check [documentation](https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md).

BrRoBo, 2020-12-30:
Wow.  You guys are rolling out the features!

mikeherriman, 2020-12-30:
I can also comment on the BLE implementation.  IT IS FANTASTIC!! And really quite easy to get up and running (Follow the Bluetooth.MD) file in the doc folder. My install uses Win 10, BLE Dongle and USB Trainer.  Just completed a Zwift ride this morning using Zwift on iPhone11.  My PC has 2 ANT+ dongles and 1 BLE dongle.  This will allow for iPhone11 use, AND/OR PC applications for riding.  
IF I were in your position, I would ditch the old Fortius software and go with the newer support apps mentioned above.  As for getting the USB driver for your trainer, the quickest and easiest way is to download Zadig and install the driver using that software.

Good Luck!

switchabl, 2020-12-30:
> As for getting the USB driver for your trainer, the quickest and easiest way is to download Zadig and install the driver using that software.

There is now also a pre-packaged installer for the USB driver, just download and run: https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T19xx.exe

ws25iy93hk, 2021-01-02:
Thanks for all the help so far.  I have the software seeing my T1902 \ iMagic device.  I'm using the TDA on a laptop without a built in ANT+ device right now.  So this is my next question.  Sorry if this is all fundamental stuff.
Is there any way to use the current setup as is?  I attempted to use this setup but didn't get any resistance from the trainer or make any of the gauges move on the screen.
If I have to use another app, has anyone attempted RGT?  In the middle of the US, and will my 10+ years off, I'm not looking to have a full time annual subscription for 4 months of usage.

mikeherriman, 2021-01-02:
I use both RGT and Zwift.  When using RGT, I use the -G50 switch to “best simulate” the Zwift experience.  This sends 1/2 the slope value to the trainer. Zwift uses a slider bar within the app to control the slope level sent to the trainer, and is at 50% by default.

BrRoBo, 2021-01-02:
@ws25iy93hk from my experience and feedback online about Ant+ FE-C connectivity, that broadcast signal can takes some time to show up in an application like TDA, Zwift or whichever.  I have had this occur on occasion, but for the most part it does eventually appear in the TDA - Device Manager or the Zwift - Paring screen. 

You can verify the FortiusAnt startup looks all good by adding the debug switch "-d 127" and reviewing the output text for apparent errors.  Perhaps even upload that text to this thread for review. 

To verify you have FortiusAnt working with your trainer, one option is to go into the "Runoff" mode, do some riding and see if the FortiusAnt gauges all react as expected.

WouterJD, 2021-01-02:
Run in manual mode; then you can fully control and check

ws25iy93hk, 2021-01-07:
So I want to start by saying that I have not yet added an ANT+ device to the laptop that I'm using.  I have been able to get the iMagic to work with the original Tacx software now so I'm for sure the trainer is working.
When I start the FortiusANT+ software, I'm getting the DOS screen and then the GUI but the only button that is not grayed out is the Locate HW button.  None of the other buttons ever become active, just gray.  I have a screen shot of this and I then did the debug switches one at a time from 1 to 32.  
[FortiusANT.2021-01-06 17-49-09.log](https://github.com/WouterJD/FortiusANT/files/5778761/FortiusANT.2021-01-06.17-49-09.log)
[FortiusAntGUI.2021-01-06 17-49-45.log](https://github.com/WouterJD/FortiusANT/files/5778762/FortiusAntGUI.2021-01-06.17-49-45.log)
[FortiusANT.2021-01-06 17-49-44.log](https://github.com/WouterJD/FortiusANT/files/5778764/FortiusANT.2021-01-06.17-49-44.log)
[FortiusAntGUI.2021-01-06 17-50-21.log](https://github.com/WouterJD/FortiusANT/files/5778766/FortiusAntGUI.2021-01-06.17-50-21.log)
[FortiusANT.2021-01-06 17-50-20.log](https://github.com/WouterJD/FortiusANT/files/5778767/FortiusANT.2021-01-06.17-50-20.log)

[FortiusANT.2021-01-06 17-50-20.json.log](https://github.com/WouterJD/FortiusANT/files/5778769/FortiusANT.2021-01-06.17-50-20.json.log)
[image]

[FortiusAntGUI.2021-01-06 17-44-32.log](https://github.com/WouterJD/FortiusANT/files/5778752/FortiusAntGUI.2021-01-06.17-44-32.log)
[FortiusANT.2021-01-06 17-44-31.log](https://github.com/WouterJD/FortiusANT/files/5778754/FortiusANT.2021-01-06.17-44-31.log)
[FortiusAntGUI.2021-01-06 17-47-29.log](https://github.com/WouterJD/FortiusANT/files/5778755/FortiusAntGUI.2021-01-06.17-47-29.log)
[FortiusANT.2021-01-06 17-47-27.log](https://github.com/WouterJD/FortiusANT/files/5778756/FortiusANT.2021-01-06.17-47-27.log)
[FortiusAntGUI.2021-01-06 17-48-27.log](https://github.com/WouterJD/FortiusANT/files/5778757/FortiusAntGUI.2021-01-06.17-48-27.log)
[FortiusANT.2021-01-06 17-48-25.log](https://github.com/WouterJD/FortiusANT/files/5778758/FortiusANT.2021-01-06.17-48-25.log)

 
[FortiusAntGUI.2021-01-06 17-49-10.log](https://github.com/WouterJD/FortiusANT/files/5778759/FortiusAntGUI.2021-01-06.17-49-10.log)

marcoveeneman, 2021-01-08:
@ws25iy93hk Just my 2 cents, as far as i know (correct me if i'm wrong) you need an ANT+ dongle connected in order to get past the Locate HW button. If you don't want this, you can specify manual mode (-m) to operate without ANT+ dongle.

ws25iy93hk, 2021-01-08:
Ok, manual mode!  Sorry, I read that mode differently. Thinking it would
ignore power curve settings.  I'll have an update on about an hour.

ws25iy93hk, 2021-01-08:
Ok, it is working.  That was my bad.  I have been holding off getting the ANT+ USB dongle pending getting this working as I don't plan to do group rides or racing in the Zwift\TrainerRoad setups.  Just looking to keep a record of the training and maybe some day have a coach help keep me more fit.

I want to thank you again because after using the -m switch, i have a moving display.  I will be getting on Amazon later and making an order for a USB ANT+ device for my laptop since I have a USB cable attachment from the iMagic trainer.

BrRoBo, 2021-01-08:
Once you get the required dongle hardware, I suggest signing up for a trial of the Tacx Desktop App, or Rouvy, or RGT Cycling.  I think you'll be surprised how nice these newer, modern apps are than the old buggy Tacx trainer apps.

ws25iy93hk, 2021-01-08:
I already have the Tacx desktop and RGT installed.  Been reading reviews of the ANT+ USB devices to see if one if better than others.

closed 2021-01-18 by WouterJD