WouterJD/FortiusANT #189: Donation link?

BrRoBo, 2020-12-29:
Wouter, this is a great solution.  I don't normal seek to spend money, but I'd like to suggest that you provide a means for us to contribute something for this project effort.

WouterJD, 2020-12-30:
@BrRoBo thanks for the suggestion which has been done already a couple of times before.

The project provides different rewards.

For a starter, my own Tacx works well. Secondly it's real fun to see how world-wide collaboration works; many people added to the software by just asking questions or even providing significant effort. Not only quality improved, but also the support for the different trainers is a joint effort. Donation would be a kind third reward.

For me - this is the first time working in an environment like this ("old fashioned" in-house software developments untill now) and I have no experience how contributors look at a donation page.

It is certainly worth while, and suggestions welcome, since this project may live quite some time and support will be required. If not only asking questions then also making some tweaks where requested.

lello72, 2021-01-04:
I agree with BrRoBo. This work is creating a lot of value (fun of training + excitement from giving a second life to something condemned to obsolescence by dumb marketeers) and personally I'd like to contribute even if I have no IT skills.

WouterJD, 2021-01-26:
@BrRoBo @lello72 thanks for your suggestion; and so I learn from you guys.
@all I entered another new world; the github sponsorship.

Like the whole project: a great adventure!

Feel free to visit https://github.com/sponsors/WouterJD

hasise, 2021-03-11:
Tested to find "Wouter Dubbeldam" on Pay Pal, no hit I'm afraid :(

WouterJD, 2021-03-11:
@hasise ;

wouter.dubbeldam [at] xs4all.nl is the email address.

WouterJD, 2021-03-11:
@hasise I appreciate that; thank you!