WouterJD/FortiusANT #191: Error running latest version

tombandy, 2021-01-03:
[image]

Hi, 

FortiusANT has been working for me with the exception of ERG Mode in Zwift making pedalling impossible. To avoid any known issues I downloaded the latest version but now it won't run (see screenshot attached). I'm running FortiusAnt.bat in the startup folder as I have done in the past.

Many thanks, and thank you for creating this software :-)

Tom

tombandy, 2021-01-03:
Also in case it's relevant I had Norton Security flag a potential issue with FortisANT.exe - had to restore it from quarantine - 
[image]

WouterJD, 2021-01-04:
@tombandy 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

requests: should be in requirements.txt (but is added at a later version); `pip install requests` should solved the issue

quarantine: perhaps the exe is special in the eye of the virus-protector (it will contain special constructs); add it to the exclusion list

tombandy, 2021-01-09:
Thank you @WouterJD I've just successfully got it to run using the PIP command you suggested. I hope to setup the trainer tomorrow and try a ride out.

One thing I have found is even after buying a Tacx trainer specific tyre and carefully calibrating after warming up the tyre for a few minutes, the runoff times are very sporadic and will fall between 5s and 7.5s with no adjustments. I have the tyre pressure at 100psi but still it doesn't seem consistent. 

Then when performing an FTP test (the hour and 15 min one) in Zwift, with ERG mode turned on, when the resistance is increased the brake literally gets thrown on really hard and is impossible to pedal, to the point where Zwift turns off ERG mode automatically after a short time as my cadence hits the floor. I've done this a few times and wish there was a calibration feature in Zwift where I could check the trainer is working without a long warm up because it takes time to find it's not working again. I'd be keen to hear if you have any ideas.

Many thanks and again thank you for providing this software to make the older trainers work online.

Tom

WouterJD, 2021-01-09:
Easiest to try -m (manual power) so you are in control, create an export (either json or tcx) and analyse what happens

closed 2021-01-24 by WouterJD