WouterJD/FortiusANT #192: Installation instructions Raspberry - Bluetooth

wtal710174, 2021-01-04:
The intend is to create installation instructions for RaspberryPi OS. Since I have a RaspberryPi 3B this will be mostly base on this device. Let's exchange here to make this an instruction for all Raspberries around. 

I am running a Fortius Trainer (T1930) with a T1932 head unit and the T1941 motor combined with the VR Steering Frame.

My original instruction can be found here: https://gist.github.com/wtal710174/23081b986ab05c9a17facfec2d738202. However I would like to continue the work in this issue.

Using the instructions I have the Raspberry control the Fortius Trainer with only the USB cable connected to it. I run Zwift on my iPad or Laptop connected using Bluetooth. I run the FortiusANT without the GUI in a terminal.

# FortiusANT BLE on Raspberry PI 3B Installation

This is a short description of the necessary steps for the installation of the FortiusANT project by [WouterJD](https://github.com/WouterJD). The project is located here (https://github.com/WouterJD/FortiusANT).
The first part of this document is taken from the USB ANT Installation Instruction written by [martin-vi](https://gist.github.com/martin-vi) which is located here (https://gist.github.com/martin-vi/5206b6b26faf6f2d6121f6c8f300e44d).

## Prerequisits

- Raspberry Pi 3b or similar with a clean RaspiOS Buster full installation (https://www.raspberrypi.org/software/)
- Python3 version > 3.6 (Version 3.7.3 is included in the Buster full installation December 2nd 2020)
- Standard user pi or any other user which belongs to the groups sudo, ssh, dialout, audio, video, plugdev, users, input, netdev (not sure if you need all of these, however this worked for me)
- Enable ssh and/or VNC in the Raspberry Pi Configuration in Preferences (Interfaces tab), if you want to connect remotely to your Raspberry Pi
- Update your system (did not work for me during the initial setup, make sure that you run the apt update command at least before installing any other packages, running both commands does not hurt, just takes some time):
```
$ sudo apt update
$ sudo apt full-upgrade
```
- FortiusANT repository downloaded to the user home directory
via git:
```
$ sudo apt install git
$ git clone https://github.com/WouterJD/FortiusANT.git
```
or via download
```
$ curl -L https://github.com/WouterJD/FortiusANT/archive/master.tar.gz | tar -xzvf -
```
Now you should have a FortiusAnt folder in your home. Check with ls -all if the folder belongs to you.
## Install wxPython

Download pre-build [wxPython package](https://drive.google.com/drive/folders/1owWkRTsvcnvq7ZSRc6zHS5dip2AMvW79?usp=sharing) to home folder. This wheel package works for an ARM7l CPU.

Install package with
```
$ sudo pip3 install wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
```
If you can not use the prebuild package, you need follow the install instructions below to build your own wxPython by replacing the version with 4.1.1 (This can take up to 18 hours depending on the RaspberryPi you are using):

https://www.marcdobler.com/2020/05/17/how-to-compile-and-install-wxpython-on-raspberry-pi/
## Install required packages
```
$ sudo apt install -y python-virtualenv python3-virtualenv build-essential bluetooth bluez libbluetooth-dev libudev-dev
$ sudo apt-get install -y libatlas-base-dev # for numpy, https://github.com/numpy/numpy/issues/11110
```
## Setup virtual environment for Python
```
$ cd ~
$ virtualenv -p python3 --system-site-packages venv
```
Afterwards, the installation of packages for FortiusANT always requires the virtualv to be activated first. This also applies to starting FortiusANT.

The virtual environment needs to be activated for the following with this command:
```
$ source ~/venv/bin/activate
```
## Node JS installation
Follow the installation instruction on https://www.makersupplies.sg/blogs/tutorials/how-to-install-node-js-and-npm-on-the-raspberry-pi

For the Raspberry Pi 3B the following steps apply for version 14.15.3:
```
# Go to home
$ cd ~
# Download node 14.15.3
$ wget https://nodejs.org/dist/v14.15.3/node-v14.15.3-linux-armv7l.tar.xz
# Untar downloaded archive
$ tar xf node-v14.15.3-linux-armv7l.tar.xz
# Go into folder with unzipped files
$ cd node-v14.15.3-linux-armv7l/
# Copy folder content to 
$ sudo cp -R * /usr/local/
# Check if installed correctly
$ node -v
# Output: v14.15.3
$ npm -v
# Output: 6.14.9
```
After installing node to the system, you have to set it up for FortiusANT
```
$ cd ~/FortiusANT/node/
$ npm install
```
The npm install command will display several warnings which can be ignored.

Last step is to grant the node binary cap_net_raw privileges, so it can start/stop BLE advertising without root access.
```
$ sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)
```
## Install FortiusANT dependencies
```
# remember that an activated virtualenvironment is required
$ pip3 install -r ~/FortiusANT/pythoncode/requirements.txt  
``` 
## Create startup script for FortiusANT
```
$ cd ~
# Deactivate virtual environment
$ deactivate
$ touch FortiusANT.sh
# Open the new file with any editor you like, I prefer pico or nano
$ pico FortiusANT.sh
# Insert the following text including the !/bin/bash comment

#!/bin/bash

echo "Enter the sudo password, please."
read -s PW
echo Password entered. Thank You!

echo $PW|sudo -S service bluetooth stop
echo Bluetooth Service stopped
source /home/pi/venv/bin/activate
if [[ "$VIRTUAL_ENV" != "" ]]
then
  echo Virtual Environment set
else
  echo Virtual Environment not set
fi
echo Virtual Environment Set
echo $PW|sudo -S hciconfig hci0 up
echo Bluetooth for BLE on internal adapter started
cd /home/pi/FortiusANT/pythoncode
echo Starting FortiusANT ...
# Use any command line parameter as required. '-b' does start the command line based bluetooth interface 
# only, without GUI. Add '-g' for the GUI. More Details in the FortiusANTUserManual in section 4.2 
# starting on page 29. 
python3 FortiusAnt.py -b
deactivate
if [[ "$VIRTUAL_ENV" != "" ]]
then
  echo Virtual Environment set
else
  echo Virtual Environment not Set
fi
echo $PW|sudo -S service bluetooth start
echo Standard Bluetooth Service restarted
```
Save file with control+O and exit file with control+x and make script executable with the following command
```
$ chmod u+x FortiusANT.sh
```
## Give user access to USB port
Run the following commands to create an access file in /etc/udev/rules and reboot Raspberry to activate rule
```
$ sudo -i
$ cat << EOF > /etc/udev/rules.d/10-usbaccess.rules
# Allow users in group usbtacx to access usb
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664", GROUP="usbtacx"
EOF
# Create group usbtacx
$ addgroup usbtacx
# add user pi to this group
$ adduser pi usbtacx
# Reboot to activate rule
$ reboot now
```
## Let's fire it up
Connect your trainer to one of the USB ports of the Raspberry Pi and execute FortiusANT with
```
$ ~/FortiusANT.sh
```
Next, the calibration procedure will start with a pedal kick. After the procedure is complete, the bluetooth starts broadcasting identifying the Raspberry Pi as periphal device with the identifier 'FortiusANT Trainer'. This one you can connect to your CTP.

After the calibration you might see a couple of error messages. This happens while the http server is being established and the program is not able to access it. After 5-10 secondes the connection should be stable and you will see the normal output in the terminal like the one below.

```
20:08:31,698: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:32,700: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:33,702: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:34,703: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:35,705: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:36,706: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
```

switchabl, 2021-01-05:
Don't have a Raspberry PI 3B so can't test this directly, but it looks quite thorough! I have a few suggestions on how to improve this further (well, I hope so at least):
- I suggest putting the venv inside the FortiusANT directory, not directly in home
- Maybe node.js can be installed locally as well (instead of system-wide in /usr/local) with something like nvm (similar to venv). I think with nodeenv it might even be possible to install node directly inside the python venv.
- Passing through additional command line options from the script to FortiusANT would probably be nice.
- I don't think you need to activate the venv in the script, it should be enough to run `~/venv/bin/python FortiusAnt.py`.
- I wonder if hciconfig is actually necessary? I thought that the system Bluetooth stack is actually bypassed completely with the current implementation but I have to admit I didn't look at it in detail.
- The USB permissions could be made a bit narrower (similar to https://github.com/WouterJD/FortiusANT/issues/176#issuecomment-749847993), but on a Raspberry PI it is maybe not so critical I suppose.

Some more general ideas that may or may not make sense to persue:
- A simplified no-GUI version of the guide. I think if you run without GUI, you don't actually need wxPython anymore. So it might be sufficient to install the numpy, requests and pyusb packages with apt and do away with venv.
- It should be possible to give the pi user permission to start/stop Bluetooth with polkit, but I would need to look up how. That way it might be possible to run the whole script without sudo (and storing the password, I am not sure "read -s" is actually secure).
- Alternatively we could run FortiusANT as a systemd service and with a dedicated user. That won't work with a GUI, but it could be particularly attractive if you use a dedicated, headless Raspberry PI for your trainer. Then you might just set the the service to auto-start on boot. I think if you declare a conflict with the bluetooth service in the unit file it will even auto-stop that for you.
- I wonder if we could just make a docker container to save people the hassle of setting it up manually.

switchabl, 2021-01-05:
> Am I correct that I won’t need a Ant dongle anymore?

If the device you use Zwift/Rouvy/etc on has Bluetooth LE (or if you are using Zwift on PC with the Companion app on the phone) as well and you are not using any ANT-only power meter/sensor, yes, no ANT dongle required.

> I’m willing to share an image if I succeed. That should work for other users with the same RPi then right?

I think so, yes. There are a number of potential disadvantages to building images that way:
- they may not work on other Raspberry PI-models
- if you expanded the partition to the whole SD card, the image may become quite large (and can only installed on cards at least as big as yours)
- they may contain unnecessary (temp/log) files
- it is relatively time-consuming if you wanted to update it regularly

For that reason, images are usually built in an automated way on a PC/server instead. But that is quite a bit more work to set-up, so unless someone volunteers to do that, I think your image could still be quite helpful for others!

Not sure what the best options are long-term:
- distributing normal Debian/rpm packages and relying on the OS for dependencies would be the best solution, but that may not be feasable anymore with the node.js dependencies for Bluetooth
- an automatically built Linux image would be great for people using a dedicated Raspberry PI, but I am not sure how much work that is
- a Docker container would be more flexible and probably easier to build
- there is also AppImage, which would be similar to the binaries provided on Windows (that should also work with the GUI)

switchabl, 2021-01-05:
Not sure what the best options are long-term:
- distributing normal Debian/rpm packages and relying on the OS for dependencies would be the best solution, but that may not be feasable anymore with the node.js dependencies for Bluetooth
- an automatically built Linux image would be great for people using a dedicated Raspberry PI, but I am not sure how much work that is
- a Docker container would be more flexible and probably easier to build
- there is also AppImage, which would be similar to the binaries provided on Windows (that should also work with the GUI)

wouterstrand, 2021-01-05:
I'm stuck at the following step:

_The virtual environment needs to be activated for the following with this command:_

`$ source ~/venv/bin/activate`

I get the message the file or folder does not exist? Using command ls -all, it does not exist indeed in the home folder?

wtal710174, 2021-01-05:
Thanks @switchabl very good and helpful comments. Here my two cents:
- I will try and test putting venv in the FortiusANT directory, that should be an easy change
- 

> 
> 
> I'm stuck at the following step:
> 
> _The virtual environment needs to be activated for the following with this command:_
> 
> `$ source ~/venv/bin/activate`
> 
> I get the message the file or folder does not exist? Using command ls -all, it does not exist indeed in the home folder?

test "$ source --help", does this give you the help messages? If not something went wrong with your virtual environment install.

wouterstrand, 2021-01-05:
Wooops, noob at work here. I skipped a step, I've succeeded now. Continuing install on the RPi3.

wtal710174, 2021-01-05:
@switchabl Thanks a lot for the great comments. I will pick this up and try to work through your proposal. I do agree with almost all of it. I did try to run without stopping the bluetooth service and not using the hciconfig command, however this did not work for me. Maybe you have a suggestion there. Regarding the virtual environment in the script, this was just to make the startup as easy as possible. I have already altered the text to include the command line parameters for startup. Very nice suggestion there.

Regarding your general ideas:
- Non Gui version:  Yes this is a great idea. This will require changes in the requirement.txt files to prevent the check for the wxPython or a completely altered program version without the GUI calls. Not sure which option to choose. I prefer the changes in the Python to detect if wxPython is present and in case it is not ignore all the GUI options.
- It should be possible to give the pi user permission to start/stop Bluetooth with polkit: Not requiring sudo would be perfect. I have no experience with polkit. I will look it up and start working on this.
- Alternatively we could run FortiusANT as a systemd service: This would be my favorite solution, since it would allow me to power up the Raspi and start cycling without logging in and typing any command. I do think that you will still need the calibration process. Keeping the FortiusANT on Raspi running all the time seems to cause issues over time. I need to look into that more closely.
- Docker: YES! Unfortunately I have no experience at all with the docker concept. Therefore I could use some help to get this done. This would be a very convenient way of setup. I would as well prefer this to an image.

wouterstrand, 2021-01-05:
Well, Thanks a bunch. I successfully connected my iPhone to the Tacx via BLE. Tomorrow I will set my bike up on the Tacx and make an attempt with my laptop.

wtal710174, 2021-01-05:
Nice! Thanks for the feedback. Good luck with the test tomorrow.

switchabl, 2021-01-05:
> I did try to run without stopping the bluetooth service and not using the hciconfig command, however this did not work for me. Maybe you have a suggestion there.

It seems I was wrong, I looked at the bleno docs (the library uesd for the Bluetooth functionality) and apparently they are both necessary.
But it should be possible to set up permissions so we don't need root for them:
- `hciconfig`: According to https://unix.stackexchange.com/questions/96106/bluetooth-le-scan-as-non-root, this should do the trick:
```
sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hciconfig`
```
- bluetooth service: add a polkit rule (see https://wiki.archlinux.org/index.php/Polkit#Allow_management_of_individual_systemd_units_by_regular_users, replace wpa_supplicant with bluetooth)

> Regarding the virtual environment in the script, this was just to make the startup as easy as possible.

Yes, if you are using it interactively, it definitely makes sense. In a script it apparently isn't necessary most of the times as long as you call the right Python interpreter. If it is (in case FortiusANT expects the Python interpreter in PATH or something), I believe you can at least skip the `deactivate` step. The changes to the environment should revert once the script terminates.

> * Non Gui version:  Yes this is a great idea. This will require changes in the requirement.txt files to prevent the check for the wxPython or a completely altered program version without the GUI [...]

I was under the impression that the latest version of FortiusANT doesn't import wx in CLI mode. So I was hoping that
```
sudo apt install git python3-usb python3-numpy python3-requests
python3 pythoncode/FortiusAnt.py -a
```
may be all that is required (apart from the permissions and Bluetooth part). But maybe not, I haven't tried yet.

> * Alternatively we could run FortiusANT as a systemd service: This would be my favorite solution, since it would allow me to power up the Raspi and start cycling without logging in and typing any [...]

Yes, I wouldn't be surprised if running it continously uncovered some stability bugs.

I have some plans to improve calibration (save calibration value between sessions; allow calibration on request, not on startup; allow calibration requests through ANT/Bluetooth) that would probably help with the head-less case as well.

TacxBiker, 2021-01-06:
Hi, by excecuting all the commands wtal710174 it sure works perfectly on a raspberry 3b+
Also the GUI works

Adding 
/home/pi/FortiusANT.sh -a -b
to the /etc/rc.local starts up FortiusANT. 
One pedal kick and the calibration starts. After calibration the FortiusANT trainer is visible by bluetooth on a PC or iPAd

wouterstrand, 2021-01-06:
It seems as if there is no calibration for me? It just immediately outputs data and transmits BLE. Is it because I use the magnetic T190 brake with T1932 head unit?

mikeherriman, 2021-01-07:
I can also attest that the natallation and configuration steps work on a raspberry Pi 4.  I have a 7” touchscreen connected to the raspberry Pi HDMI port allowing for startup/shutdown without external keyboard or mouse.  Happily able to run Zwift app on iPhone, Zwift app on Windows with FortiusAnt connected via iPhone Zwift Companion app.

There does seem to be some GUI related issues, in that the GUI becomes unresponsive, and requires a kill command to close out.  Even though the GUI becomes unresponsive, FortiusANT continues to function.

Mike

wouterstrand, 2021-01-07:
How do I start the GUI? I attempted with -g but it didn’t work.

TacxBiker, 2021-01-07:
It is also possible to start FortiusANT by simply clicking in the script /home/pi/FortiusANT.sh in the filebrower.
Then the GUI is visible.

wouterstrand, 2021-01-07:
Thanks!

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

TacxBiker, 2021-01-07:
One thing, it is not possible to add opperants when clicking on the script.
I hust eddited the scrip by addeing te opperants -a -b -g

wtal710174, 2021-01-07:
I created a little add on which might be useful to others as well. When I started the GUI today for testing some stuff I noticed that the GUI is showing new versions in the status bar if available (very nice feature :-)). Therefore I implemented an automatic update in the script, to check for the latest version during start up and update the new files. In order to do this you need to implement the following.

Step 1:
```
$ cd ~/FortiusANT/
$ git remote add upstream https://github.com/WouterJD/FortiusANT
```
Step 2 - Get new code from repository (needs to be done in the FortiusANT directory)
```
$ git pull upstream master
```
If you implement this line in the script after the password section with
```
git -C /home/pi/FortiusANT pull upstream master
```
the script will always check for updates and download the files before starting FortiusANT.

wtal710174, 2021-01-07:
# Bluetooth without sudo for pi

I followed @switchabl 's advice. This worked kind of for me. So step 1 is complete ... pi can start and stop bluetooth, however a password is still required. Did anybody get this working without password request for pi so far?

wtal710174, 2021-01-07:
> One thing, it is not possible to add opperants when clicking on the script.
> I just eddited the scrip by addeing te opperants -a -b -g

@TacxBiker Sounds like you do not need a sudo password for the script anymore, how did you get this done? Or are you starting the script in a terminal after double clicking?

wtal710174, 2021-01-07:
On more idea for improvement. Some of the users here talk about running the Pi headless and just attaching it to the trainer to make it kind of an interface that does not require any input through a terminal or touchscreen. @switchabl  mentioned the service before. What about connecting an on/off switch to the IO-ports of the raspberry, which starts the script or service in on position and stops the script or service in off position. Anybody that did something similar in the past and has an idea how to implement this for the user pi (script) or as a service that listens to the switch status?

wtal710174, 2021-01-07:
@mikeherriman Thanks for the feedback!

wtal710174, 2021-01-07:
### Bluetooth without Sudo and updated script

_Based on the remarks from @switchabl above._ 

First give user the right to use hciconfig without sudo

```
$ sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hciconfig`
```
Then enter the command 'sudo visudo' to edit the sudoers file. In the section 
'# Allow members of group sudo to execute any command' add the following lines:
```
%usbtacx ALL=NOPASSWD: /bin/systemctl stop bluetooth.service
%usbtacx ALL=NOPASSWD: /bin/systemctl start bluetooth.service
```
This will grant all member of the group usbtacx the right to start bluetooth service without entering a password.
Now you can use `sudo service bluetooth stop` and `sudo service bluetooth start` as user pi without entering a password.

_(Comment: I tried to run the command without sudo, since this works for the hciconfig command. If you ommit the sudo you will have to enter a password)_

Now edit the FortiusANT script with `$ pico ~/FortiusANT.sh` and replace the content with the following code:
```
# FortiusANT Startup Script v0.2

# Remove comment below if you want automatic updates
# Execute 'git remote add upstream https://github.com/WouterJD/FortiusANT' in FortiusANT directory first before removing comment
#echo Getting FortiusANT updates from Github Repository
#git -C /home/pi/FortiusANT pull upstream master

sudo service bluetooth stop
echo Bluetooth Service stopped
source /home/pi/venv/bin/activate
if [[ "$VIRTUAL_ENV" != "" ]]
then
  echo Virtual Environment set
else
  echo Virtual Environment not set
fi
hciconfig hci0 up
echo Bluetooth for BLE on internal adapter started
cd /home/pi/FortiusANT/pythoncode
echo Starting FortiusANT ...
python3 FortiusAnt.py $@
sudo service bluetooth start
echo Standard Bluetooth Service restarted
```
Save the changed file and give it a try. Let me know if this works.

Now you should be able to start the script without any issue by double clicking the file in the file browser in the graphical interface or as before in a terminal with `~/FortiusANT.sh -b` to just launch the bluetooth interface.

mikeherriman, 2021-01-08:
I just updated the sudouser file as instructed, edited the FortiusANT.sh file as instructed.  I also added the switches inside the script (-a -g -H0 -c12).

Created desktop shortcut to the script.
Now can double-click the shortcut from Desktop and everything fires up.  
One thing that I am not sure about is when I double click the shortcut, I am prompted to Execute, Execute In Terminal, Cancel.
When I select Execute in Terminal, everything goes but within a minute or so, the GUI and terminal window become unresponsive, requiring CTRL+C to stop the script.  When I select Execute the GUI starts with no background terminal window and all seems to operate much better.  Any ideas?

TacxBiker, 2021-01-08:
@wtal710174 
I'm not sure why that is. when auto start the script by RC.local I don't need to enter a password.
when using the same script by clicking or command line in a terminal I do need to enter a password.

wtal710174, 2021-01-08:
@TacxBiker Which script are you using? Please make sure you have `#!/bin/bash`in the first line. I forgot to add this yesterday, which caused the issues I was having.
Regarding RC.local startup should launch the script as root, which could explain the behavior. You can double check by logging the output of the script in a text file using `whoami` to print the user name when executing the script.  
I have updated the script in the comment above to include the reference to the bash.

wtal710174, 2021-01-10:
### FortiusAnt One Button Controller for Raspberry

I have created a little python program (actually my first one ever) to control the headless Raspberry with a very simplistic interface (1 Button and a traffic light LED Module). See picture below.

If implemented in the .bashrc of the user pi it will start when the  raspberry is booted and indicate the status using the 3 LED's.

This is a quick and dirty first build, however it makes the control much easier and I do not need to use a keyboard or terminal anymore to stop and start or shutdown.

GPIO Pins are:
- GPIO 24 for the switch input
- Pin 17 for the 3.3V power to the switch
- RED LED on GPIO 22 (PIN 15)
- YELLOW LED on GPIO 27 (PIN 13)
- GREEN LED on GPIO 17 (PIN 11)
- GROUND is connected to PIN 9

If this is of any interest to anybody else, I would be interested in a more professional and rigid module to be connected to the GPIO pins.

Feedback and help to improve the code appreciated.  

[image]
[image]
[image]

[FortiusAntControl.py.zip](https://github.com/WouterJD/FortiusANT/files/5792800/FortiusAntControl.py.zip)

MeanHat, 2021-01-10:
Hi @wtal710174 - looks very good. I was planning to use an illuminated LED switch such as this https://thepihut.com/products/rugged-metal-on-off-switch-with-red-led-ring?_pos=3&_sid=0f4c526eb&_ss=r to shutdown the RPi Zero. At present my neighbour uses a keyboard to command a poweroff/shutdown.
[image]

wtal710174, 2021-01-10:
@MeanHat Thanks for sharing this. I like the button idea. Looks very nice. Maybe I can find a Case with GPIO Connector that is big enough to fit this button. However I checked on the website and I think I like this switch even better:
https://thepihut.com/products/rugged-metal-pushbutton-16mm-6v-rgb-momentary
Check it out. It has a RBG LED ring, which allows you to show the status of the process, similar to what I am doing with the traffic light. Letting the RED LED ring blink in different ways could to the same.

wtal710174, 2021-01-10:
I am having some stability issues with the nodejs server. The server process just disappears after a random time (3-6 minutes) while riding. The CPU load is at 12% max when this happens and the memory is about 25% used. So I do not see any resource issue. Does anyone have an idea? I will check the repository if I can find it in another issue. 
Interesting is that this happens when the FortiusAnt.py is started using the FortiusAntController.py program. The FortiusAnt program continues to operate (just no Bluetooth signal anymore). If I am starting the FortiusAnt program using my FortiusANT.sh bash script, this does not happen.

MeanHat, 2021-01-11:
Hi @wtal710174 
I now have my RPi 3B up and running in BLE mode and communicating with an Android phone running Zwift. I am short of a head unit but borrowed my neighbour's T1932 and it seemed to work OK (though I was not able to check the commands from Zwift were reaching the trainer. Brilliant - thank you!
By way of feedback (1) using FortiusANT.sh I received a "too many arguments" error when I entered 4 arguments (-a -b -d127 -s) and (2) with -b I received an error ("expected binary...).
I had previously set up a RPi Zero using system.d to automatically start FortiusANT which worked very well. I was interested in your script approach.
My next step (as well as sourcing a new head unit) is to set up another RPi Zero for Bluetooth. I was planning to do this without the GUI. I think there are a number of us interested in adding some install instructions for RPi in the user manual @WouterJD has created. I wonder if it would be helpful to have 3 sections such as (1) RPi baseline with ANT+ (2) RPi with Bluetooth and (3) RPi with Bluetooth without GUI (avoiding the wxPython issues). What do you think? I'd be happy to contribute/test etc.
Cheers, Keith

wtal710174, 2021-01-11:
@MeanHat Could you check the script you are using? I have altered the run without sudo script in the comment above, since the bash command line options where lacking the quotation marks. Not sure what this could cause. Since I did this I did not get any more errors. If this does not help I need figure out how to exchange with PM's to help you troubleshoot.
Regarding the manual: @WouterJD asked me to open this thread to get the bluetooth installation on Raspberry mature. Once we have all the bugs out we can start writing the instruction for the manual. If I remember correctly this is done in Word. The basic ANT instruction by martin-vi can be found here: https://gist.github.com/martin-vi/5206b6b26faf6f2d6121f6c8f300e44d as you probably remember. This will need some updates, since it is not based on the latest source code and requirements. 
I am still testing some more features for the virtual environment as suggested by @switchabl like the installation of the node.js directly in the virtual environment. Once I am making progress there I will post. 
The other thing remaining is the issue with the crashing node.js when starting the bash script using my FortiusAntController.py this might get solved with the implementation of the node.js in the virtual environment.
Once all that is done and approved by WouterJD to be mature, I would gladly accept some help getting all this wrapped up and ready for WouterJD's manual.

Lp64, 2021-01-12:
Hi All,
nice project. I excecuted all the commands from wtal710174 on a raspberry 4b like mikeherriman did. 
By executing the command  ./FortiusANT -b -g the GUI works. 
Clicking on Locate HW responds in: No (free) ANT-dongle found, Bluetooth interface available.
Terminal shows: 
17:36:05,374: Connected to Tacx Trainer T1932
17:36:05,678: Motor Brake Unit Firmware=6402 Serial=    0 year=2000 type=T190 Version2=0 MotorBrake=False

Clicking on Start in the GUI responds in: Connected to Tacx Trainer 1932
No (free) ANT-dongle found, Bluetooth interface opened to CTP

Terminal shows:
... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb002b050>: Failed to establish a new connection: [Errno 111] Connection refused'))

GUI is freezing. FortiusANT trainer is visible by bluetooth on a PC, but no chance to connect.
Any ideas; what is wrong? 
Ralf

wtal710174, 2021-01-12:
@Lp64 As long as you see this error:
´´´
... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb002b050>: Failed to establish a new connection: [Errno 111] Connection refused'))
´´´
your node server is not running. Which is normal for a few seconds until node is up and running. To eliminate the GUI as possible source of the failure, start your FoertiusAnt with -b only and check if it works. After a view seconds you should see a similar line like during calibration. Any other terminal output you have posted is perfectly normal.

MeanHat, 2021-01-12:
> @MeanHat Thanks for sharing this. I like the button idea. Looks very nice. Maybe I can find a Case with GPIO Connector that is big enough to fit this button. However I checked on the website and I [...]

Hi @wtal710174  - I like the RGB momentary pushbutton switch!
The script on the RPi3 is "FortiusANT Startup Script v0.2". I'm happy to try again. Working on an RPi Zero implementation currently.

MeanHat, 2021-01-17:
@wtal710174 I have used this mini on/off shim with Bluetooth FortiusAnt on an RPi 3. It works fine and was easy to install and set up. It's not as elegant as your solution with traffic lights:  https://shop.pimoroni.com/products/onoff-shim. It then works headless using a systemd service to start up FortiusAnt to run with Zwift.

[image]

trygvelu, 2021-01-20:
I did a zwift training session with the pi zero W, using an Ant dongle to communicate with my Genius trainer and BLE connection to the zwift computer. I had to download a node version for armv6 for the pi zero, as explained in the link in the Node JS section. I have a 7" LCD connected to the pi, but the wxPython gui does not work well for me. It freezes before I get to connect to the trainer. That got me thinking about alternative GUI options for embedded/headless solutions.
With node.js running (for BLE) it would maybe be feasible to implement a web based dashboard/frontend? There are many nice libraries available for building webpages with real-time data. 
Another option is tempting for me - build a pygame based dashboard, which can be run from the command line, and without running X11/wm in the background.

wtal710174, 2021-01-23:
Thanks for your comment @trygvelu. Yes the wxPyhton GUI is quite unstable on the raspberry. Not sure why this is happening. I have been testing with a tkinter interface to use a 7" touch display to control the FortiusAnt application. As of now I am facing similar issues. Additionally the node server started by the FortiusAnt programq called using this interface, does not run stable at all. Therefore pygame could be another option to get a stable interface for the raspberry.  In case you come up with something working, please share. It might still have the issue with the unstable node server as long as you do not use it as a replacement for the GUI class in the FortiusAnt program itself.

wtal710174, 2021-01-23:
### Node in virtual environment and updated startup script
I have been working on moving the node server into the virtual environment. Additionally I have moved the virtual environment into the FortiusAnt folder. This certainly does require an update of the script for starting FortiusAnt as well. In the following I have put a step by step instruction how to set this up.

#### Node in virtual environment
First you need to set up and activate a new virtual environment in the FortiusAnt folder
```
$ cd ~/FortiusANT/
$ virtualenv -p python3 --system-site-packages venv
$ source ~/FortiusANT/venv/bin/activate
```
Next you install wxPython (Download the [pre build package](https://drive.google.com/drive/folders/1owWkRTsvcnvq7ZSRc6zHS5dip2AMvW79?usp=sharing) and save to your Downloads directory in home)

```
$ pip install ~/Downloads/wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
```
Next install node to virtual environment
```
Download node 14.15.3
$ cd ~/Downloads/
$ wget https://nodejs.org/dist/v14.15.3/node-v14.15.3-linux-armv7l.tar.xz
# Untar downloaded archive
$ tar xf node-v14.15.3-linux-armv7l.tar.xz
# Go into folder with unzipped files
$ cd node-v14.15.3-linux-armv7l/
# Copy folder content to 
$ cp -R * ~/FortiusANT/venv/
```
and install the packages required for the FortiusANT node server
```
$ cd ~/FortiusANT/node/
$ npm install
```
Last step for node is to grant the node binary cap_net_raw privileges, so it can start/stop BLE advertising without root access.
```
$ sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)
```
Next Install the FortiusANT dependencies in the virtual environment
```
$ pip install -r ~/FortiusANT/pythoncode/requirements.txt
```
To start FortiusANT add the following bash script to your home

```
$ cd ~
# Deactivate virtual environment
$ deactivate
$ touch FortiusANT.sh
# Open the new file with any editor you like, I prefer pico or nano
$ pico FortiusANT.sh
# Insert the following text including the !/bin/bash comment

#!/bin/bash

#FortiusANT Startup Script v0.3 running node in virtual environment

#Checking if script is running already
FILENAME=`basename $0`
pids=$(pgrep -x $FILENAME)
for pid in $pids
do
   if [ $pid -ne $$ ]; then
      echo "$FILENAME is already running. Exiting"
      exit 0
   fi
done

#Checking command line options

if [[ -z "$@" ]]
then
[... 81 lines cut ...]
# -b for command line bluetooth startup, add any command line parameter as required, like -g to start with GUI
$ ~/FortiusANT.sh -b
```
Next, the calibration procedure will start with a pedal kick. After the procedure is complete, the bluetooth starts broadcasting identifying the Raspberry Pi as periphal device with the identifier 'FortiusANT Trainer'. This one you can connect to your CTP.

After the calibration you might see a couple of error messages. This happens while the http server is being established and the program is not able to access it. After 5-10 secondes the connection should be stable and you will see the normal output in the terminal like the one below.
```
20:08:31,698: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:32,700: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:33,702: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:34,703: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:35,705: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
20:08:36,706: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 15
```
Let me know if this works for you, could be that I forgot to add some steps. Difficult to write this without a clean system. 

You can easily start this script automatically by adding the following command line to the end of your .bashrc file
```
# edit with 
$ pico ~/.bashrc
# add the following line
/home/pi/FortiusANT.sh &
```
This will start the FortiusANT every time you start your Raspberry. Opening a terminal will not start it again. All you need to do is kick a pedal about 1 minute after you started the raspberry. 
I expect to post an update on the traffic light visual control soon, which will help to visualize the status of the FortiusAnt program.

wtal710174, 2021-01-23:
### Visual Status Indicator for FortiusANT
As stated before, I still can not get the script in the post [FortiusAnt One Button Controller for Raspberry](https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-757495938) to work reliably, since the node server always stops working after latest 10 minutes.
Therefore I had the idea to use the traffic light indicator at least to display the status of FortiusAnt, while not using it to launch FortiusAnt. Currently I am starting FortiusAnt using my FortiusANT.sh script for running the program with node in the virtual environment and I am launching the Status Indicator program in the same way. So I have 2 programs running independently. The Status indicator does not even run in a virtual environment. Both are being launched in the .bashrc file.

The Status indicator shows red if nothing is running, yellow (blinking), if FortiusAnt is running and node has not started the http server for broadcasting and green once everything is up and running. Additionally a long press (>3s) of the push button shuts the raspberry down. The light status changes, if anything stops working at some point.

Currently this seems to work fine. Below you find the source code attached.
 
[FortiusAntStatusIndicator.py.zip](https://github.com/WouterJD/FortiusANT/files/5860549/FortiusAntStatusIndicator.py.zip)

povey76, 2021-02-10:
I have tried a couple of times now with updated link (reformatting and starting again) and I am getting an error when it comes to: 
$ cp -R * ~/FortiusANT/venv/
The error reads cp: target 'home/pi/fortiusant/venv/' is not a directory? Every thing has worked up until this point.

I have looked in the files and the file structure has fortiusant and within that node and venv are present.                                                                                                                                

Any suggestions would be appreciated, took some time to get to this point - very new at this!

wtal710174, 2021-02-10:
@povey76: Difficult to diagnose this based on your description. The only thing that seems a possible cause is the fact that Linux is case sensitive. So if you have named your folder "FortiusANT" and you use "fortiusant" in your command it will not work, because this is using different names for the directory. Hope this helps, otherwise we have to try to troubleshoot some more in detail.

povey76, 2021-02-10:
> @povey76: Difficult to diagnose this based on your description. The only thing that seems a possible cause is the fact that Linux is case sensitive. So if you have named your folder "FortiusANT" and [...]

Thanks, obvious really!! Amended and all good to try, although:

18:04:26,084: Dongle - No expected replay from dongle, Bluetooth interface available?

Anything that I need to apply for the Garmin ANT usb to be recognised?

wtal710174, 2021-02-10:
@povey76: If you do not want to use the built in bluetooth while connecting the Tacx Head Unit using USB you need to refer to the Instruction that @martin-vi posted on his Git here: [Ant Linux Instruction](https://gist.github.com/martin-vi/5206b6b26faf6f2d6121f6c8f300e44d). There you will find the rights you need to set for the Ant Dongle. Additionally you need to make sure that you start the FortiusANT script without the -b option for bluetooth.

povey76, 2021-02-10:
> @povey76: If you do not want to use the built in bluetooth while connecting the Tacx Head Unit using USB you need to refer to the Instruction that @martin-vi posted on his Git here: [Ant Linux [...]

Ok, so I don't need the ANT dongle if using this method - even better! So as it is, I can connect the usb from the fortuis and when peddling/calibration I should see the response on the Screen (started with -g) and will transmit via bluetooth to Apple TV (zwift)? Appreciate the support- if this works later I'll be made up! Not doubting code, my newness to raspberry and code!!😂

wtal710174, 2021-02-10:
Yes, this is the way it should work. Start with -b -g, if you want the GUI Interface. -b is sufficient if you are OK with just the output in the terminal.
Good look and have a great first ride!

povey76, 2021-02-10:
> Yes, this is the way it should work. Start with -b -g, if you want the GUI Interface. -b is sufficient if you are OK with just the output in the terminal.
> 
> Good look and have a great first ride!

Thanks, when kids are in bed I'll be testing!! Is there an addition that you don't have to enter a command code? I have a spare screen so may mount that (looks good), but would be ideal to start peddling or hit a button to switch off and kill when done? I've seen some comments on this but still a novice to grasp where it needs to to copied and pasted!! Sorry, and really appreciate the help & support - what a group!!👍👍👊

wtal710174, 2021-02-10:
There is several ways, for the beginning I would use the screen and observe the output after starting the program manually in a terminal. This way you will understand if the Bluetooth is connected, and the node server is running and you will see all the possible failures that might happen. It is much simpler to troubleshoot by posting the output and errors you get. 
Regarding the start without interaction, there are several ways of making the unit headless. Two are described in this issue here.

[Traffic Light](https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-757495938) which I am using with the script in this [link](https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-766138122).

@MeanHat has a pretty nice and simple solution with his [on/off shim](https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-761776533).

The best solution for the future is probably going to be the [pull request](https://github.com/WouterJD/FortiusANT/pull/232) by @decodeais, which is for ANT systems, however can be used for the bluetooth solution as well, with some minor modifications.

So lots of possibilities and things to learn.

povey76, 2021-02-10:
> There is several ways, for the beginning I would use the screen and observe the output after starting the program manually in a terminal. This way you will understand if the Bluetooth is connected, [...]

Good new's is that she is alive and communicating with zwift! Although I have some errors as picture? What's required?[image]

wtal710174, 2021-02-11:
> Good new's is that she is alive and communicating with zwift! Although I have some errors as picture? What's required?[image]

Not really sure, what this can be. I guess the USB cable is connected correctly. Maybe try another cable. If this does not work, check if you have given the User the right to access the USB port. Not sure if this can be the issue since FortiusAnt is finding the hardware in the first lines. 

Let me know how this works. If it does not the next step would be to run FortiusAnt in debug mode and post the logfile.

switchabl, 2021-02-11:
@wtal710174 This seems to be the same problem as #251 and #253, so it probably isn't something that is wrong at your end.

povey76, 2021-02-11:
> @wtal710174 This seems to be the same problem as #251 and #253, so it probably isn't something that is wrong at your end.

Thanks for reply, seems to communicate ok - so not concerned - just wanted to check it wasn't an installation issue!

WouterJD, 2021-02-25:
@wtal710174 great instruction-set; RPI4 is working in console-mode.
Next step is BLE and then wxPython.

Please explain, what is the reason to use the virtual environment.

wtal710174, 2021-02-26:
@WouterJD The reason for the virtual environment is that python can not differentiate different package version if you are running more projects on your Raspi. In the virtual environment it will have the package versions it requires for the project. It is more or less just good practice in python to handle this issue. If your project is going to be the only python project on the raspi you can skip the virtual environment and install and run directly. A good excessive explanation you can find here: [Python Virtual Environments: A Primer](https://realpython.com/python-virtual-environments-a-primer/)

WouterJD, 2021-02-26:
Many thanks, I expected this but wondered if it had to do with BLE. Thanks to your support my RPI is now working well. Next are: BLE and wxPython. Moving forward😉

WouterJD, 2021-02-26:
> * Non Gui version:  Yes this is a great idea.

There is a UseGui constant in constants.py, which was not tested (by me) untill now.
In FortiusANT 5.3 this variable will be set to False when wx is not available, and this requirement is met then.
-g is then not available either.

WouterJD, 2021-02-26:
> #echo Getting FortiusANT updates from Github Repository
> #git -C /home/pi/FortiusANT pull upstream master

Automatc updating is NOT implemented by design; at the moment of implementation this check it was requested to inform, not to update.
I agreed with that requirement: it avoids that a working system is "improved" with new errors.

wtal710174, 2021-02-26:
@WouterJD for wxpython just install the pre build package. It will work on the RPI4. 
BTW: I am having some issues with the headless version and the calibration. If I fire up the RPI and immediatly calibrate as soon as I can, the calibration fails (Runs until timeout). If I wait some time and calibrate then the calibration works. Not sure why, would be intersting if I am the only one that is experiencing it.
Nice idea checking the presence of the wx package and disabling gui automatically. Maybe still accept the -g option, but starting without gui with a message in the console, instead of stopping the execution, because of an invalid option. Hope I understood this correctly, did not check / notice the version 5.3 code yet.

WouterJD, 2021-02-26:
> BTW: I am having some issues with the headless version and the calibration. If I fire up the RPI and immediatly calibrate as soon as I can, the calibration fails (Runs until timeout).

Perhaps RPI not fully booted yet. Add sleep in script and see what happens.
Timing will not be revealed by logfile

WouterJD, 2021-02-26:
> Maybe still accept the -g option

It's how it is. If you don't want gui, don't specify -g flag. If you do: error message. Easy😉

[40 more replies]

closed 2021-03-10 by WouterJD