WouterJD/FortiusANT #193: Tacx Fortius 110V wheel speed limitations

TriScott, 2021-01-04:
The original 110V (North American) version of the Tacx Fortius had a limitation whereby the resistance dropped out dramatically when the wheel speed exceeded ~26mph / 42km/h.  Tacx subsequently released an updated version of the 110V Fortius hardware that somewhat addressed this issue.  However, with the second version of hardware, the resistance (and thus power) still seems to ramp down as the wheel speed exceeds 42km/h.

Fast forward to today and the 110V Tacx Fortius limitations are now affecting Tacx Fortius + FortiusANT + Zwift.  Essentially any flat road sprint segment I contest in Zwift will see the wheel speed get above 42km/h and the actual resistance start to ramp down.  I haven't measured the impact, but my guess is that it is providing an unfair advantage as the reported power continues to ramp up while the actual effort plateaus down.  The same occurs when trying to accelerate during a prolonged downhill.

The Tacx Fortius resistance also becomes very choppy at wheel speeds less than ~15km/h when simulating steep climbs, so the effective useful range of a 110V Tacx Fortius is between 15km/h and 42km/h at the wheel.  Note, this is not the Zwift reported speed, but the Tacx reported speed as displayed by FortiusANT.  The Zwift range is a lot broader so you can do a lot of quality riding without ever experiencing this.

I am wondering if any other FortiusANT users with 110V Tacx Fortius trainers are experiencing similar issues?  If so, I am curious what some of the Tacx experts (@WouterJD @totalreverse) think about the potential of dynamically scaling the resistance to encourage the wheel speed to stay within the sweet spot range.  This could be an additional FortiusANT command line option for 110V Tacx Fortius users.

The goal would be to keep the reported power accurate by encouraging the wheel speed to stay within the 15km/h-42km/h range through automatically scaling the resistance by a correction factor proportional to the proximity to the limit, resulting in a value different from the FE-C requested grade / incline by the application.  For example, as wheel speed exceeds 40km/h, ramp up the resistance, and as wheel speed decreases below 17km/h ramp down the resistance.  When the wheel speed returns to the supported range, reduce and then remove the correction factor.  In all cases, the reported power being generated by the rider would stay true.

Thoughts on this approach?  Thanks!

mattipee, 2021-01-07:
Hi, @TriScott 

I've been working on getting Fortius working in GoldenCheetah (see https://github.com/GoldenCheetah/GoldenCheetah/pull/3725).

Also some discussion here (https://github.com/WouterJD/FortiusANT/issues/171).

I don't know if anything behaviours are 110V specific, but there are perhaps T1932/T1942 differences, and as the T1942 is able to take a firmware update, differences between T1942 units.

There is a "known" issue at low wheel speeds. @WouterJD has a routine in FortiusANT to "encourage higher wheel speed", but limiting the resistance applied at low wheel speeds. I did some data capture and came up with a different scheme which is currently in FortiusANT, but under discussion. Graphs from TACX for other models suggest power limit is linear with wheel speed (suggesting a constant maximum force... Power (W) = Speed (m/s) * Force (N).)

You can see a comparison here:

[image]

@ericchristoffersen found this document with graphs... https://www.qbp.com/diagrams/TechInfo/Tacx/MA9570.pdf

Indeed this suggests that the device has an upper limit... beyond "42kmh", it can no longer apply "100N" of force (or whatever the value is for the Fortius - Genius appears to be ~118N). I can't push 1000W+ for long, so I can't really test this. :)

But in terms of ride feel, I can see that wheel speeds in excess of the upper limit will mean reduced resistance available, but still a constant upper power limit.

[image]

The values reported by the trainer for current force and wheel speed do appear to be good or are assumed to be so, so I'd be surprised if the reported power value (obtained simply by multiplying those two values, once scaled to standard units) would over-report.

So there are two main issues in my head at the moment:
- not all headunits may behave the same
- there is an inherent force limit (which may be non-constant, and a function of speed)

Add to that, "virtual gearing" whereby you can get "additional low gears" or "additional high gears" by scaling the resistance... the former allows you to climb gradients beyond the device's limits by artificially increasing wheel speed to get it up into a more capable zone, and the latter allows you to push against more resistance without spinning out. A tangent, but relates to "scaling resistance".

Anyway, a jumble of different thoughts for you. Hope some of it makes sense, I'm still feeling my way through it.

@switchabl @michelrdagenais fyi.

totalreverse, 2021-01-07:
Very interesting discussions here and in the GC respository group.
I cannot add valuable information about virtual gearing, but I can say, that there is no "magic" inside the solid blue T1942 headunits.
All firmwares I know, just forward the commands from the PC software to the powerback and return the answers from powerback unit in bytes 25..47 (skipping the last checksum byte) . 
Any controlling magic must be inside the powerback firmware or in the PC software.

michelrdagenais, 2021-01-08:
If I remember correctly, when I got my T1942 Fortius, we had to do an upgrade of the Brake unit firmware to solve "stuttering" issues. Now that you mention it, it was very difficult going downhill (i.e. over ~40kph) to put out any significant power with Tacx software. I have a 110v unit.

mattipee, 2021-01-08:
Couple of related comments here... https://www.cyclingforums.com/threads/tacx-fortius.371170/

WouterJD, 2021-01-08:
@Triscott do you use the python- or windows.exe version of FortiusAnt?
In case of python, I can give hints how to experiment.

TriScott, 2021-01-11:
Thanks to everyone for pointing me to a lot of the excellent work people have done to reverse engineer this trainer.  It has taken me a lot of time to go through it all. @mattipee I used to use GoldenCheetah frequently for data analysis.  I did try Fortius control with it some years ago but did not have a good experience.  It sounds like you and @michelrdagenais have made tremendous progress!

I have two observations so far regarding my original issue:

1) The V2 of the 110V Fortius hardware does not have the 26mph / 42km/h limitation and I was mistaken to suggest that it does.  The easiest way to illustrate that is to use the trainer in erg mode with a fixed power (e.g. 300 watts) and you can easily get up to 55km/h wheel speed or more with it still feeling perfectly smooth.  V1 of the 110V hardware definitely has that issue and my original proposal would still be relevant to those devices.

2) I have more or less pinpointed the reason for choppiness during sprints and have a patch that I need to test.  I am optimistic this can be fixed with sw alone.  I will update again with my findings soon.

@WouterJD I am using the Windows.exe version of FortiusANT as I have never installed python under Windows before.  I will need to do that next.

WouterJD, 2021-01-11:
@TriScott python is not needed, but may come in handy to make patches and/or do some quick tests

BrRoBo, 2021-01-16:
My Fortius usage feedback just for a datapoint...

I have the solid blue Fortius head unit T1942.  

The motorbrake is reported by FortiusAnt as a T1978, year=2006, version2=0, firmware=0.  I do seem to recall applying a firmware upgrade to this unit, but since it was so long ago I can't be 100% certain.

Regardless, this Fortius works perfectly with the 2x Ant+ dongles along with either Zwift or the current TDA for over 600 miles (1000km) to date.  I have climbed many steep grades > 15% at a relatively slow wheel speed and cadence and had no stutter/jerkiness issues.  The only time we experience any stutter/jerkiness is if the speed is lower than ~5mph or if the cadence sensor is not properly installed.

My further testing will be vicariously through new owners of this Fortius (or i-Magic) as my wife & I anted up for a new direct drive unit... instead of traveling this winter we stayed home because of COVID.  I am selling both Fortius and i-Magic for cheap just to free up some space in our small condo/garage.

WouterJD, 2021-01-16:
You make me cry: sell this magnificent setup😪

michelrdagenais, 2021-01-16:
Your T1978 is surprising, this is not documented as an expected answer. The version2=0 and firmware=0 are similarly unexpected. I checked with my Fortius and indeed, I sometimes get strange results such as (for the same unit):

Motor Brake Unit Firmware=4100 Serial=828 year=2007 type=T1946 Version2=2078
Motor Brake Unit Firmware=73569 Serial=7712 year=2059 type=T1978 Version2=8014
Motor Brake Unit Firmware=35153 Serial=24800 year=2033 type=T1977 Version2=9255
Motor Brake Unit Firmware=8273 Serial=51936 year=2017 type=T1977 Version2=3570
Motor Brake Unit Firmware=67791 Serial=54784 year=2096 type=T1976 Version2=12278

Note that I am not using FortiusAnt but similar code that I added to GoldenCheetah to get the version number and model. However, it appears that there is a problem in both code. I will experiment further to try to understand the problem.

> The 110v motorbrake is reported by FortiusAnt as a T1978, year=2006, version2=0, firmware=0. I do seem to recall applying a firmware upgrade to this unit, but since it was so long ago I can't be 100% certain.

totalreverse, 2021-01-16:
Probably, there is still a problem with out-of-sync decoding of "resistance command" answers as "version command" answers. 
https://github.com/totalreverse/ttyT1941/issues/20#issuecomment-757467896

BrRoBo, 2021-01-20:
> 
> 
> You make me cry: sell this magnificent setup😪

The Fortius has found a very deserving home with a healthcare worker in my area.  Still to find a home for the i-Magic.

WouterJD, 2021-01-20:
> Note that I am not using FortiusAnt but similar code that I added to GoldenCheetah to get the version number and model. However, it appears that there is a problem in both code. I will experiment further to try to understand the problem.

Please test with FDortiusAnt 5.1 Test IV or higher

closed 2021-01-24 by WouterJD