WouterJD/FortiusANT #194: Flow Excel i-Magic T1901 Brake & T1682 Head unit

rockfreak, 2021-01-04:
Hi all,

Is it possible to get the following setup working?

Brake Unit: T1901 (Flow Excel i-Magic) 
Head Unit: T1682 

The head unit seems to be connected with an ethernet style cable not usb.

Has anyone got experience with this?

I don't have hands on as this is my mums old trainer and she'd like to try Zwift without the mass outlay of a smart trainer.

Happy to provide more info and help out to get it working if you think it's doable.

cyclingflow, 2021-01-05:
No. The cable connects to the brake, not to the PC (not usb as you noted).
The T1682 is a manual unit. 
Will only work if you have a separate power meter on the bike, and manually adjust 
the resistance of the brake using the T1682. This also requires an ANT+ usb dongle,
for receiving the power meter.

Cheapest solution is to get a used t1932 unit (with or without brake/trainer),
or any head unit that has the USB connection.
Many sell their units because their TTS software is no longer installable.
This requires two ANT+ dongles and FortiusAnt.

See manual for more detailed information.

rockfreak, 2021-01-05:
Hmm I wonder if we can replicate the signals sent by the T1682 head unit.

As far as I can work out from the manual the T1901 will work it's the head unit that is the issue?

switchabl, 2021-01-05:
@rockfreak In principle, yes. I have done that before for testing. You can find the pin-out for the T1901 at https://github.com/totalreverse/ttyT1941/wiki ("Inside the T1902 and T1942"). You would need to:
- count the pulses on PIN1 for cadence (one pulse per pedal revolution)
- count the pulses on PIN6 for wheel speed (four pulses per roller revolution)
- detect the zero crossings on PIN 4 (+- 20V sine)
- output a PWM signal synchronized with the zero crossings on PIN3 to control the brake resistance

Most USB-capable microcontrollers would probably be able to do the job along with a little bit of analog circuitry. Emulating the old T1902 protocol would be the easiest choice.

Unless you enjoy the DIY aspect, still easier to buy a suitable head unit (T1902, T1904 or T1932) of course. The brake should work fine.

cyclingflow, 2021-01-06:
DIY project
https://hackaday.io/project/164276-tacx-flow-ant-conversion

closed 2021-02-04 by WouterJD