WouterJD/FortiusANT #195: Manual adjustment of the requested power on the headunit

TacxBiker, 2021-01-05:
In the manual of FortiusANT:
4.1.5 Buttons on the Tacx head unit
There are four buttons: Cancel, Enter, Up, Down.
• Cancel is active in all modes.
• If not in an active mode, Up/Down navigate through
the menu, Enter activates the selected button and
Cancel exits FortiusAnt.
• In runoff or manual mode, Up/Down modify the
required power with ±50Watt. OK *) resets the
power to the initial value of 100Watt.
• In manual grade mode, Up/Down modify the slope
with ±1degree. OK *) resets the slope to the initial
value of 0 degrees.
• In normal resistance mode, Up/Down modify the
resistance of the Fortius by ±10%. OK *) resets the
resistance is reset to the initial value of 100%.

Does this mean when in ERG mode de buttons don't work?
As for now I can't manualy adjust the requested power in ERG mode

Is this as it should work or am I doing something wrong

WouterJD, 2021-01-08:
If you are connected to a CTP (Zwift) then that program tells what power to ride. The idea is that -if you are doing an ERGmode training- it does not make sense to adjust the power.

If you are running in manual power mode (-m flag on command line) you can set the power.

TacxBiker, 2021-01-08:
Hi, your right. of course it should be like this. But when riding a hevy training it can occour on some occasioins the the asked power is to high (not frequently of course..). 
But when you're not able to deliver the power it not possible to peddal. But the programs goes on pauze (one is not peddaling) so the problem stays.
What i was wondering, should it be possible to lower the requested power manualy, until the new power value is received from the CTP (The Sufferfest in my case)

WouterJD, 2021-01-11:
Version 5.1 Test III is available with this issue implemented.
(for details see [wiki](https://github.com/WouterJD/FortiusANT/wiki#version-51-test-iii-available-for-evaluation-jan-11-2021))

Please test

TacxBiker, 2021-01-11:
That works perfect! 
I didn't have a gui at the time I tested. Is this acomplished by virtual gears isn't it? 
no matter what received value from CPT, the power to produced is lower or higher. When a new value is received, the offset is still used.
So nice add on, thanks.

WouterJD, 2021-01-12:
Thanks for confirmation

closed 2021-01-12 by WouterJD