WouterJD/FortiusANT #197: Stuttering resistance on pedals during ride on Fulgaz

ManInAGarden, 2021-01-08:
I have a strange problem with Fulgaz and FortiusAnt. In any paring dialog I can paddle smoothly and Fulgaz shows reasonable numbers for power and cadence. As soon as start a ride resistance on the pedals starts to feel very bumpy and jerky.

My configuration: I run FortiusAnt on a raspberry pi 4B with an update Buster operating system. On startup via a terminal and ssh Fortiusant.py begins with the follwing output:
python FortiusAnt.py -a
18:41:34,949: Dongle - Using Dynastream Innovations dongle
18:41:34,965: Connected to Tacx Trainer T1942
18:41:35,330: Motor Brake Unit Firmware=54979 Serial=88288 year=2033 type=T19157 Version2=2304 MotorBrake=True
18:41:35,531: Tacx   - Connected to Tacx Trainer T1942
18:41:35,531: AntHRM - Heartrate expected from Tacx Trainer
18:41:36,619: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
18:41:36,747: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
18:41:36,875: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
18:41:37,004: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
18:41:37,132: FortiusANT receives commands from an ANT+ Generic Remote Control
18:41:37,311: Ctrl-C to exit
18:41:37,311: Tacx   - * * * * G I V E   A   P E D A L   K I C K   T O   S T A R T   C A L I B R A T I O N * * * *

I can than start calibration via a pedal kick.

Pairing with Fulgaz seems to work.

Is there anything I can try now?

WouterJD, 2021-01-09:
run with -m (manual power) and create export tcx or json and analyse what happens; the json file shows the data received/sent
use the sample excel  in supportfiles to interpret the json

ManInAGarden, 2021-01-09:
@WouterJD Thank you. Meanwhile I also tested it with Zwift. It works flawlessly. So that problem seems to be with Fulgaz or somewhere around communication with Fulgaz. Does that test with -m still make sense in that case? Or should I provide some other test.

mattipee, 2021-01-09:
@ManInAGarden hi - I've been looking into stuttering a little too much recently. Just a couple of thoughts if I may...

- is your weight correctly set in both Zwift and Fulgaz?
- what happens if you increase Zwift "trainer difficulty" slider in the settings up to max?

Also @WouterJD ...

> Motor Brake Unit Firmware=54979 Serial=88288 year=**2033** type=T19157 Version2=2304 MotorBrake=True

@ManInAGarden - what's the serial number on the underside of your motor power unit?

ManInAGarden, 2021-01-09:
The serial number is: 41061903

ManInAGarden, 2021-01-09:
I did another test with Fulgaz. First I should mentian that before I tried Zwift I readjusted the screw that  presses the motor unit against the wheel so that it as tighter. So that was in effact with Zwift also.
Then I worked myself through the parts of the manual that I just had quickly passed by during my first read. It's the part with the formulas,,,, There I fund that with Rouvy one should use the parameter -G50. I did that and now I only had a little stuttering at the beginning of the ride when speed was still low. (I did not have this with Zwift). So maybe that -G50 is also a good starting point for Fulgas and maybe I even have to reduce this down to -G30.

ManInAGarden, 2021-01-09:
@mattipee I double checked my weight in both applications. It is/was OK in both Zwift and Fulgaz.

WouterJD, 2021-01-09:
@mattipee you underline next line; the info is informatory, the essential is that MotorBrake=True; that is because this is a T1942 head unit, which supports motor brake only

> Motor Brake Unit Firmware=54979 Serial=88288 year=**2033** type=T19157 Version2=2304 MotorBrake=True

Type T19157 was not expected, though.
Look also at https://github.com/totalreverse/ttyT1941/wiki#tacx-product-number-overview

Would T1932 also support this motorbrake?

WouterJD, 2021-01-09:
@ManInAGarden -G50 means that the grade (as transmitted by Rouvy) is multiplied by 50%
This is done by Zwift by default (can be changed in settings) where Rouvy evidently has decided to transmit the real grade and leaves it to the user to change gears (as in reality).

See https://github.com/WouterJD/FortiusANT/issues/120#issuecomment-756836672 for upcoming feature :-)

WouterJD, 2021-01-09:
@mattipee for T1932, the default setting is Magnetic Brake
If the motor-brake-message does not return data AND -t Motorbrake is specified then type = Motor Brake
If the motor-brake-message returns type = 41 or 46 then type = Motor Brake

This has grown like this, because I assumed that T1932 would consistently return type. Question here is what default to take.

Ref [motor brake version message](https://github.com/totalreverse/ttyT1941/wiki#t1941-motor-brake-version-message)

mattipee, 2021-01-09:
> @mattipee you underline next line; the info is informatory, the essential is that MotorBrake=True; that is because this is a T1942 head unit, which supports motor brake only
>
>
> Type T19157 was not [...]

I was just drawing attention to the year "2033"...

WouterJD, 2021-01-10:
@mattipee Interesting!

tt-YY-##### (tt=41 (T1941), YY=year, ##### brake individual serial)

So I have to convert Serial to string and then interpret left-to-right; since ##### may be #### ### ## # ??

For a serial-number 1903 I expected 410601903 :-)
https://github.com/WouterJD/FortiusANT/blob/5.1%2B4.2-Quality-upgrade/pythoncode/usbTrainer.py#L2948

WouterJD, 2021-01-18:
Please test with version 5.1 Test V