WouterJD/FortiusANT #199: Success on Mac, Though Issue with Rolldown

ted-westwood, 2021-01-09:
Hey Wouter and All,

I've finally had success on Mac, I just ran VMWare Fusion instead to Bootcamp Windows and managed to get it all working! 

On initial calibration, I am attempting to get the Rolldown time right, however when I stop pedalling the brake starts to slow down and then continues spinning on its own accord. Have we got any ideas as to what the issue is here? 

Cheers,
Ted

ted-westwood, 2021-01-09:
It also appears that cadence (when it settles) is reading about double what it should be.

WouterJD, 2021-01-09:
> Have we got any ideas as to what the issue is here?

Yes, this is as it should be.
Do you have suggestions to improve below text?

See manual 4.1.3 Start

_FortiusAnt starts to calibrate the trainer (if supported and -n is not specified). Calibration means that the brake rotates the wheel at 20 km/hr and returns the resistance found. As soon as the resistance is constant, the calibration stops. The calibration time is at least 30 seconds (warming up the tire) and stops when the resistance value is constant.
Note that the calibration starts when you turn the pedal as if starting to cycle, which is the only physical action to take. Note that, starting the motor automatically would be a risk for physical injury, therefore the confirmation with a pedal-kick is required.
**When calibration is started, do not pedal, the process completes automatically.**_

ted-westwood, 2021-01-09:
Yeah, I had this issue pre "start"ing calibration, during the rolldown phase and it was rolling by itself. That was down to my sloppiness so I apologise! I've managed to get through this now however after running "Start" to calibrate, I am now getting:

```
22:39:37,778: Tacx returns insufficient data, len=24
22:39:37,779: To resolve, check all cabling for loose contacts.
```

Initially I got the "Tacx returns insufficient data, len=XX To resolve, try to run without Pedal Stroke Analysis" dialogue, and after removing -A I ended up with the loose contacts message. Which I would presume is cadence, however it registers fine and accurately in the rolldown.

I'm very happy to revise any wording and to offer steps for my process in VMWare for the record when I succeed! Hahaha.

WouterJD, 2021-01-10:
HI @ted-westwood The short message error is not 100% clear when it is given.
The head unit does not manage to respond and returns an incomplete message.

It can be caused when the head-unit is polled too fast.
It is also possible that there are loose contacts, including the power cable. **Perhaps** because -when loose- it causes interference.

Please check.
If persistent, create a logfile (with -d127 on the command line) and share.
Also - use the newest version please; which is in testmode - checkout from https://github.com/WouterJD/FortiusANT/tree/5.1+4.2-Quality-upgrade

ted-westwood, 2021-01-10:
Hey @WouterJD, I've run two instances of this incase it helps - one with Pedal Stroke Analysis, another without. Attached here:
https://www.dropbox.com/sh/xyu17wy19qbjtg5/AAA6tJXEukmIgfWI-mopBJ31a?dl=0

I ran this through the newest version - the test mode is very handy, thank you!

WouterJD, 2021-01-12:
I have looked at your logfile and have the following observations:
- The short-message warning is given only once per session, this can be ignored.
- I see some "hickups" as if FortiusAnt is blocked  for some time (see below) which may be related to your VMware installation; VMware may be less ideal for a real-time application like FortiusAnt and/or Zwift
- Also the ANTdongle produces many messages (third copy below); I think I usually do not see that

But bottom-line, the logfile does not show big surprises.
Please describe what issues you experience in the sessions, you have attached the lgfile for.

```
12:50:02,938: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
12:50:04,980: AntDongle.Read() returns: ...
```

```
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
12:50:07,017: AntDongle.Read() returns: ...
```

```
12:50:04,980: devAntDongle.read(0x81,1000,20) returns ...
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe4, info="01 01 03" (ch=1, msg=0x1)
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
12:50:04,980: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)

```

WouterJD, 2021-02-01:
Suggestions are added to manual 5.2

closed 2021-02-01 by WouterJD