WouterJD/FortiusANT #200: T1932 + Magnetic brake - Question to all

WouterJD, 2021-01-10:
The T1932 headunit supports the motorbrake and the magnetic brake.

From version 4.0 onwards, I try to automatically detect what type of brake is connected - but that causes many issues (see #173).

After startup, the following messages are displayed:
```
19:12:01,123: Connected to Tacx Trainer T1932
19:12:01,456: Motor Brake Unit Firmware=**** Serial= **** year=**** type=T19** Version2=**** MotorBrake=*****
```
I would appreciate that everybody who reads this, copies the given messages into this issue + the version of FortiusAnt used; please also respond if NO message is given.
- - - 

Currently for T1932:
- FortiusAnt behaviour is for a Magnetic brake by default
- If type=T1941 or T1946 OR -t Motorbrake is specified (from version 5.1 onwards [see branch](https://github.com/WouterJD/FortiusANT/tree/5.1+4.2-Quality-upgrade)) behaviour is for a Motor brake 

Current issues:
- The motorbrake message is not always succesfully returned
- There appear to be more motor brake types (T1949 is reported by @DoubleH63)

Possible alternative is to invert the behaviour rules as defined above
To make a decision, your feedback is very helpful

WouterJD, 2021-01-11:
@bikebeppe63 thanks!

[image]

Tacx Trainer T1932
Motor Brake Unit Firmware = 6402 Serial = 0  year=2000 type=T190 Version2=0 MotorBrake=False

WouterJD, 2021-01-11:
Version 5.1 Test III is available with this issue implemented.
(for details see [wiki](https://github.com/WouterJD/FortiusANT/wiki#version-51-test-iii-available-for-evaluation-jan-11-2021))

Please test

TriScott, 2021-01-13:
Using 5.1 Test III, I get

`Motor Brake Unit Firmware=4103 Serial=11734 year=2038 type=T1978 Version2=0 MotorBrake=True`

switchabl, 2021-01-20:
With 5.1 Test VI:
- T1901 Flow -> T1932
```
19:58:25,355: Connected to Tacx Trainer T1932
19:58:25,657: Motor Brake Unit Firmware=0x1902 Serial=0 MotorBrake=False
```
- T1941 Fortius -> T1932 (first time after plugging it in)
```
20:13:59,297: Connected to Tacx Trainer T1932
20:13:59,797: Tacx head unit returns incorrect header 0x21303 (expected: 0xc03)
20:13:59,997: Motor Brake Unit Firmware=0x1004 Serial= 2784 year=2008 type=T1941 Version2=526 MotorBrake=True
```
[FortiusAnt.2021-01-20 20-13-58.log](https://github.com/WouterJD/FortiusANT/files/5844745/FortiusAnt.2021-01-20.20-13-58.log)
- T1941 Fortius -> T1932 (second time and after)
```
20:14:16,469: Connected to Tacx Trainer T1932
20:14:16,769: Motor Brake Unit Firmware=0x1004 Serial= 2784 year=2008 type=T1941 Version2=526 MotorBrake=True
```
[FortiusAnt.2021-01-20 20-14-15.log](https://github.com/WouterJD/FortiusANT/files/5844747/FortiusAnt.2021-01-20.20-14-15.log)

BikerVibor, 2021-02-09:
Hai Wouter

Wow, many hours of work in this great program, a lot of respect for this. I hope I can get my Fortius up and running again through FortiusAnt. Until now I just have to find my way. ... and wait for my Ant dongles to arrive.

Is it possible to test the Fortius with Fortius Ant without an Ant Dongle?

The start-up data:
22:51:27,701: No Tacx trainer found
22:51:51,404: Connected to Tacx Trainer T1932
22:51:51,953: Tacx head unit returns incorrect header 0x21303 (expected: 0xc03)
22:51:52,170: Motor Brake Unit Firmware=0x1007 Serial= 8865 year=2010 type=T1941 Version2=790 MotorBrake=True
22:51:52,278: FortiusAnt applies the MotorBrake power curve

[image]

switchabl, 2021-02-09:
@BikerVibor That already looks good. You can test in manual mode: click on the settings button (the gear) and choose either manual power or manual grade. Use the up/down buttons on the head unit to change the target power/grade.

WouterJD, 2021-02-17:
Hi @BikerVibor 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

-m and -M can be used for ANTless testing indeed.
I assume issue can be closed.

closed 2021-02-17 by WouterJD