WouterJD/FortiusANT #201: Improve Motor Brake Version Message

WouterJD, 2021-01-10:
@mattipee Interesting!

tt-YY-##### (tt=41 (T1941), YY=year, ##### brake individual serial)

So I have to convert Serial to string and then interpret left-to-right; since ##### may be #### ### ## # ??

For a serial-number 1903 I expected 410601903 :-)
https://github.com/WouterJD/FortiusANT/blob/5.1%2B4.2-Quality-upgrade/pythoncode/usbTrainer.py#L2948

_Originally posted by @WouterJD in https://github.com/WouterJD/FortiusANT/issues/197#issuecomment-757451056_

mattipee, 2021-01-10:
Easiest first step would be to include the raw value in the printout, alongside the "parsed" fields.

WouterJD, 2021-01-10:
I suggest following change
https://github.com/WouterJD/FortiusANT/blob/5.1%2B4.2-Quality-upgrade/pythoncode/usbTrainer.py#L2950

raw data is in the log, this print should be informative for the user.

mattipee, 2021-01-10:
Much easier to read!

WouterJD, 2021-01-11:
Version 5.1 Test III is available with major update on Motor/Magnetic brake determination.
(for details see [wiki](https://github.com/WouterJD/FortiusANT/wiki#version-51-test-iii-available-for-evaluation-jan-11-2021))

Please test

TacxBiker, 2021-01-11:
seems to work just fine

WouterJD, 2021-01-13:
see https://github.com/totalreverse/ttyT1941/issues/20#issuecomment-759674338

WouterJD, 2021-01-18:
Implemented in 5.1 Test V

closed 2021-01-18 by WouterJD