WouterJD/FortiusANT #202: trying to get it to work

laurenshaijma, 2021-01-10:
Hi All,
I followed the manual and speeded up to 40km/hr then let it roll out
calibration will stop it self but after that it keeps saiing don't pedal
so is it finished then?
[image]

and if all is finished how do I connect zwift to it?
I put on heart monitor zwift imidiatly found it but nothing else
can anyone help me?

WouterJD, 2021-01-11:
@laurenshaijma 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Locate HW is disabled because it's already done and trainer and ANT-dongle found
How did you start FortiusANT (what parameters)
It looks like you did a runoff; that's OK.

Now press Start to activate the program.

laurenshaijma, 2021-01-11:
Hey Wouter,
I'm The guy harrasing you on messengerLaurens Haijma from Stiens just above Leeuwarden
dont know what you mean  by what bundelI bought a tacx fortius T1941 (it sais on motor/brake)T1941.50 (it sais on the box attached to the motor/brake)and the head unit sais T1932

I started FortiusANT.exe the windows executable just double click the .exeIm running windows 10
then do the runoff and after that kick pedal to calibratein the screen it sais CYCPLUS ( you told me could give problems but someone on GIT said this one would work I even checked ID nr. and it was the same as that of him (revering to "totalreverse" on your git)
when I turned heartrate on zwift found itbut what do I connect to zwift there are speed, cadense, and I think it was 2 other options which do I need to connect?
oke I hope you have enough info with thisI'll try again tonight see if I can connect....
regards,Laurens Haijma

WouterJD, 2021-01-12:
Voorlopig doet de ANT-dongle het.

Je kunt Zwift aansluiten de het FE-C device 57591:

FortiusAnt broadcasts the following devices
57591 FE-C Controllable Fitness equipment
57592 HRM Heartrate monitor (optional, according -H parameter)
57595 SCS Speed/cadense sensor
57596 PWR Power monitor

In fact, only the FE-C (57591) is required. SCS and PWR are implemented so you can attach a Garmin Edge on your bike, it should not be required by Zwift. If you use an ANT+ heart rate monitor, Zwift will connect to that and the FortiusAnt HRM is not required.

laurenshaijma, 2021-01-12:
when I press start it keeps saying calibrating do not pedal! but you say press start and connect to zwift
I don't get it
[image]

after that start zwift and get into
[image]
this screen 
I kind of assumed that I have to press
[image]

this button but it wont find anything

you say:"Je kunt Zwift aansluiten de het FE-C device 57591:" but I don't know what it means
can anyone help?

laurenshaijma, 2021-01-12:
[FortiusANT.2021-01-12 21-27-26 - Copy.log](https://github.com/WouterJD/FortiusANT/files/5804803/FortiusANT.2021-01-12.21-27-26.-.Copy.log)

laurenshaijma, 2021-01-12:
[FortiusANT.2021-01-12 22-11-50.zip](https://github.com/WouterJD/FortiusANT/files/5804838/FortiusANT.2021-01-12.22-11-50.zip)

WouterJD, 2021-01-14:
> [FortiusANT.2021-01-12 21-27-26 - Copy.log](https://github.com/WouterJD/FortiusANT/files/5804803/FortiusANT.2021-01-12.21-27-26.-.Copy.log)

It looks like proceeding normally:
- YOu give a pedal kick
- The trainer starts to run and returns a low resistance (Speed=20.0 TargetRes=2227 CurrentRes=-26)
- When the resistance is stable, the trainer is stopped at 21:28:31,811

At the first command sent to the ANT-dongle, an error occurs
21:28:31,811: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: The device does not recognize the command.\r\n'
21:28:31,814: devAntDongle.read exception: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: The device does not recognize the command.\r\n'
21:28:31,814: ANT Dongle not available; try to reconnect after 1 second

21:28:33,366: GetDongle - Check for an ANT+ reply
21:28:33,367: GetDongle() returns: Using CYCPLUS dongle
21:28:33,367: ANT Dongle reconnected, application restarts

Because of the restart, the calibration is done again (although that is not necessary)
|* * * * G I V E   A   P E D A L   K I C K   T O   S T A R T   C A L I B R A T I O N * * * *
The Speed=20.0 TargetRes=2219 CurrentRes=-10 is received
|* * * * C A L I B R A T I N G   (Do not pedal) * * * *
For some reason Speed=0.0 TargetRes=2228 CurrentRes=0 is given as if the calibration stopped

Could you please describe what exactly happens during this test-cycle?
- What do you do
- What does the trainer do

laurenshaijma, 2021-01-14:
My bike is on the trainer next to me, I'm behind the laptop.
when asked give pedal a kick I give it a kick, the wheel starts rotating 
then the trainer takes over the wheel runs for it self for a while while I wait 
after a while it stops rotating the wheel slows down ant stops
text of calibrating changes from calibrating to connected to for a second and then changes to calibrating  dont pedal again

WouterJD, 2021-01-14:
That is what I saw in the logfile.
Try to run without calibration (specify -n option on command line) and see what happens.

You are investigating two issues:
- the ANTdongle giving issues
- the second calibration loop receives info from the previous loop, therefore displaying "do not pedal"

Trying without calibration avoids this, let's see first whether you get the ANT-interface working

WouterJD, 2021-02-01:
Apparently working

closed 2021-02-01 by WouterJD