WouterJD/FortiusANT #203: ANT dongle not disconnected properly, then reported "No (free) ANT-dongle found"

WouterJD, 2021-01-12:
You just had to make sure that each of the Apps had fully disconnected from the dongle and there were no "orphaned" processes hanging about using Task Manager if required

I tried running Bkool, RouvyAR and Zwift at this point but (of course) they couldn't see the brake unit as I only had the one Ant+ dongle. The second dongle turned up this morning and I have just successfully run pairing of the brake unit with all three Apps. It is the case that swapping between the Apps can sometimes leave orphaned processes, and I needed to "end task" on a few occassions to release the new Ant+ dongle if the App had been closed, but had some components still running in the background.

WouterJD, 2021-01-12:
First [reported by @RichardE-UK](https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-757476726).

I have this issue now and then myself; to be investigated and suggestions welcome.
It not only occurs when FortiusAnt was killed but also when apparently closed normal. So perhaps dongle not closed properly.

WouterJD, 2021-01-18:
@switchabl do you have suggestions how to properly close the ANTdongle so that it does not remain connected?

switchabl, 2021-01-18:
@WouterJD Normally that should happen automatically (but I think I've had this issue a couple of times myself). Maybe calling `usb.util.dispose_resources` would help. Maybe not.
https://github.com/pyusb/pyusb/blob/a16251f3d62de1e0b50cdfb431482d08a34355b4/docs/tutorial.rst#dont-be-selfish
https://github.com/pyusb/pyusb/blob/ffe6faf42c6ad273880b0b464b9bbf44c1d4b2e9/usb/util.py#L206

MeanHat, 2021-02-17:
I've been testing a Tacx Fortius using Windows and Dynastream Ant dongle today and have come across the same error report. FortiusANT worked initially but is now reporting "No (free) ANT-dongle found" despite not opening a CTP and rebooting PC. I've used Task Manager to close any obvious processes but with no success. I wanted to let you know @WouterJD to provide another data point.

WouterJD, 2021-02-18:
@MeanHat @switchabl The "only" way to reset is to remove the dongle and re-insert.
It seems the dongle itself is hanging and/or the physical re-insert resets the device-driver.

WouterJD, 2021-02-18:
@switchabl I have added:

```
        # https://github.com/pyusb/pyusb/blob/ffe6faf42c6ad273880b0b464b9bbf44c1d4b2e9/usb/util.py#L206
        AntDongle.devAntDongle.reset()
        #usb.util.release_interface(AntDongle.devAntDongle, ??)
        usb.util.dispose_resources(AntDongle.devAntDongle)
```
and will see whether this helps.

I have not added release_interface, have to look for parameter

WouterJD, 2021-02-22:
The additional "cleanup" code does not change the behaviour.
Sometimes python does not end, sometimes the dongle gets in undefined state and must be reset (remove/insert from USB port)

WouterJD, 2021-03-10:
THis happens and no additional suggestions received. Closed

closed 2021-03-10 by WouterJD