WouterJD/FortiusANT #204: No Trainer‘s found

stefantacx, 2021-01-13:
I can open the application but i can't find the Trainer. 
The Trainer works in this computer with Tacx Software.

marcoveeneman, 2021-01-14:
Hi Stefan, did you follow the installation instruction from the manual? I suspect you didn't install the correct usb drivers, the default Tacx driver doesn't work with FortiusANT, we need libusb. See chapter 3.5 of the manual.
Alternatively you can use an installer that does this automatically for you, see https://github.com/WouterJD/FortiusANT/issues/168#issuecomment-747721270

stefantacx, 2021-01-14:
Thank you, Marco!
I think this is my Problem. I World test it.

WouterJD, 2021-01-14:
hi @stefantacx 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

closed 2021-01-18 by WouterJD