WouterJD/FortiusANT #205: anfFE ANT+ Managed Network Document – Fitness Equipment Device Profile, Rev 5.0

inonoob, 2021-01-13:
Hey again, 

I checked your code [link](https://github.com/WouterJD/FortiusANT/blob/fc67aadb4a354e99f95bc9c7ad767b7729fccb57/pythoncode/antFE.py#L44) for antFE.py 

I also check the spec "ANT+ Managed Network Document – Fitness Equipment Device Profile, Rev 5.0" page 24 8.3. 
It says  a default pattern should be: 

64 Pages of repeating data page pattern + 2 Common Pages + 64 Pages of repeating data page pattern + 2 Common Pages

But in your code you seems to send do the following: 

30 Pages + 2 Common Pages + 30 Pages + 2 Common Pages. 

Why ? What do I miss here ?

WouterJD, 2021-01-14:
Page 24 8.3 Suggested Transmission patterns describes the suggested patterns; we are in sync on that.

Pattern b) is used (approximately) in FortiusAnt and is unchanged from antifier (the predecessor software) and I do not know why these choices were made.

inonoob, 2021-01-14:
Thanks @WouterJD for the answer. it just came up because I 'm currently reviewing my code which I partly use from your code. I'm trying to document my code so I can understand what I did and why. And as I wrote my text, I double checked with the specification and it came out that it's not as the suggested pattern b which I also use. It seems to work. Maybe the key word here is suggested pattern for older device.

WouterJD, 2021-01-15:
In your new code, I would properly follow the documentation.
Since you are in testmode, you will test the code.

I do not touch, since it works and why change - not knowing the reasons what it was ever decide this way and what impact it might have.

Please keep me informed.

closed 2021-01-19 by WouterJD