WouterJD/FortiusANT #206: FortiusANT master or exe fails to download

BikeSwimNotRun, 2021-01-15:
Hi there all.  I've tried several times and added various exceptions for the files to my AV but, still, these files fail to download from 'github/wouterJD/' every time.  Any ideas.

WouterJD, 2021-01-15:
Try to download .exe directly:
https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe

Tel us what antivirus program you have, what exception yiu registered and how.
PS. Try to disable AV and see what happens

WouterJD, 2021-01-15:
Google on "python executable windows antivirus" provides a lot of info, but that does not help you by now.

WouterJD, 2021-01-15:
@BikeSwimNotRun 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

BikeSwimNotRun, 2021-01-15:
Wouter, thank you for getting back.

To answer your questions.......
I'm using Bitdefender Total Security.
I have disabled the AV, restarted, tried and failed.
I have set exceptions for: raw.githubusercontent.com; github.com;
codeload.github.com;
Firefox browser: 84.0.2

Something is still getting in the way.  But, I do now have a copy of
FortiusANT.exe so I may be able to build a workable system with this based
on Bryan's post:

https://www.lunicus.org/Forum/viewtopic.php?f=16&t=483

Thanks again.

Stephen Thurgood

(M-UK) +44 7971 160808

WouterJD, 2021-01-15:
I do not know your AV bit I think you have te define an exception for the target-file:

c:/github/WouterJD/FortiusANT/WindowsExecutable/FortiusANT.exe

(Or similar)

BikeSwimNotRun, 2021-01-15:
Yep did this and the same issue.

Several days ago I had 'FortiusANT.exe' on my system and running.  I then
explored your manual and started playing with the setups.

Between then and now the problem has evolved.

I have recovered 'FortiusANT.exe' to a usb drive but cannot copy it to my
laptop.

The regular error code is 0x800700E1 which means windows thinks it contains
a virus or unwanted software.

I have used my AV to scan it and it reports it to be clean.

On-line searches highlight that windows can sometimes report this error and
there is nothing wrong.

I have disabled my AV (Bitdefender) and the same happens.  I have tried
adding exceptions but no luck.

I notice this unresolved and closed issue details similar problems:

https://github.com/gibaBR/Switch-Backup-Manager/issues/33

Any thoughts?

WouterJD, 2021-01-15:
Any thoughts?

I would suggest to carefully register what message is given by what program.

If you have disabled your AV, there should be jo issue. "means windows thinks it contains
a virus or unwanted software" Windows itself does not do AVdetection.

Are you sure you do not have two anti-virus solutions installed?

BikeSwimNotRun, 2021-01-16:
I have removed my AV, disabled Microsoft Defender, Firewall, pretty much
anything I can think of that may interfere and yet, the core system still
reports an error on attempting to download files:
[image: image.png]

or trying to transfer FortiusANT.exe from a USB onto my laptop:
[image: image.png]

Stephen Thurgood

WouterJD, 2021-01-16:
Please upload image through web interface; does not arrive by email

BikeSwimNotRun, 2021-01-16:
[image]
[image]

WouterJD, 2021-01-16:
I don't know
Why do you have two AV's?
A firewall does not block USB copy

There is something else...

BikeSwimNotRun, 2021-01-16:
I understand that MS antivirus kicks in (on Win10) in the absence of a
third party app.
And, as for the firewall,  I'm not an IT expert so anything that might get
in the way was disabled.

WouterJD, 2021-01-16:
MS antivirus + defender + your own AV?
Disable them all

Then what happens with copy from USB stick?

BikeSwimNotRun, 2021-01-16:
The previous graphicas show you.

Stephen Thurgood

(M-UK) +44 7971 160808

taylor42788, 2021-01-17:
I think I've noticed the same issue with version 5.0 on an HP PC running HP Sure Sense (comes as default) that detects FortiusANT.exe as a threat and puts it in quarantaine. 
Now, I've retried with version 4.1.1 that installs with no problem.
Has there been changes between 4.1.1 and 5.0 that make the security tool detect a threat in latest version and not in previous one?
Thanks

BikeSwimNotRun, 2021-01-17:
Ah. My Hp has Sense installed. But, it did allow the program to run on
initial download. A reboot may explain my issue as this would have
initiated a system check.

switchabl, 2021-01-17:
Unfortunately some high-profile antivirus engines are flagging the file (Microsoft, McAffee, Symantec) based on their heuristics:
https://www.virustotal.com/gui/file/0f3a4ebd88f312d91fd53db271eba0d2807897b5890044858e291960fa8d19cb/detection

My last autobuild (https://github.com/switchabl/FortiusANT/actions/runs/452848386, using #160) seems to do better although it should be the essentially the same code. A few false positives (due to bundled Python), but mostly obscure ones:
https://www.virustotal.com/gui/file/2c96eb06e3e595017d8f3bce208075556d601ce1886f881deb49008fa3149573/detection

Not sure what exactly makes the difference, maybe different Python/package version. A proper installer might help in the long-term but right now I don't have a good idea. If Windows Defender flags the file, a lot of people will have this problem and telling everyone to switch off their antivirus is not a great solution.

This is yet another reason to remove the binaries from the repo. Then you can at least clone the source without the antivirus interfering.

WouterJD, 2021-01-17:
Windows users will want to download a .exe version; I guess it makes no difference where it was built.

I think the AV does not to be disabled, the .exe should be excluded. But it will be difficult to provide clear instructions what to do.

It's unclear why one windows10 blocks and the other does not.

... to be continued

switchabl, 2021-01-17:
I've run the latest master through the Github build: https://github.com/switchabl/FortiusANT/actions/runs/492240850
https://www.virustotal.com/gui/file/2a3d9ad9e265f567d93d9c015f78cfb358f1e728e37f2baca4ef668c28ded412/detection

There are fewer false positives, but they are different again. And they include "Microsoft", but local Windows Defender thinks it is fine. All very random and confusing. Maybe the problem will just go away by itself with 5.1.

Anyway, @BikeSwimNotRun and @taylor42788, you can try this one, see if it works for you:
https://github.com/switchabl/FortiusANT/suites/1846357836/artifacts/35762318

WouterJD, 2021-01-18:
Thanks all for investigating.
Pleasse refer to [this message](https://github.com/pyinstaller/pyinstaller/issues/4633#issuecomment-688886522) in pyinstaller's own issue list.

I do not think there's much we can do; other than follow the recommendations as in mentioned post.

taylor42788, 2021-01-19:
Thank you. I could install the .exe generated by @switchabl and HP Sure Sense did not block it, but I've not tried it yet.
Thanks for the support!

WouterJD, 2021-01-24:
Since there seems not much we can do, issue closed.

closed 2021-01-24 by WouterJD