WouterJD/FortiusANT #208: Dongles don't recognise command

k4r1vince, 2021-01-16:
Hi guys, 

If someone could help me with this issue please, when fortiusant is calibrating i get this code..

[image]

I thought it was originally the cycplus dongles I had so I bought 2 Anself dongles from Amazon but i'm getting the same issue still
[image]
[image]

Help please

Karl

WouterJD, 2021-01-18:
Hi @k4r1vince 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
The CYCPLUS are known as notories "difficult" ANT-dongles; see #61 
Please attach logfile so I can check whether it's a CYCPLUS issue

k4r1vince, 2021-01-18:
Hi Wouter,

Thanks for replying, I read the issue with the cycplus dongles so I purchased the anself dongles but I'm still getting the same error. I will get a log tomorrow

Thanks

MeanHat, 2021-01-19:
Hi @k4r1vince 

I experienced the same problems with CYCPLUS ANT dongles and also Anself ones (from Amazon) which turned out to be rebadged CYCPLUS units. I swapped them for Garmin ones for one installation and Lifeline ones (https://www.wiggle.co.uk/lifeline-ant-usb) for another and both worked without problems (#61). A third installation uses Bluetooth (see #192) which removes the need for the dongles completely.

Good luck - Keith

WouterJD, 2021-01-19:
@all what is it that "we" do that the dongle goes banana's, where others use it to zwift with wahoo.
Use one channel only?

#Curious

k4r1vince, 2021-01-19:
Hi @MeanHat

Thanks so much for replying!

I did notice when i plugged the ANSELF dongles in it still said CYCPLUS dongles on the interface but just passed it off as the software, do the Garmin and Lifeline Dongles also read CYCPLUS or do they read their own name?

I've ordered 2 Lifelong dongles and they're coming Tomorrow so should hopefully have it working.

Thanks

Karl

switchabl, 2021-01-19:
@WouterJD I am not sure this even true. I have read numerous complaints about CYCPLUS dongles not working properly with Zwift.

What I think happened is this: the manufacturer decided to save money and roll their own ANT stack insteading of licensing the official one from Dynastream. But their implementation is buggy and not 100% compatible. It's probably not a single thing. One major thing is that apparently continuous scanning is broken (which Zwift normally uses but we don't). But I am pretty sure there are more subtle bugs. Zwift developers may or may have put in more work to identify those and work around them by now. There is also a chance that there exist several different revision by now, with some things fixed but not others.

@k4r1vince No, Garmin and Lifeline (and the other similar-looking ones) should read "Dynastream" (a Garmin subsidary that holds the ANT patents/trademarks and makes the official dongles/firmware stack).

MeanHat, 2021-01-19:
Hi @k4r1vince - I can confirm the comment from @switchabl that the Lifeline and Garmin dongles so indeed show up as "Dynastream". Good luck.

switchabl, 2021-01-19:
On a more constructive note, I would also be curious what exactly is going wrong and if it can be worked around. So if someone would be willing to send me a known-bad CYCPLUS they are not using I could take a closer look (I'm based in Germany). Although if you still can, I strongly suggest returning it for a refund instead. It's the only way they will stop selling broken ones.

k4r1vince, 2021-01-19:
Hi @switchabl 

I bought my Fortius turbo trainer from eBay last week and it came with the 2 CYCPLUS dongles which is why i then bought the ANSELF dongles. If the Lifeline dongles to the trick I can send both CYCPLUS dongles to you. I will return the ANSELF dongles for a refund.

Thanks

Karl

WouterJD, 2021-01-19:
FortiusANT recovers from some errors, not the now reported one.
A lot of time has already been invested; I will put energy in the other open issues.

WouterJD, 2021-01-19:
> complaints about CYCPLUS dongles

And CYCPLUS has confirmed that there are issues with (some versions of) their dongles

switchabl, 2021-01-19:
Yes, I know. It is quite frustrating. Unfortunately it still seems to be the single most frequent issue (more than drivers I think). What do you think about adding a big-fat warning in the README?

> Communication with Trainer Road, Rouvy, Zwift and others

> Communication with a Cycling Training Program (CTP) can be done using an ANT- or BLE-dongle (BLE = Bluetooth Low Energy)

**ANT dongles made by CYCPLUS (identical-looking ones sold as Anself and probably others) have been known to cause problems, avoid buying those.** Refer to {the manual, issue #X, issue #Y} (or something) for more info.

> The CTP can be run on another computer or Smartphone.

WouterJD, 2021-01-19:
It's at the top of the issues
https://github.com/WouterJD/FortiusANT/issues/174

WouterJD, 2021-01-19:
Readme adjusted; I don't write "don't buy these" to avoid formal/legal issues.

k4r1vince, 2021-01-22:
Hi @WouterJD @switchabl & @MeanHat I can confirm the Lifeline dongles work perfectly! They were reduced to less than the ANSELF dongles on Amazon so ended up being cheaper and having no problems at all.

I can send you the CYCPLUS dongles @switchabl if you send your address please

MeanHat, 2021-01-23:
@k4r1vince - ghreat news. Thanks for confirming

closed 2021-01-24 by WouterJD

switchabl, 2021-01-25:
@k4r1vince That's great to hear! Please send me an e-mail to `switchabl at mailbox dot org` about the CYCPLUS dongles.