WouterJD/FortiusANT #210: PowerFactor input on v5.0 different to v3.2.1 - option '-p 1.3' works on 3.2.1, errors on 5.0

pcuttriss, 2021-01-18:
Fortius Antifier v3.2.1

Works well with an option of ‘-p 1.3’

 
[image]

19:50:55,192: FortiusANT started
19:50:55,192: -a
19:50:55,197: -d 11 (0b1011)
19:50:55,197: -g
19:50:55,197: -H 0
19:50:55,197: -n
19:50:55,197: -p 1.3
19:50:55,197: ------------------
19:50:55,849: FortiusAnt GUI started in child-process
19:50:56,271: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
19:50:56,896: Connected to Tacx Trainer T1932
19:50:58,083: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
19:50:58,271: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
19:50:58,489: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
19:50:58,677: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595

 
Issue:
Passing in the same options as my working install of 3.2.1 the antifier shows 100W while the PowerTap is running at around 65W. Plus there is the error of:
Command line error; -p incorrect power factor=1.3

Fortius Antifier v5.0

[image]

19:59:00,990: Command line error; -p incorrect power factor=1.3
19:59:00,992: FortiusANT started
19:59:00,992: -a
19:59:00,992: -c False
19:59:00,993: -C 0/0
19:59:00,993: -d 11 (0b1011)
19:59:00,993: -D None
19:59:00,993: -g
19:59:00,993: -H 0
19:59:00,994: -n
19:59:00,994: -p 1.0
19:59:00,994: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
19:59:00,994: -t False
19:59:00,994: ------------------
19:59:01,916: FortiusAnt GUI started in child-process
19:59:02,457: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
19:59:03,081: Connected to Tacx Trainer T1932
19:59:03,421: Motor Brake Unit Firmware=4103 Serial= 6639 year=2010 type=T1941 Version2=3086 MotorBrake=True
19:59:04,806: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
19:59:04,993: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
19:59:05,210: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
19:59:05,416: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
19:59:05,635: FortiusANT receives commands from an ANT+ Generic Remote Control

In the current docs the ‘-p’ option has the following example:
-p FACTOR Adjust target Power by multiplying by this factor for static calibration.
If the power, displayed by FortiusAnt is always 10% higher than your power-meter,
specify -r90 to reduce resistance with 10%.
This is somewhat confusing as the option -r is mentioned within this description. Is this a typo?
Passing in the option -r90 just crashes the process….even the log doesn’t get generated.
Passing in the option -p90 results in the debug showing ‘-p 0.9’
Passing in the option -p110 results in the debug showing ‘-p 1.1’
Passing in the option -p130 results in the debug showing ‘-p 1.1’

Looking at the master branch I found the following in FortiusAntCommand.py
        #-----------------------------------------------------------------------
        # Get powerfactor
        #-----------------------------------------------------------------------
        if args.factor:
            try:
                self.PowerFactor = max(0.9, min(1.1, int(args.factor)/100 ))
            except:
                logfile.Console('Command line error; -p incorrect power factor=%s' % args.factor)

Is there a reason that the factor is limited to between 0.9 and 1.1? If there is a reason, then could you point me in the right direction to calibrating the brake appropriately please? I have tried the calibration mechanism, but it doesn’t seem to make any difference.

WouterJD, 2021-01-18:
Hi @pcuttriss 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

-r90 is a typo in the documentation, will be corrected next upload
I would expect next error message instead of a crash, that aside:
`FortiusAntCommand.py: error: argument -r/--Resistance: ignored explicit argument '90'`

The limit 90...110 % is introduced as a basic safeguard against odd numbers.
If practice shows that a 50...150% is better, please let me know and I will correct.

pcuttriss, 2021-01-18:
Hey Wouter. Thanks for the reply. I did put a quick message in issue #14 to introduce myself. I have been running this great app you created for a while now and was looking at the new version when I hit this issue.

I have just run from a command window and in that window you do get the message you expected.....previously I was running from a shortcut which provided no feedback before closing the shell.

C:\Users\PC>C:\Users\PC\Downloads\FortiusANT-5.0.exe -n -g -a -H0  -r90 -d 11
usage: FortiusANT-5.0.exe [-h] [-a] [-A] [-b] [-c CALIBRATERR] [-d DEBUG] [-D ANTDEVICEID] [-g] [-G GRADEADJUST]
                          [-H HRM] [-m] [-M] [-n] [-p FACTOR] [-P] [-r] [-R RUNOFF] [-s] [-t] [-x]
FortiusANT-5.0.exe: error: argument -r/--Resistance: ignored explicit argument '90'

If you could expand the option from 50 to 150 easily that would be great. I assume it would only require changing the max, min values in the PowerFactor section? I have been trying to run the latest master branch in python on windows but I keep getting crashes on the hardware scan.....will open another issue with the details when I get some better debug output. If I can get that running then I will have a hack at the code myself to see what it does :-)

I really do appreciate all of the work you have done on this as it keeps my old Fortius doing exactly what I need it to.....I am not a full time cyclist any more but having the turbo trainer running is helping me maintain a decent level of fitness while we wait to get back to Brazilian Jiu Jitsu. 

Dankje.

Paul

switchabl, 2021-01-18:
> The limit 90...110 % is introduced as a basic safeguard against odd numbers.
> If practice shows that a 50...150% is better, please let me know and I will correct.

> If you could expand the option from 50 to 150 easily that would be great.

+1 My Fortius reads more like 15-20% high as well. I also suggest adding a warning message if the input is out of range. I didn't even realize the limits exist.

WouterJD, 2021-01-18:
Warning messages would be the correct way to go; but again more code. 
I will extend the range.

Do you understand how it looks like with -p150, zwifting to your poor friend? That's why a tight limit was choosen. Too tight though

WouterJD, 2021-01-18:
I will also review other min/max limitations

switchabl, 2021-01-18:
> Do you understand how it looks like with -p150, zwifting to your poor friend? That's why a tight limit was choosen. Too tight though

150% will be fine though, that will be like riding your grandmother's bike. 50% on the other hand will let you fly up the KOM... (maybe 50% _is_ excessive, that's double the power)

Anyway, my personal philosophy has always been to let users do stupid things if they want to. I prefer to check for obvious typos/errors only, not limit what I think people should do. There is also always a chance it turns out to be useful in some weird scenario I haven't thought of (maybe doubling power makes sense if you have an injury and are pedaling with a single leg or something). But I don't race online and I guess it makes cheating very convenient. So the good folks at ZwiftPower may disagree with me.

WouterJD, 2021-01-19:
Corrected: Fortius Antifier v5.1 test VI

closed 2021-01-19 by WouterJD

jujroy, 2021-01-20:
@WouterJD @pcuttriss Hello everyone.  I tested the latest version (test VI) this morning using my Fortius (Motor Brake 110 Volts) and Zwift.  Using the configuration file, I inserted 130 in the -p Box.  Calibration and startup process went with no error.  I do not have a Power Meter to do a comparison.  My feeling (legs, hill, shifting, etc) is much closer to reality with this version and the 130 setting.  For the first time in six weeks, I had to do down to my 34 ring in a 10% climb.  My FTP went down as expected. Also worth mentionning that this time, I only used the virtual gearbox.  DI2 was left in 50/17 for the complete ride.  I would appreciate if riders with Power Meters could give us an idea of the -p number that they use along with the results and trainersetup.  Great work guys.  Will continue riding with this version and -p (130)  (https://www.strava.com/activities/4652438238)

WouterJD, 2021-01-21:
@jujroy 
Thanks for response
I think the activity is not public