WouterJD/FortiusANT #211: High power after 30 seconds

WouterJD, 2021-01-18:
Hi Wouter,

I have just bought a Fortius from e-bay.
I thank you for your time and effort on this project
I am from Glasgow Scotland.

I know nothing about the Fortius unfortunately 

I have blue and white control unit on the bottom it looks like it has a product number of T1932 Fortius
The unit the wheel connects with I assume you call this the Brake has T1941

As I am new to this I have not signed up to Zwift etc yet  I was looking to use the trainer in manual mode

I have used the following command to run the trainer: from my reading of the manual the following command should emulate a flat road 

FortiusANT.exe -g -d127 -a -H0 -G0/0 -m

19:10:39,792: FortiusANT started
19:10:39,793: -a
19:10:39,794: -c False
19:10:39,794: -C 0/0
19:10:39,795: -d 127 (0b1111111)
19:10:39,795: -D None
19:10:39,795: -G defines Grade = antGrade * 0 [* 0 (downhill)]
19:10:39,796: -g
19:10:39,796: -H 0
19:10:39,796: -m
19:10:39,797: -p 1.0
19:10:39,797: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
19:10:39,798: -t False
19:10:39,798: -x
19:10:39,798: ------------------
19:10:40,551: FortiusAnt GUI started in child-process
19:10:40,923: Connected to Tacx Trainer T1932
19:10:41,252: Motor Brake Unit Firmware=4103 Serial= _9999_ year=2009 type=T1941 Version2=3103 MotorBrake=True
19:13:49,344: FortiusAnt GUI ended
19:13:49,417: FortiusAnt ended

I run the calibration

I  can cycle for about 30 seconds and all appears fine then there is a large resistance on the trainer 
I have a 700c road bike and running with 52/12 gearing  ( seen on some of your posts speed of wheel is important hence the gear ratio )
 
Iam not sure if there is an issue with the trainer

I have uploaded the debug output can you help thanks

[FortiusANT.2021-01-18 19-10-39.log.txt](https://github.com/WouterJD/FortiusANT/files/5832206/FortiusANT.2021-01-18.19-10-39.log.txt)
[FortiusAntGUI.2021-01-18 19-10-40.log.txt](https://github.com/WouterJD/FortiusANT/files/5832207/FortiusAntGUI.2021-01-18.19-10-40.log.txt)
[FortiusANT.2021-01-18 19-12-10.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5832208/FortiusANT.2021-01-18.19-12-10.tcx.txt)
[FortiusANT.2021-01-18 19-10-39.json.txt](https://github.com/WouterJD/FortiusANT/files/5832209/FortiusANT.2021-01-18.19-10-39.json.txt)

_Originally posted by @Kennyac in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-762454236_

WouterJD, 2021-01-18:
Hi @Kennyac
I will look into it shortly.

One question, you run in manual power (-m) and defined -G0/0; why is that?

mattipee, 2021-01-18:
Hi, @Kennyac

I don't see Cadence being reported... have you got the cadence sensor connected and positioned. Usually after 4 seconds or so (in my experience), if the unit detects no cadence, it will "stop" the motor.

Kennyac, 2021-01-18:
I don't have a cadence sensor connected

The pedal connection was missing

Getting one later this week guy forgotten to include it

So that probably explains it

Thanks for your help

WouterJD, 2021-01-19:
Thanks @mattipee I will review doc to see whether some more info can be put there

WouterJD, 2021-01-19:
See manual paragraph 5.6; I have added some additional text for next upload.

closed 2021-01-19 by WouterJD