WouterJD/FortiusANT #212: The CTP doesn't change the power

kromis23, 2021-01-19:
Hi, I'm new to the forum and cycletrainers, I bought an Fortius T1932 this autumn and connected it to my MTB. I'm bridging the ant+ to bluetooth via an Viiiiva 4iiii to Rouvy, everything are connecting but Rouvy doesn't increase/decrease the power. If I increase/decrease the cadense I can see it changes but with a constant cadence it stays ~100W, i'm starting w/o parameters so the default values are used.

I have read the manual and forum to find something that could explain the this, the closest are the one below.
Manual adjustment of the requested power on the headunit (in ERGmode) #195

Are there something more I should do or should this work ?

I'm located in Sweden.

/ Lars-Erik

switchabl, 2021-01-19:
The Viiiiva doesn't appear to support forwarding the FE-C commands (used for controlling power/slope) according to this:
https://support.thesufferfest.com/hc/en-us/articles/360015030400-The-4iiii-Viiiiva-Heart-Rate-Monitor-Ant-to-Bluetooth-Bridge

The latest version of FortiusANT supports now Bluetooth directly but you will need a Bluetooth LE dongle.

kromis23, 2021-01-20:
Thanks for clarifying this, I had missed that Viiiiva couldn't send back the information.

Great that bluetooth works, I will try that out instead.

closed 2021-01-20 by kromis23

WouterJD, 2021-01-20:
Hi @kromis23 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -