WouterJD/FortiusANT #215: Calibrate Vortex using FortiusAnt

WouterJD, 2021-01-20:
Hello, 
first of all I am really excited about the way this project is improving and its community is growing. You are doing a great job and thank you for that.

I am using i-Vortex (T1961) with head unit (T1962) and Tacx supplied dongle and after some trial & error everything seems to be working and I can use Zwift on my smartphone. I was just wondering (could not find the information in the manual neither in issues) how can I calibrate Vortex when using FortiusANT since Rundown procedure can be used only with motor breaks and the calibration process (with running the break automatically) is probably not supported either.

Will just calibrating my Vortex using the head unit and then using both HU (in PC mode) and Vortex during the ride do? And if so, do I need to aim for calibration with offset as close as possible to 0?

Thanks in advance for the answer.
Cheers

_Originally posted by @cermatej in https://github.com/WouterJD/FortiusANT/issues/46#issuecomment-763547268_

WouterJD, 2021-01-20:
The fact that rundown is not implemented for Vortex is because of it would require a serious redesign of the software. If it would be required of course it could be done.

Whether calibration is possible I do not known, not having a Vortex myself and/or the specs how it could work.

`Will just calibrating my Vortex using the head unit and then using both HU (in PC mode) and Vortex during the ride do?`
I would suppose yes.

`do I need to aim for calibration with offset as close as possible to 0?`
I cannot tell you that; what is the procedure, using the original software?

cermatej, 2021-01-20:
Thanks for the quick answer. 

So far I have been riding Zwift without the head unit connected since it thought it does not make any difference on the watts produced.

Neither Vortex nor HU manual does not say anything about being as close as possible to 0 value when calibrating. It just adjusts the "resistance bias" based on the calibration test (30km/h and immediate stop). It was just my heuristic that when tightening the brake too much produced positive bias values and vice versa.

My understanding was that the Vortex would be similar to the Bushido in this regard, which according to the manual works with FortiusANT based on the HU calibration.

I can try today both extremes with calibrating on:
- too loose brake
- too tight brake
and see if the actual output is the same when pedaling at similar intensity (which would mean that FortiusANT works with the Tacx calibration).

And maybe trying the same with just the HU at first to see if there is actually a need to be close as possible to 0 or the output values will be adjusted according to the calibration.

switchabl, 2021-01-20:
That's an interesting one. The Bushido saves the calibration value internally in the brake, so if you initiate calibration using the head unit and then ride with FortiusANT, everything should be fine. With the Vortex it is less clear. TTS4 re-sends the calibration value with every message (like with the Fortius, where it is not saved by the brake). But the GoldenCheetah implementation doesn't, so maybe it is not strictly necessary.

Usually, if the calibration value is too high, simulating 0% realistically might not be possible anymore (because of the minimum resistance) and if it is too low, the tyre can slip. But there should be no need to get to exactly 0. If it turns out the brake doesn't retain the calibration with FortiusANT then tuning to exactly 0 might be a useful work-around though.

Implementing calibration doesn't look too hard, but I don't have a Vortex either. So that makes it somewhat cumbersome.

i-Vortex manual: https://www.libble.eu/tacx-i-vortex-computer/online-manual-595470/

cermatej, 2021-01-20:
I performed mentioned tests and the results were following:

- zero value seems to reflect a optimal pressure (unsuprisingly) on the wheel since whenever the brake was too loose the cal value would go to negative numbers (down to around -5 when the brake was still tight enough not to slip). On the other hand up to around +3 when the brake was too tight (after a few seconds of pedaling the tyre was getting quite warm).
- in around half of trials to run the script with HU running the HU did not seem to respond (no sign of PC mode and screen stayed the same).
- whenever running Zwift with HU on (PC mode or default) the resistance was unstable - probably jumping from Zwift target res and HU's target res
- although when the HU stayed in default mode and no CTP was running, the power output was the same for FortiusANT and HU

But so far it seems that the HU cant be used with CTP bc of the resistance drops which is fine for me. I was just curious whether the numbers produced were correct in CTP.

switchabl, 2021-01-20:
I suspect the problem is that we send the message to switch to PC-mode only once. If that message is lost, the HU stays in stand-alone mode and keeps trying to override the resistance setting. For the Bushido, we keep re-sending until we get a confirmation. I can try to make a patch if you are willing to test it.

cermatej, 2021-01-20:
The thing is that it seems that the resistance is being overriden in PC mode as well so I'm not sure if the fix would work.

switchabl, 2021-01-22:
@cermatej If that is indeed the case, that complicates things (sounds hard to debug without direct access). But one step at a time I guess. Fixing the mode-switching should not be too hard. And there's still a chance if we can switch the HU to PC-mode properly (and keep it there!) it will just work.

WouterJD, 2021-01-22:
From memory I recall that a sort of keep-alive is sent. I did not have this experience when tedting; I had a vortex for test for some time

switchabl, 2021-02-11:
I now have a Vortex for testing, so calibration directly from FortiusANT will very likely happen at some point (probably as a part of the big calibration update for all trainers that I am working on, but that will take some more time). For now, I have verified that if you calibrate with the head-unit, the calibration is also applied in FortiusANT and the brake seems to retain the calibration value even when you turn it off (very similar to Genius and Bushido brakes).

@cermatej PR #252 should now make sure that the HU always switches to PC-mode (among other things). A Windows binary is available here: https://github.com/switchabl/FortiusANT/suites/2021188431/artifacts/40537444

I couldn't reproduce the other problem so far, but I will keep trying (see #219). Can you also check your brake and head-unit firmware versions? (Turn on head-unit, hold left+right key for 3 seconds, brake and HU versions are in the second row) Maybe this is just something that Tacx fixed in firmware at some point.

cermatej, 2021-02-12:
Hi! first of all thanks for the update. So far I have been using HU only for calibration and afterwards I turn it off when riding Zwift (and the percieved effort seemed to be similar to the numbers I have been producing) so I was ok with that.

ad #219 Actually at first when testing this project I recall power drops when using manual mode (from value set by HU down to 100W target) but I will test this one as well as Zwift with HU powered on hopefully sometime today and post the results.

My versions:
 - HU: 0.1.5
 - brake: 0.2.1

switchabl, 2021-02-13:
@cermatej In #219 @ditisdirkdewit has reported that the problem is solved by upgrading the head unit firmware to version 0.2.5

cermatej, 2021-02-13:
Hi! That's great 😊 I tried to test it today morning, but the could not download the executable. Now it works fine so I will test it with the new firmware as well.

WouterJD, 2021-02-17:
Since there is no communication here, I assume resolved

closed 2021-02-17 by WouterJD

switchabl, 2021-02-17:
@WouterJD The head unit issue is resolved (I think) and I have confirmed that calibration through the head unit unit works.

Adding calibration directly from FortiusANT, for convenience and if you don't have a working head unit, is still on my todo list. I don't know if you want to keep the issue open for that or not.

orrmany, 2023-11-17:
@switchabl : I have a T1961 brake/ T1962 HU combo. I have power stability problems with CTP, see issue #442.
Do you think that firmware might be the reason? (brake ver. 0.1.5, HU ver is 0.2.2)

WouterJD, 2023-11-17:
I will respond on #442 shortly; I think it's not related.
I do not have (or ever had) a Vortex and have no information regarding power stability.
I did not hear about required upgrades before.

Please explain the power stability issues in more detail.