WouterJD/FortiusANT #217: Issue using ID of Garmin Tri HRM in startup options

pcuttriss, 2021-01-23:
Hi Wouter

The power factor input #210 is working brilliantly....just been setting up  5.1+4.2 quality upgrade (#198) for a session this afternoon.

I thought I would do a specific option for the Garmin HRM Tri unit I have which has ID 730926 but there appears to be an issue with the length of that ID as shown below.

struct.error: ushort format requires 0 <= number <= 0xffff

Paul C

######################################
14:58:04,070: FortiusANT started
14:58:04,071: -a
14:58:04,071: -c False
14:58:04,071: -C 0/0
14:58:04,071: -d 11 (0b1011)
14:58:04,072: -D None
14:58:04,072: -g
14:58:04,072: -H 730926
14:58:04,072: -n
14:58:04,072: -p 1.35
14:58:04,073: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
14:58:04,073: -t False
14:58:04,073: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
14:58:04,073: ------------------
14:58:04,983: FortiusAnt GUI started in child-process
14:58:05,638: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
14:58:06,248: Connected to Tacx Trainer T1932
14:58:06,814: Tacx head unit returns incorrect header 0x21303 (expected: 0xc03)
14:58:07,033: Motor Brake Unit Firmware=0x1007 Serial= 6639 year=2010 type=T1941 Version2=3086 MotorBrake=True
14:58:07,143: FortiusAnt applies the MotorBrake power curve
14:58:08,548: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
14:58:08,737: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), id=730926 only
Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "FortiusAnt.py", line 422, in Tacx2DongleThread
  File "FortiusAnt.py", line 150, in Tacx2Dongle
  File "FortiusAntBody.py", line 556, in Tacx2Dongle
  File "FortiusAntBody.py", line 643, in Tacx2DongleSub
  File "antDongle.py", line 803, in SlaveHRM_ChannelConfig
  File "antDongle.py", line 1302, in msg51_ChannelID
struct.error: ushort format requires 0 <= number <= 0xffff

WouterJD, 2021-01-23:
@pcuttriss 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

The ANT protocol defines a number with max 65535, ... Please retry without -H and look what HRM is found.
Why do you want to spevify the ID?

pcuttriss, 2021-01-24:
Hey Wouter

Thanks for the response. I have done some digging and I see what you mean by the ANT+ spec for the ID. It seems that some manufacturers sometimes do some weird stuff with ID's and it seems that Garmin is one of them. I did some more digging and from Golden Cheetah it seems that the base ANT+ ID is in fact 10030.

No idea why there is this difference or what purpose it serves. 

To your other question though: I wanted to make sure that I was receiving data from the HRM strap rather than my watch. But in this case it is irrelevant as the HRM value is already logged in Golden Cheetah.

The ID that was logged when I dropped the -H0 option wasn't the ID that GC uses.....not sure where it came from.

I don't see this is a valid issue any more....but I have learnt a bit more about ANT+ so thank you :-)

Paul C

WouterJD, 2021-01-24:
The ID for ANT is hopefully unique amongst alll HRM's in the neighborhood and limitted to two bytes.
Then it's up to the manufacturer to generate a unique number. Could be & 0xffff but of course anything else.

closed 2021-01-24 by WouterJD

WouterJD, 2022-01-04:
ANT ID's may be larger, see also #342