WouterJD/FortiusANT #218: Problem to get Bluetooth to work on Windows

kromis23, 2021-01-23:
I am trying to get the Bluetooth interface working, I have followed these instructions.
https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md
From the errors it looks to be related to the nodejs but I can't see what I have missed.

In section "6. Install dependencies" I manually created the directory node "C:\Fortiusant\node" and typed the command "npm install" and get the following message. It looks like the package.json are expected to be there, should the node directory been created there in an earlier stage and how ?

C:\Fortiusant\node>npm install
npm WARN saveError ENOENT: no such file or directory, open 'C:\Fortiusant\node\package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open 'C:\Fortiusant\node\package.json'
npm WARN node No description
npm WARN node No repository field.
npm WARN node No README data
npm WARN node No license field.

up to date in 0.552s
found 0 vulnerabilities
 

I started FortiusAnt with the following parameters -a -g -b -H0 -G50 -d64 -n -s
 

The nodejs directory are "C:\Program Files\nodejs"
 
This is the trace file
-----------------------------------------------------------------------

16:25:58,387: FortiusANT started
16:25:58,387:     Restart=False debug=64
16:25:58,387: -a
16:25:58,387: -b
16:25:58,387: -c False
16:25:58,388: -C 0/0
16:25:58,388: -d 64 (0b1000000)
16:25:58,388: -D None
16:25:58,388: -G defines Grade = antGrade * 0.5
16:25:58,388: -g
16:25:58,388: -H 0
16:25:58,389: -n
16:25:58,389: -p 1.0
16:25:58,389: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
16:25:58,390: -s
16:25:58,390: -t False
16:25:58,390: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
16:25:58,390: ------------------
16:25:58,391: Windows executable started
16:25:58,391: Version info for the components
16:25:58,651: Version=Fortius Antifier v5.1, on github=Fortius Antifier v5.1.
16:25:58,654: Fortius Antifier v5.1
16:25:58,654:            FortiusAnt = 2021-01-10
16:25:58,654:             antDongle = 2020-12-30
16:25:58,654:                 antFE = 2020-12-28
16:25:58,654:                antHRM = 2020-12-27
16:25:58,654:                antPWR = 2020-12-28
16:25:58,654:                antSCS = 2020-12-27
16:25:58,654:             bleDongle = 2021-01-04
16:25:58,654:             constants = 2021-01-18
16:25:58,654:                 debug = 2020-12-18
16:25:58,655:        FortiusAntBody = 2021-01-14
16:25:58,655:     FortiusAntCommand = 2021-01-19
16:25:58,655:         FortiusAntGui = 2021-01-18
16:25:58,655:               logfile = 2020-11-25
16:25:58,655:            RadarGraph = 2020-05-07
16:25:58,655:              settings = 2021-01-19
16:25:58,655:       structConstants = 2020-01-25
16:25:58,655:             TCXexport = 2020-12-20
16:25:58,655:            usbTrainer = 2021-01-12
16:25:58,655:              argparse = 1.1
16:25:58,655:                 numpy = 1.19.4
16:25:58,655:                    os = nt
16:25:58,656:               windows = 10.0
16:25:58,656:                pickle = 4.0
16:25:58,656:              platform = 1.0.8
16:25:58,656:          sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
16:25:58,656:                   usb = 1.1.0
16:25:58,656:                    wx = 4.1.1
16:25:58,656: FortiusANT code flags
16:25:58,656:    UseMultiProcessing = True
16:25:58,656:                UseGui = True
16:25:58,657:          UseBluetooth = True
16:25:58,657: ------------------
16:26:00,985: BleInterface.Open() ...
16:26:00,986: ... Popen(C:/node,['node', 'server.js'])
16:26:00,991: ... completed
16:26:00,991: FortiusANT exchanges data with a bluetooth Cycling Training Program
16:26:00,991: ---------- BLE-devices are activated ----------
16:26:00,994: BleInterface.Write({'heart_rate': 0, 'watts': 12.25, 'cadence': 99.0})
internal/modules/cjs/loader.js:883
  throw err;
  ^

Error: Cannot find module '../index.js'
Require stack:
- C:\node\server.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (C:\node\server.js:2:10)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ 'C:\\node\\server.js' ]
}
16:26:05,093: ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x057E3460>: Failed to establish a new connection: [WinError 10061] Det gick inte att göra en anslutning eftersom måldatorn aktivt nekade det'))
16:26:05,094: Tacx2Dongle; Processing time 4.100 is 3.850 longer than planned 0.250 (seconds)
16:26:05,094: BleInterface.Write({'heart_rate': 0, 'watts': 23.9153125, 'cadence': 99.99})
16:26:09,172: ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x057E3910>: Failed to establish a new connection: [WinError 10061] Det gick inte att göra en anslutning eftersom måldatorn aktivt nekade det'))
16:26:09,174: Tacx2Dongle; Processing time 4.081 is 3.831 longer than planned 0.250 (seconds)
16:26:09,175: BleInterface.Write({'heart_rate': 0, 'watts': 33.091639453125, 'cadence': 100.98})

kromis23, 2021-01-24:
I created the file package.json in C:\Fortiusant\node from the link below
https://github.com/noble/node-bluetooth-hci-socket/blob/master/package.json

Next error was that it server.js in directory c:\node, I found server.js in directory C:\Fortiusant\node\node_modules\hwk\lib and copied the file to c:\node (probably wrong server.js but I didn't find anything in the github.com/noble/node-bluetooth-hci-socket forum or other place on my PC)

After that the bluetooth seems to start (se picture) but I get errors from the server.js and it also looks like the http server is not started or at least doesn't answer on port 9999, wrong port or antivirus ? (add added an exception for fortius.exe and cmd file w/o result). There are no registred ports in C:\Windows\System32\drivers\etc\services on 9999 or python.

[FortiusAnt.2021-01-24 10-04-45.log](https://github.com/WouterJD/FortiusANT/files/5861855/FortiusAnt.2021-01-24.10-04-45.log)
[image]

WouterJD, 2021-01-24:
I see 
`C:\FortiusAnt\node`
and
`C:\node`
which seems inconsistent

kromis23, 2021-01-25:
I tried to follow the instructions "(https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md".
I have done the following changes to solve the errors in the trace file, after that the bluetooth interface ssems ok according to the logs, but I can't connect to bluetooth from my phone.

I have done the following changes compared to "6. Install dependencies"
----------------------------------------------------
- <FortiusANT_location> = "C:\FortiusAnt"
- <node_location> = "c:\node"
  I can't se any relations between <FortiusANT_location> and <node_location>, but <node_location> needs to be "c:\node" according to trace file
- I created the file "package.json" from the 51 lines in the URL "https://github.com/noble/node-bluetooth-hci-socket/blob/master/package.json"
- I copied "crypto.js", "utils.js" and "server.js" from directory "c:\node\node_modules\hawk\lib" to "c:\node" since the trace showed that they was expected in "C:\node"
- I Started the python HTTP Server manually from a command prompt "python -m http.server 9999"
  For the URL"http://localhost:9999" i get the following, just to verify that the server works
  Directory listing for /
    . build/
    . lib/
    . node_modules/
    . crypto.js
    . utils.js
    . server.js
- If I removed the files "crypto.js", "utils.js" and "server.js" from c:\node I got errors in the trace file and have therefore kept them.

Trace files
----------------------------------------------------
After the changes above the trace file looks like this which seems OK (I also attached bluetooth/full trace files).
20:19:48,700: Connected to Tacx Trainer T1932
20:19:49,037: Motor Brake Unit Firmware=0x1004 Serial= 5020 year=2008 type=T1941 Version2=2052 MotorBrake=True
20:19:49,145: FortiusAnt applies the MotorBrake power curve
20:19:49,254: Retry because short buffer (len=64) or incorrect header received (expected: 0x21303 received: 0xc03)
20:19:50,401: BleInterface.Open() ...
20:19:50,402: ... Popen(C:/node,['node', 'server.js'])
20:19:50,408: ... completed
20:19:50,408: FortiusANT exchanges data with a bluetooth Cycling Training Program
20:19:50,409: ---------- BLE-devices are activated ----------
20:19:50,413: BleInterface.Write({'heart_rate': 0, 'watts': 0, 'cadence': 0})
20:19:50,668: BleInterface.Write({'heart_rate': 0, 'watts': 0, 'cadence': 0})
20:19:50,932: BleInterface.Write({'heart_rate': 0, 'watts': 0, 'cadence': 0})

I get an incorrect header information, i'm not sure what this means
20:19:49,254: Retry because short buffer (len=64) or incorrect header received (expected: 0x21303 received: 0xc03)

When I look on the python HTTP server I get the following messages
::1 - - [25/Jan/2021 20:19:50] "POST /ant HTTP/1.1" 501 -
::1 - - [25/Jan/2021 20:19:50] code 501, message Unsupported method ('POST')
::1 - - [25/Jan/2021 20:19:50] "POST /ant HTTP/1.1" 501 -
::1 - - [25/Jan/2021 20:19:50] code 501, message Unsupported method ('POST')

These messages matches FortiusAnt writing to the BLE adapter, I think Fortius are sending a wrong Post to the HTTP server.
20:19:50,413: BleInterface.Write({'heart_rate': 0, 'watts': 0, 'cadence': 0})
20:19:50,668: BleInterface.Write({'heart_rate': 0, 'watts': 0, 'cadence': 0})

[Bluetooth-Trace__FortiusAnt.2021-01-25 20-19-47.log](https://github.com/WouterJD/FortiusANT/files/5869382/Bluetooth-Trace__FortiusAnt.2021-01-25.20-19-47.log)
[Bluetooth-Trace__FortiusAntGUI.2021-01-25 20-19-48.log](https://github.com/WouterJD/FortiusANT/files/5869384/Bluetooth-Trace__FortiusAntGUI.2021-01-25.20-19-48.log)
[Full-Trace__FortiusAntGUI.2021-01-25 20-29-47.log](https://github.com/WouterJD/FortiusANT/files/5869386/Full-Trace__FortiusAntGUI.2021-01-25.20-29-47.log)
[Full-Trace__FortiusAnt.2021-01-25 20-29-46.log](https://github.com/WouterJD/FortiusANT/files/5869387/Full-Trace__FortiusAnt.2021-01-25.20-29-46.log)

Some outstanding questions:
----------------------------------------------------
- I get an "incorrect header received (expected: 0x21303 received: 0xc03)" at every startup, is this an issue ?
- The FortiusAnt Post gives an HTTP 501 error, is this an issue ?
- I started the python server manually from a command prompt, can this be added to the FortiusAnt executable or have I missed some steps ?
- The bluetooth interface seems to be active from Fortius but I can't see it from my phone, what more do I need to do ?
[image]

Best Regards, Lars-Erik

marcoveeneman, 2021-01-28:
> In section "6. Install dependencies" I manually created the directory node "C:\Fortiusant\node" and typed the command "npm install" and get the following message. It looks like the package.json are expected to be there, should the node directory been created there in an earlier stage and how ?

Hi @kromis23,

Hold on, where did you read that you need to create a node folder? The manual states you need to go to the node folder in the FortiusANT direcory which contains all the required files, this folder is already present. It doesn't say you need to create the node folder yourself since that will of course not work.

Hope this helps.

Marco

kromis23, 2021-01-28:
Hi Marco, I think it’s clearer, I have used the Windows  executable for ANT+ and it worked to have it in any folder and I think that’s the confusion. Therefore I have used Windows executable also for Bluetooth, I started the HTTP server manually and changed the directory because of the errors in the trace file.

Does this mean that the Windows executable shouldn’t be used and I should follow the Python installation to get Bluetooth working for Windows ? (that was not clear to me when reading the Bluetooth installation)

Lars-Erik

kromis23, 2021-01-28:
When I used ForiusAnt with ANT+ I had only the executable downloaded (no python installed since it was not needed), when I followed the bluetooth instructions I was confused about the \node directory.

After your comment I read through the pdf documentation again and followed the "3.4 Install python" with "pip install -r requirements.txt" and after that it worked.

In the "https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md" documentation there could be a new step 6, and change stpe 6 to 7 as below. Maybe obvious but it confused med since I was using the windows executable.

5. Replace the driver for your bluetooth dongle using Zadig
6. For Windows installations only
   Make sure you have followed "3.4 Install python" step in "FortiusANTUserManual.pdf" manual if you have used the Windows executable.
7. Install dependencies

Thanks for the help, Lars-Erik

closed 2021-01-28 by kromis23