WouterJD/FortiusANT #219: Power drop on Vortex

ditisdirkdewit, 2021-01-23:
Hi,

Thanks Wouter for this amazing piece of software. Until now I was using Zwift in combination with my tacx Vortex & head unit only with a Bluetooth cadence and speed sensor. So I am very exited to get my Vortex back to full functionality. 

I was able to successfully pair everything with Zwift, bot noticed that as soon as resistance is required, the resistance works for approx 5 sec and the drops back to 0 for 20 sec and then another 5 sec of resistance. (Resistance is stable when set on head unit without ant connection).

Same behavior when I run in manual mode. Do you have any tips on how to resolve this?

Thanks!

WouterJD, 2021-01-24:
Hi @ditisdirkdewit 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

It's not clear immediatly; you might do a test-run (as short as possible, showing the issue) with -d127 and post the logfile.

WouterJD, 2021-01-24:
Since the headunit is not pairing, it might be that head-unit and FortiusANT are both changing the power.
Since the headunit is only used for the buttons; perhaps try to keep the headunit switched off.

ditisdirkdewit, 2021-01-25:
Hi @WouterJD 

Thanks for the quick response. I'm from NL. I have a Tacx I-Vortex (Headunit T1962 & brake T1961). Bought it new already quite some years ago. Laptop that ran the Tacx software crashed and since then I'm using my bluetooth speedsensor.  Bicycle used is a Specialized Allez.

I'm using FortiusAnt on a Windowslaptop. 

The headunit seems to be paired, see attached the log files. As you might be able to see it takes quite some time before the brakes kicks in and when it works it only works for a few seconds.

[FortiusAnt.2021-01-25 07-29-48.json.txt](https://github.com/WouterJD/FortiusANT/files/5864972/FortiusAnt.2021-01-25.07-29-48.json.txt)
[FortiusAnt.2021-01-25 07-29-48.log.txt](https://github.com/WouterJD/FortiusANT/files/5864973/FortiusAnt.2021-01-25.07-29-48.log.txt)
[FortiusAntGUI.2021-01-25 07-29-49.log.txt](https://github.com/WouterJD/FortiusANT/files/5864974/FortiusAntGUI.2021-01-25.07-29-49.log.txt)
[FortiusANT.2021-01-25 06-29-52.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5864975/FortiusANT.2021-01-25.06-29-52.tcx.txt)

ditisdirkdewit, 2021-01-25:
Thanks for suggestion on the head unit, was able to test it without the head unit just now. 
The brake works now properly with Zwift, so indeed it is probably an issue with the head unit sending different power signals. 
Still strange that the pairing seems to be fine?

switchabl, 2021-01-25:
> Still strange that the pairing seems to be fine?

Not just that, according to the logs the head unit clearly reports it is in PC/config mode, similar to what @cermatej reported in #215. I wonder what we are missing. It _shouldn't_ interfere in PC mode, it is what TTS uses as well.

ditisdirkdewit, 2021-01-25:
Agreed, let me know if there are certain tests I can do to identify the issue.

WouterJD, 2021-02-04:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2021-02-04 by WouterJD

switchabl, 2021-02-11:
@ditisdirkdewit Sorry, it's been a while, but I now have a Vortex here, so I have a somewhat better idea what to do.

First, I have made a small patch (#252) which should make the Vortex somewhat more reliable overall. It may also help with this issue. Chances are it doesn't, but it may be worth a try.
You can find the Windows binary here: https://github.com/switchabl/FortiusANT/suites/2021188431/artifacts/40537444

I couldn't reproduce the problem on my T2171/T1962 but I wonder if it might be firmware-related. Can you check your firmware version? (Turn on head-unit, hold left+right key for 3 seconds, brake and HU versions are in the second column).
I have brake version 0.2.7 and head-unit version 0.2.5 on mine (which I think are the latest ones).

ditisdirkdewit, 2021-02-13:
@switchabl Thanks! Really appreciate the effort. I checked my firmware and it seems the brake firmware is far behind. The brake is on 0.1.5 (T1961) and headunit was on 0.2.2 (T1962). I was able to upgrade the headunit to 0.2.5 but couldn't find any firmware upgrade for the brake. 

But on the bright side, after the headunit firmware upgrade, it seems to work properly after doing a quick test :-). Also with the update binary. Thanks for the suggestion!

switchabl, 2021-02-13:
@ditisdirkdewit That's great to hear. Tacx didn't use the head-unit the way we do until TTS 4.12 and firmware 0.2.5 was released around the same time, so there is a good chance it just wasn't really supported with the old one.

Latest firmware binaries:
- Vortex HU: 0.2.5 (2014/10/13)
https://web.archive.org/web/20160825162210/http://patches.tacxdata.com/data/files/firmware/Vortex_computer_025_0.bin
- Vortex brake: 0.2.7 (2015/07/03)
https://web.archive.org/web/20160825162159/http://patches.tacxdata.com/data/files/firmware/VER_firmware_0274.bin

I don't have the changelogs unfortunately (not sure if they were ever released).

@WouterJD This is something to keep in mind, and possibly put in the manual. There are probably many old Vortex and Bushido trainers around that have only ever been used stand-alone and are still on the original firmware version.

ditisdirkdewit, 2021-02-15:
Thanks @switchabl! Updated the firmware and did a longer test ride. works much more stable now.

cermatej, 2021-02-15:
Hi @ditisdirkdewit! I am having a similar problem with my i-Vortex  and I was not able to update either firmware. Tried using TacxUpdater.exe from my TTS4 installation, but that just checks the internet for updates and end up telling me that I have the latest version of the FW. How do I then install these .bin files? Thanks

ditisdirkdewit, 2021-02-15:
Hi @certmatej, I also used TacxUpdater.exe from the TTS4 installation. But with me, once it is not able to find the firmware on the Internet it provides an option to select a local .bin file. 
Have you installed the complete TTX software? If not you receive less options to choose from. When you start the Tacxupdater do you  have the ability to select the Vortex from the menu?

cermatej, 2021-02-15:
Yes, I installed the whole TTS version 4.4.* although I could not activate it since I I bought the Vortex 2nd hand and the activation code provided did not work, but I think there is no need to activate the software to run TacxUpdater.exe. Yep, I have about 6 trainers to choose from as well as the whole TTS update, but when I choose Vortex it just checks if the dongle is connected and then once the internet check fails I have only option to quit the updater. Could you maybe provide me just the updater executable or the link for the TTS (if you have installed from the internet)?

ditisdirkdewit, 2021-02-15:
@cermatej I can. please see here:
[tacxUpdate.zip](https://github.com/WouterJD/FortiusANT/files/5980301/tacxUpdate.zip)
pass: 12345678

Let me know when you have it, I'll remove the link/package.

cermatej, 2021-02-15:
Worked as a charm! Both FWs updated. Seems like TTS 4.4.* did not have this option.
Thank you

ditisdirkdewit, 2021-02-15:
@cermatej Great! hope everything works properly for you as well now.

VanSalimH, 2022-03-16:
I have a T2171 i-Vortex that was on brake firmware 0.2.3 and HU firmware 0.2.1.  Upgrading to 0.2.7 for the brake and 0.2.5 for the HU fixed the powerdrop problem for me.  Everything works very nicely now.