WouterJD/FortiusANT #223: Flat resistance on magnetic brake at low wheel speed

WouterJD, 2021-01-26:
@bikebeppe64 has a T1932 headunit and a T1901 brake

FortiusAnt 5.1 reports:
```
16:29:50,303: Motor Brake Unit Firmware=0x1902 Serial=0 MotorBrake=False
16:29:50,428: FortiusAnt applies the MagneticBrake power curve
```

A ride is done, using Rouvy, and attached files are provided for analysis.
[FortiusAnt_CCS - Bormio FA 51VI_TEST.log](https://github.com/WouterJD/FortiusANT/files/5872796/FortiusAnt_CCS.-.Bormio.FA.51VI_TEST.log)
[FortiusAntGUI_CCS - Bormio FA 51VI_TEST.log](https://github.com/WouterJD/FortiusANT/files/5872798/FortiusAntGUI_CCS.-.Bormio.FA.51VI_TEST.log)

WouterJD, 2021-01-26:
The TCX file is from the Garmin Powermeter; json and logfiles are from FortiusAnt.
Combined together it provides the following image:
[image]

WouterJD, 2021-01-26:
Explanation:
Input:
[1] TargetGrade as provided by Rouvy (multiplied by 100 for scaling purpose)
[2] SpeedKmh (the speed of the bicycle wheel) = speed of the brake roller
Calculated:
[3] TargetPower
[4] TargetResistance (divided by 10 for scaling purpose)
Obtained from brake/headunit:
[5] CurrentResistance
Calculated:
[6] CurrentPower
Obtained from Garmin powermeter:
tcxWatts

WouterJD, 2021-01-26:
### Displayed Power

Observation: the tcxWatts (power meter) and CurrentPower (FortiusAnt and Rouvy display) curves nicely match; there seems to be a constant factor.
The -p command-line option can be used to adjust this.

switchabl, 2021-01-26:
The brake resistance seems to be maxed out most of the time. From the data, it looks like the rolling resistance is significantly lower than what is set. So instead of changing -p, I suggest either
- adjusting down -c or
- increasing the roller pressure with the adjustment knob (preferred, as it actually increases max resistance)
until the displayed power matches. I guess it shows the need for a proper calibration routine again.

The brake was sold as "6% max" I believe, but unfortunately the actual limit may be ever lower depending on rider weight, wheel speed and roller pressure. Decreasing the grade factor -G may give a better (more dynamic) ride feeling.

WouterJD, 2021-01-26:
### TargetResistance

The TargetResistance appears to be quite flat.
The WheelSpeed is around 10 km/hr
The -G parameter is NOT specified
And Rouvy is used as an application

This causes the following: Rouvy transmits a slope-percentage which results in TargetPower. Since this power is quite high (compared to the athlete) the speed will be low and -to achieve the power- the resistance is high.

If the wheel-speed would be *2 the curve would be as shown below ([4]TargetResistanceVBA(s*2)/10), which follows the grade-curve:
[image]

WouterJD, 2021-01-26:
### Recommendation

FortiusAnt -p80 -G50

- -p Adjusts the displayed power so that it better matches the Garmin Powermeter
- -G adjusts the requested grade from Rouvy and enables a higher wheelspeed, making the trainer more flexible

WouterJD, 2021-01-26:
### Performance finetuning recommendation

> The brake was sold as "6% max" I believe, but unfortunately the actual limit may be even lower depending on rider weight, wheel speed and roller pressure. Decreasing the grade factor -G may give a better (more dynamic) ride feeling.

@switchabl I fully agree

@bikebeppe64 You can do a lot on performance finetuning and I hope this issue helps you to understand.
- Check the runoff time; the brake should not be too tight/too loose
- -G adjusts the requested grade from Rouvy and enables a higher wheelspeed, making the trainer more flexible
- -c to adjust the powercurve as described by @switchabl so that it better matches the Garmin Powermeter
- -p As last adjustment of the reported power

WouterJD, 2021-01-26:
This is the excel sheet for reference.
[FortiusANT JSON Analysis.xlsx](https://github.com/WouterJD/FortiusANT/files/5873077/FortiusANT.JSON.Analysis.xlsx)

BikeBeppe64, 2021-01-26:
Done a new test today and it's real more than acceptable for a big hill 
Rouvy : Teglio Climb 9/22%
-c13 virtual gear 50/50 real gear 50/15 (+/-) cadence 55~70

WouterJD, 2021-01-26:
Great news; happy rides!

closed 2021-02-04 by WouterJD

WouterJD, 2021-02-04:
Added to manual