WouterJD/FortiusANT #224: Trying to run T1901 / T1902 combo natively on OS X

goxmail, 2021-01-26:
Hi, looking for ideas on how to proceed with the above; I have a OS X 10.14.6 machine where I've installed Python and I believe all the pip requirements - when launching the application through python launcher, I'm getting the no numpy module found error, however I can successfully get to the GUI loading through the command line.

However, the Locate HW button is greyed out and I can't seem to figure out if the USB connection to the trainer or the ANT+ stick is even working.

I did install libusb through brew.

WouterJD, 2021-01-29:
Hi @goxmail 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

If you did not do special actions to bypass the numpy error, I assume FortiusAnt crashes. This should be visible in the console (black window).

Please provide some more info, like screen shots to explain in what state you are.

WouterJD, 2021-02-01:
No more info received

closed 2021-02-01 by WouterJD

goxmail, 2021-02-01:
Hi Wouter,
I'll gladly add myself to the list once I got things running ;)

The numpy error is just due to the python launcher I believe, if I check the requirements in terminal, they are all satisfied.

So I can load the GUI, I can enter settings, but I can't search for HW and if I set it to a specific Tacx model, that doesn't change anything..