WouterJD/FortiusANT #226: no Taxc trainer found

manurip, 2021-01-27:
Hello,
I have a 1932 Taxc Fortius multiplayer for several years (and don't want to delelte it) running on one Laptop with licence and want to install ur genius idea of FortiusAnt.
I read and try to follow ur procedure but when I run Fortius Antifier it don't found my trainer.
here under the file I've got.
I'm stuck and didn't know what to do further.

thanks for ur support.
kind regards.

Manu

[image]
[image]
[image]

memgard51, 2021-01-27:
Did you install new driver for the tacx USB connection?  (3.5.1 in the manual)

switchabl, 2021-01-27:
Easiest way is to grab the all-in-one driver package from https://github.com/switchabl/t19xx_usb (not yet in the manual).

manurip, 2021-01-27:
Hello, It's work with your file. Great Thanks to you. Sportively youManuEnvoyé depuis mon téléphone Huawei-------- Message original --------De : switchabl <notifications@github.com>Date : mer. 27 janv. 2021 à 17:52À : WouterJD/FortiusANT <FortiusANT@noreply.github.com>Cc : manurip <manu.ripet@skynet.be>, Author <author@noreply.github.com>Objet : Re: [WouterJD/FortiusANT] no Taxc trainer found (#226)
Easiest way is to grab the all-in-one driver package from https://github.com/switchabl/t19xx_usb (not yet in the manual).

manurip, 2021-01-27:
Hello, Yes I did. But it's working with the file from switchable. Thanks for your support. Have a nice day and take care of you Manu Envoyé depuis mon téléphone Huawei-------- Message original --------De : memgard51 <notifications@github.com>Date : mer. 27 janv. 2021 à 17:19À : WouterJD/FortiusANT <FortiusANT@noreply.github.com>Cc : manurip <manu.ripet@skynet.be>, Author <author@noreply.github.com>Objet : Re: [WouterJD/FortiusANT] no Taxc trainer found (#226)
Did you install new driver for the tacx USB connection?  (3.5.1 in the manual)

MeanHat, 2021-01-28:
Hi @WouterJD and @switchabl - I have a T1901 Flow brake and frame which is lacking a head unit. I have been unsuccessful in obtaining a new head unit so have resorted to building a controller based on @totalreverse's wiki (https://github.com/totalreverse/ttyT1941/wiki). This is obviously not picked up as a Tacx trainer/controller and has VendorID=0403 and ProductID = 6001. I have tried some simple changes to the code but without success. Do you have any suggestions how I might be able to use FortiusANT with this non-Tacx controller?

switchabl, 2021-01-28:
@MeanHat If you are referring to the serial interface that will unfortunately not work with the T1901. The T1901 uses an analog interface instead. Building a DIY head unit for that is possible in principle but would be a bit more involved. I can give you the details on the interface but I don't have a ready-made design.

Even for the T1941 (which has a serial interface) you would have to add a new Trainer class to FortiusANT. The protocol used for interfacing the Fortius directly is somewhat different than with the head unit.

MeanHat, 2021-01-28:
Hi @switchabl - thank you very much for the the info. I'd misinterpreted the wiki. I'd be grateful for the interface details but will continue to search for a second hand head unit.

WouterJD, 2021-01-29:
> ... has VendorID=0403 and ProductID = 6001 ...
> Do you have any suggestions how I might be able to use FortiusANT with this non-Tacx controller?

Apart from the controller itself and the interface to the magnetic brake....

If you have connected with a USB-interface and libusb drivers;
changing
```
idVendor_Tacx   = 0x3561
for hu in [hu1902, hu1904, hu1932, hu1942, hue6be_nfw]:
```
into
```
idVendor_Tacx   = 0x0403
for hu in [0x6001]:
```
should do the job (assuming you are mentioning hex-code) because
`dev = usb.core.find(idVendor=idVendor_Tacx, idProduct=hu)      # find trainer USB device`
filters on vendor and product.

Of course after detection of the USB-device, it should implement the protocol.
I hope this is what you mean?

MeanHat, 2021-01-30:
Thanks @WouterJD  - that's really helpful.

WouterJD, 2021-01-30:
Please keep me updated

WouterJD, 2021-02-01:
Original reason for this issue solved, I think

manurip, 2021-02-01:
Yes it's work wonderful. Thank you for your support. Manu Envoyé depuis mon téléphone Huawei-------- Message original --------De : Wouter Dubbeldam <notifications@github.com>Date : lun. 1 févr. 2021 à 11:55À : WouterJD/FortiusANT <FortiusANT@noreply.github.com>Cc : manurip <manu.ripet@skynet.be>, Author <author@noreply.github.com>Objet : Re: [WouterJD/FortiusANT] no Taxc trainer found (#226)
Original reason for this issue solved, I think

switchabl, 2021-02-05:
> Hi @switchabl - thank you very much for the the info. I'd misinterpreted the wiki. I'd be grateful for the interface details but will continue to search for a second hand head unit.

You can find the pin-out for the T1901 at https://github.com/totalreverse/ttyT1941/wiki ("Inside the T1902 and T1942"). You would need to:

- count the pulses on PIN1 for cadence (one pulse per pedal revolution)
- count the pulses on PIN6 for wheel speed (four pulses per roller revolution)
- detect the zero crossings on PIN 4 (+- 20V sine)
- output a PWM signal synchronized with the zero crossings on PIN3 to control the brake resistance

Most USB-capable microcontrollers would probably be able to do the job along with a little bit of analog circuitry. Emulating the old T1902 protocol would be the easiest choice. You could _probably_ get away with doing it on a Raspberry PI directly (with the GPIO pins) but some care will be needed to keep latencies down.

If you decide to go for it let me know. I do have more info on exact timings and so on and can look it up if you need it.

MeanHat, 2021-02-06:
Hi @switchabl  - that's really helpful. I think I'll avoid the RPi given the latency issues you mention and try another solution (Arduino). Not sure when I'll get to it but have made a note of your info and @totalreverse's wiki.

Thank you for your offer of further help as and when I get to the next stage.

WouterJD, 2021-02-17:
I think this issue can be closed

closed 2021-02-17 by WouterJD