WouterJD/FortiusANT #227: Bluetooth instead of ANT

memgard51, 2021-01-27:
Im one of all thouse who got the Cycplus-ant and having trouble.
Might get me new dongles it the problems are to big. 
The question: With the Bluetooth-support, can I go for bluetooth insted of ANT+ ? 
Is there any pros or cons with ANT+ vs Bluetooth?

Running Win10 and T2250 Tacx Flow bundle.

WouterJD, 2021-01-28:
@memgard51 Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Yes, Zwifting with Bluetooth works.

Because of the amount of requests, quality of the CYCPLUS dongles and low availability of other ANTdongles I have ordered a second BLE-dongle and will see what additional documentation is needed.

WouterJD, 2021-01-28:
See #61

closed 2021-01-28 by WouterJD

switchabl, 2021-01-28:
I would add that Zwift can use your phone as a BLE "dongle" if you install the Zwift Companion app, so one BLE dongle/built-in BLE may be enough (it does drain the battery quite fast though).

MeanHat, 2021-01-28:
Following on from @switchabl's previous comment I would add that BLE has the advantage on Android devices in that it will work with FE-C whereas ANT will not (FE only) - see https://support.zwift.com/en_us/ant-and-ble-overview-B1S1BVgr.

WouterJD, 2021-01-29:
I've bought a second BLEdongle and will do some tests; also translation it to the manual.
Second BLEdongle so I can continue using the PC-with-flatscreen (instead of streaming the smartphone)

memgard51, 2021-01-29:
> I've bought a second BLEdongle and will do some tests; also translation it to the manual.
> Second BLEdongle so I can continue using the PC-with-flatscreen (instead of streaming the smartphone)

What BLE-dongle did you get? Better to follow suit if I'm to get some as well ;-)

WouterJD, 2021-01-29:
See #61 now also containing BLE links

switchabl, 2021-01-30:
> Second BLEdongle so I can continue using the PC-with-flatscreen (instead of streaming the smartphone)

The "Zwift Companion" app (different from the Zwift phone app) forwards the BLE to the PC, so Zwift still runs on the PC with big screen. (Just as an info for people wanting to try it, for development it is definitely good to have a second BLE dongle)

WouterJD, 2021-01-30:
Sure true - for Zwift
I will check for development and documentation purpose.