WouterJD/FortiusANT #228: RGT: anyone using it?

AndrewB53, 2021-01-28:
Not so much an issue but a question to fellow users. Is anyone using FortiusANT to connect to RGT? I signed up earlier this week and have used it three times so far and it seems quite good. However I tied the Mont Ventoux route but found on a 4% slope I could hardly turn the pedals. Thinking it might be set up in a similar way to Routy I tried a different route today with a -G50 setting and that felt quite realistic. I would be interested in other users experiences.

WouterJD, 2021-01-28:
Nice! What is RGT?

AndrewB53, 2021-01-29:
Sorry,  I should  have included more details. It's an app similar to Zwift,
www.rgtcycling.com.  It has some free content and costs about half the
price of Zwift for the full package. There is a two week  trial period,
worth a look if you don't want to commit to Zwift.

FreekvdB, 2021-01-29:
Hi AndrewB53 and  WouterDJ, i am using RGT with fortiusANT for about 5 weeks now. I have an I-magic trainer and everything is working perfectly. Especially since the issue with the power curve on the i-magic has been sloved. I am very happy for the effort everyone has been put into it. A lot of thanks for that.
I am using FortiusAnt with no extra parameters and everything feels very realistic.
I really like the program RGT a lot. At this moment a have the piad (premium version).
It really gives me fun with indoor traing at this moment

AndrewB53, 2021-01-29:
@FreekvdB I like it too, it's amazing how watching a small avatar can improve indoor cycling. I think I'll try again without the power modification, perhaps I was a bit weak on Wednesday .

mikeherriman, 2021-01-30:
@AndrewB53 I have been using a combination of Zwift/RGT/Rouvy with FortiusANT since the beginning of the Pandemic.  I have my trainer connected to a Raspberry Pi 4 b with a 7" touchscreen on it.  With the latest work done with the issue #216, it is working much better.

As for RGT, yes the -G50 is recommended.  Zwift has a default value of 50%.  To replicate the "Zwift" feel, using a -G50 works very well for me.

I use both RGT and Zwift.  Sometimes my WIN10 machine doesn't load the Zwift interface...just sits and hangs.  RGT runs no problem with the WIN10 machine and my iPhone.  It would be much better if the developers at RGT could build a 100% user interface on a PC or Mobile device, instead of having to run both.

I do not use the subscription on RGT and am happy with the free rides that are available.  I have been riding the Dirty River course lately...I really enjoy that one.  If I am looking for an easier ride, then I go a few laps on the Borrega Springs.

Cheers!
Maybe see you on a ride.

AndrewB53, 2021-01-30:
@mikeherriman I had an instance where in RGT the mobile app and the desktop app lost connection and my ride ended earlier than expected. Other than that it woeks well. I run FortiusAnt on a desk top PC and RGT on a laptop so I can see the FA GUI. I'm currently using the 14 day free trial to try out all the features, I think I'll probably the subscription, in the UK it's £6.99 about half the price of Zwift. I'll keep a lookout for you, what's your favorite route?

closed 2021-02-04 by WouterJD