WouterJD/FortiusANT #229: Problem running 5.1 on windows 10 ( need to install lib_programname,bleno,express)

Kennyac, 2021-01-30:
I had problems running the latest version did not find it in the documentation

for python I had to install lib_programname

 _python -m pip install lib_programname_

I am using a bluetooth dongle ( using FortiusAnt.exe )

Needed to install _bleno_ and _express_

npm install bleno
npm install express

This got me up and running

WouterJD, 2021-01-30:
Hi @Kennyac did you follow steps in https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md?

What OS do you use?

Kennyac, 2021-01-30:
Windows 10Pro 20H2

I had version 5.0 running on my laptop

When I ran version 5.1 I thought my firewall was blocking the connection to node as it could not connect to server 

I ran the server manually and got the error about bleno and express

I installed the missing node packages then it worked

WouterJD, 2021-02-04:
added to manual

closed 2021-02-04 by WouterJD