WouterJD/FortiusANT #230: Status LED's for the  headless Raspberry

decodeais, 2021-01-30:
I am new to Git an Python but i added successful a few Status LED's  for the RaspberryPi to your Software. Now it is possible to run it headless. I made a branch and would like to commit it.
Thanks!

WouterJD, 2021-02-01:
Hi @decodeais 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Thanks for the effort, nice work!
I will look into the PR

WouterJD, 2021-02-22:
I have ordered an RPI and will implement. 

Could you share some photo's how you completed your hardware; how you put your leds etcetera...
I would appreciate that!

decodeais, 2021-02-23:
Hello, I have took two photos
Quick and dirty, but fine working solution.
Maybe you know it allready:
I soldered the LED direct to the female connector strip. The resistors
have 470 Ohms.
If you look to the bottom of the LED, one side is flatted and the wire
is a little shorter. This side has to be connected to GND.
The button close contact when pressed.

WouterJD, 2021-02-25:
Hi @decodeais nice that you have created the foto's.
You can share them by pasting them in a post (area saying "Attached files by drag&drop...") on github (webversion)

WouterJD, 2021-03-04:
Hi @decodeais we have multiple issues on LED's and DIsplays, I close this one.
Thanks for the great work.
FortiusAnt 6.0 is in the pipeline :-)

closed 2021-03-04 by WouterJD

switchabl, 2021-03-04:
> FortiusAnt 6.0 is in the pipeline :-)

@WouterJD Can you include #252 while you are at it?

WouterJD, 2021-03-04:
Hi @switchabl will look at it shortly and sure integrate! 
Check with me if it takes too long