WouterJD/FortiusANT #233: Start Up on Mac

danlewis97, 2021-01-31:
Hi,

I have a Tacx Vortex and I am using Mac. I have followed Manual until I got to the FortiusAnt Start Up section. I cannot code this to start. I cannot open any Start Up file due to them being .bat and manually typing them in Terminal doesn't appear to work saying 'command not found'.

Any ideas Mac users?

WouterJD, 2021-02-01:
HI @danlewis97 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

WouterJD, 2021-02-01:
Who provides a good Mac-quote how to start a program on the mac; I will add to the manual in a next version.

marcoveeneman, 2021-02-01:
I'm using a mac and 'just' start it as follows on the command line.
`python3 FortiusAnt.py -a -b -g`

Note that you need to navigate on the command line to the `<your FortiusAnt directory>/pythoncode` location

WouterJD, 2021-02-04:
Some notes added to manual

closed 2021-02-04 by WouterJD