WouterJD/FortiusANT #234: Is it possible to calculate the rolling resistance from the RunOff time

GewoonGijs, 2021-02-01:
Hi All,

I was wandering. Since I don't have a power meter, is there another way to calibrate my setup. I assume it should be possible to calculate the correct -c<param> from the timing of the RunOff? Am I correct? And does anyone made this calculation earlier?

WouterJD, 2021-02-01:
I think this is difficult.

You tune your "knob" so that the RunOff time is 7seconds and therefore your input variables has become a constant.
and then -c would be the same for everybody as well.

I think you tune your "knob" so that Runoff is 7 seconds.
Then compare power-meter with FortiusAnt calculation and adjust.

Wandering how others will look at this.

switchabl, 2021-02-01:
In principle, yes. I more or less know how to do it but I still need and to do some testing, some maths and maybe a couple of tweaks to the code. The plan is to have an automatic calibration procedure like TTS does. Unfortunately I didn't have much time to work on it lately.

GewoonGijs, 2021-02-01:
I tried to do some math's myself, but could not get my head around it. What I could think of is:

Loss of kinetic Energy is absorbed as heat by the combination of the brake and the rolling resistance.

Since the power setting is pinned at 100 W it should be 100 * 7,2 s = 720 J. Am I right? But how to calculate the kinetic Energy?

From the other way around, Power at a speed of 10,56 m/s (=38 km/h) = Fres * v
Fres = m * a, with m =85 kg and a,avg = 1.43 m/s^2 (38 km/h -> 1km/h in 7.2 sec)
Fres, avg = 121,33 N, but can we assume Fres to be constant during the SpinOff. 

Maybe you can share some thoughts, I will happily do the coding :)

switchabl, 2021-02-02:
@GewoonGijs First let my point out that the 100W power target already includes the specified rolling resistance (-c). Now in principle you can use this to check the calibration: if the spin-down time is consistent with a 100W resistance, your calibration is fine, otherwise adjust either -c or the trainer knob. This is similar to what @WouterJD has suggested (but see caveats at the end of the post).

Before we get to the kinetic energy, let us first look at the (effective) inertial mass. This is determined by the moment of inertia of 1) the flywheel and 2) the rear wheel.

1) There is some info in the old Tacx catalogues:

- flywheel mass is supposed to be 2kg (https://issuu.com/eurobike/docs/t0300_tacx_catalogue_2014_en, page 32)
- flywheel effective inertia is 11.81kg (http://engbike.com/catalogos/Catalogo-TACX_2015.pdf, page 42)

I couldn't weigh the flywheel directly, but I measured the dimensions (width = 33mm, diameter = 100mm) and 2kg would put the density at 7700 kg/m^3 which is about what you would expect for stainless steel. Furthermore, the cylinder that makes contact with the wheel has a diameter of 29mm. I get an effective mass of 11.9kg from that. So that all seems to be reasonably accurate (and the 1.65kg @totalreverse has listed at https://github.com/totalreverse/ttyT1941/wiki is potentially wrong).

2) The contribution from the rear wheel is relatively small by comparison. The formula `m_eff = m_rim + m_spokes / 2` is a reasonably good approximation (`m_rim` includes all the mass at the rim: the rim itself, the nipples, the inner tube and the tyre). This will vary from bike to bike but something like 1kg should be a reasonably assumption for a road bike.

So I will assume an effective mass `m_eff` of 12.8kg overall (which is a lot lower than on the road and a major reason the ride feel on the Flow is not all that realistic).

At a given wheel speed `v` we can now calculate the kinetic energy as usual:
```
E_kin = 1/2 m_eff v^2
```
What is the time it takes for the wheel to spin down from a speed `v_1` to `v_2`? The corresponding energy loss is `ΔE = 1/2 m_eff (v_1^2 - v_2^2)`. Now assuming that the power `P` dissipated by the brake and the rolling resistance is constant (again, caveats below), the spin down time is simply
```
Δt = ΔE / P = 1/2 m_eff (v1^2 - v_2^2) / P .
```
If I plug in `v_1 = 38 km/h`, `v_2 = 1 km/h` and `P = 100W`, I get a target spin-down time of `Δt = 7.1s`.

On the face of it that seems fine. Unfortunately it will generally not give us a very accurate calibration (or if so, then only by accident). The main problem is that the brake will not actually be able to keep the power constant at 100W:
- The power dissipated by the brake is highly speed-dependent so FortiusANT will have to keep adjusting the resistance setting throughout the spin-down. Unfortunately there is some delay to that.
- The resistance range is limited. At 38km/h the power will be a _lot_ higher than 100W even at the lowest setting. And below 10km/h the power will not reach 100W even at the highest setting.

I think the most sensible path will be to keep the resistance setting fixed during the spin-down (similar to what the -r parameter does for normal riding). In principle we could try to derive the corresponding target times from our brake force model. But a simpler and more accurate approach will be for me to get on a bike and collect some power meter data specifically for this.

I also want to experiment with the parameters some more. E.g. I suspect that going down all the way to 1km/h is not a great idea (chances are very low speeds will be less consistent).

GewoonGijs, 2021-02-04:
@switchabl thanks for your explanation.

From your explanation, I tried to do some calculations. If i make a table in Excel for the actual 
Power and Force based on a rolling resistance of 15, I see that Power is more or less linear decreasing from 270 W at 38 km/h to 106 W at 17kmh. After that the algorithm starts increasing the resistance and Power will be just above 100W untill speed has dropped to 9 kmh, and then it goes down quite rapidly. If i look at the Force (Power/v) it starts at 26 N drops gradually to 23 N ar 17kmh, then goes up to 39 N at 9kmh and drops again to 19N at 1 kmh. 

Average Force is something around 27N and average power is something like 139 W. 

But if i calculate the average deceleration is is 1,44 m/s^2 which gives an estimation of 18,5N (if i may use F=m*a with the m_eff as mass, or should we calculate wit Torque and Inertia?).

If i estimate the proper t using the average power, i get 713/139 = 5.1 s. And if i take the mean Force i get a deceleration of 2,10 m/s^2 resulting in an estimation for t = 4,9 s. 

My conclusion is that for t = 7 s  Froll is 15N is overestimated, am I right? This will also lead to an overestimation in ActualPower, right?

GewoonGijs, 2021-02-04:
[RunDown analysis.xlsx](https://github.com/WouterJD/FortiusANT/files/5928878/RunDown.analysis.xlsx)
 
See my attached Excel file, I guess turning op the Knob to get around 5 s of Rundown seems to bet a much better guess for a rolling resistance of 15 N. Or, I should use -c to set a lower rolldown resistance of 3 N. But as you stated before, it would even be wiser to use only part of the rundow, where either Force is more or less constant, or Power. I guess the easiest way is to but resistance at the lowest setting and keep it there, and choose the velocity between 37 kmh and Critical speed (~ 17kmh). Force is quite flat for the whole range then. Do you agree? 

Next time I am on the trainer, I will calibrate for a Rundown time of 5 s, see what happens. At least my estimated FTP will decrease :)

switchabl, 2021-02-05:
> From your explanation, I tried to do some calculations. If i make a table in Excel for the actual
> Power and Force based on a rolling resistance of 15, I see that Power is more or less linear [...]

Yes, that seems about right.

> (if i may use F=m*a with the m_eff as mass, or should we calculate wit Torque and Inertia?).

For our purposes, working with (physical) torque or (effective) linear force is essentially equivalent. I usually stick to the latter because it corresponds naturally to the outdoor situation we are trying to simulate. Also the wheel and roller diameters cancel out. Think of it as a convenient layer of abstraction.

> If i estimate the proper t using the average power, i get 713/139 = 5.1 s. And if i take the mean Force i get a deceleration of 2,10 m/s^2 resulting in an estimation for t = 4,9 s.

Both approaches are somewhat problematic. The formula

<img src="https://render.githubusercontent.com/render/math?math=\Delta t = \frac{m_\text{eff}}{F} \Delta v">
is valid for constant force (I assume this is what you used).

The other one I gave earlier on the other hand really only works for constant _power_:

<img src="https://render.githubusercontent.com/render/math?math=\Delta t = \frac{1}{2} \frac{m_\text{eff} (v_1^2 - v_2^2)}{P}">

In general, you will have to solve this differential equation instead

<img src="https://render.githubusercontent.com/render/math?math=v' = -\frac{F(v)}{m_\text{eff}}">

In our case we end up with an integral for the spin-down time

<img src="https://render.githubusercontent.com/render/math?math=\Delta t = m \int_{v_2}^{v_1} \frac{dv}{F(v)}">

In your dataset, the force does not actually vary that much over most of the range (lowest speeds don't contribute much), so the difference is great (still ~5s). But something to keep in mind.

With that out of the way: I just did a quick real-world test. I did a few minutes warm-up, adjusted the knob so the displayed power matches my power meter (Garmin Vector 2 dual-side). Then I did the "runoff" with default parameters. The result was 6.4-6.6s!

Why the difference? Well, a number of ideas:
- We have assumed that the adjustment is instantaneous, but there is in fact some significant latency. So the actual force applied will be lower. (This _may_ also depend a little on the pedal stroke analysis option. I had it turned on.)
- It is not clear how accurate our force curve is at low speeds (there was not a lot of good low-speed data available).
- There may be systematic errors due to the way the wheel speed is measured and spin-down time is calculated.

In the end I think this purely theoretical approach will not be good enough for calibration. I'm not saying that it isn't useful. Having this discussion and writing everything down explicitly has definitely given me a better idea on how to proceed. But there is just no way around collecting more power meter data. And then do some more modeling based on that. Maybe I can find time to do this over the weekend.

In the meantime I suggest using 6.5s as a target.

> I guess the easiest way is to but resistance at the lowest setting and keep it there, and choose the velocity between 37 kmh and Critical speed (~ 17kmh). Force is quite flat for the whole range then. Do you agree?

I would indeed suggest to stick to the lowest resistance setting. There are a number of advantages to that (lower spin-down time means higher accuracy; don't need to worry about resistance adjustments). Not sure about restricting the range that much. I would prefer characterising the spin-down dynamics properly and make the spin-down as long as possible. Then again asking users to put out 300W for calibration seems somewhat excessive, too. We will see what is feasable.

GewoonGijs, 2021-02-05:
I changed my setting to 6 sec. and a rolling resistance of 8N. That seemed to give some quite realsitic behaviour (in terms of speed on a flat stretch) in Zwift.

And indeed I did the assumption that F and P are constant, which isn't true off course :)

switchabl, 2021-02-05:
> 
> I changed my setting to 6 sec. and a rolling resistance of 8N. That seemed to give some quite realsitic behaviour (in terms of speed on a flat stretch) in Zwift.

In that case you are almost certainly underestimating the power. Zwift is notorious for assuming a very low drag coefficient, so unless you ride a very aero-optimized bike and position outdoors, speeds in Zwift should be significantly faster.

Like I said, my preliminary test indicates that the target should be more like 6.5s (with 15N default setting). I trust my power meter a lot more than any of the calculations we have done here.

GewoonGijs, 2021-02-06:
Another reason to get a higher power at your pedals is the loss in the drivetrain. Since we don’t pedal, this will not be measured during Runoff. What I read is around 3% for a new well lubricated dura ace chain. My chain is definitely of less quality and probably will give a loss that’s higher. Maybe around 5-7%.

But anyway, I guess your pedals are definitely more accurate then the calculations.

switchabl, 2021-02-06:
Well, true, drive-train losses might shift things by a few watts (and power meter accuracy is another +/-2 %). From a simulation standpoint we should obviously use power at the wheel, not at the pedals. But in practice, if you look for better than 10% accuracy you really need to get a direct-drive trainer (or a power meter). Especially with the magnetic wheel-on brakes there are just too many error sources. The point of the calibration will be to make sure it is a) consistent and b) not 20% or 50% off.

And to put this into perspective: the model we are using (both in FortiusANT and for these calculation) is based on three data-sets. One was collected with a hub-based Powertap, one with a single-sided Stages (the third I don't remember). All had some data quality issues. I came up with a simple two-parameter (+ rolling resistance) model based on those. It seems to work surprisingly well but it is a far cry from the actual physics of an eddy-current brake. In particular, should be no expectation that it correctly predicts dynamic behaviour (i.e. fast speed/resistance changes like during the spin-down).

WouterJD, 2021-02-22:
I suggest to reopen the issue when information is exchanged on the topic.

closed 2021-02-22 by WouterJD