WouterJD/FortiusANT #235: OSX Issues with 1901/1902 combo

goxmail, 2021-02-02:
Hi @WouterJD ,
I'll gladly add myself to the list once I got things running ;)

The numpy error is just due to the python launcher I believe, if I check the requirements in terminal, they are all satisfied.

So I can load the GUI, I can enter settings, but I can't search for HW and if I set it to a specific Tacx model, that doesn't change anything..

_Originally posted by @goxmail in https://github.com/WouterJD/FortiusANT/issues/224#issuecomment-770792703_

goxmail, 2021-02-02:
[image]
[image]
[image]

switchabl, 2021-02-02:
I don't believe `FortiusAntGui.py` is meant to be run directly. Have you tried running `FortiusAnt.py` instead?

goxmail, 2021-02-02:
Hm, haven't  thought about that...
That gives me:
---------------------------------------------------------------
11:57:50,031: FortiusANT started
11:57:50,031: -a
11:57:50,031: -A
11:57:50,031: -c False
11:57:50,032: -C 0/0
11:57:50,032: -d 127 (0b1111111)
11:57:50,032: -D None
11:57:50,032: -g
11:57:50,032: -H 0
11:57:50,032: -p 1.0
11:57:50,032: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
11:57:50,032: -t Magneticbrake
11:57:50,032: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
11:57:50,032: ------------------
11:57:50,623: FortiusAnt GUI started in child-process

..and then python crashes.

switchabl, 2021-02-02:
Any other output (some exception maybe) or log files created? Can you post the output of `python --version` and `python -m pip --list` (for the same python executable you are using to run FortiusAnt.py)?

goxmail, 2021-02-02:
thanks for your assistance!

- The log file created in the pythoncode folder just shows the child-process message; anywhere else I should look for that ?
- python version gives 3.7.9. (which I have installed as the osx included 2.7 wouldn't work from what I gather)
- upon python -m pip --list I get a:

Usage:   
  /usr/local/bin/python -m pip <command> [options]

no such option: --list

So I guess the issue could be here somewhere....

goxmail, 2021-02-02:
further info:

Hannas-MacBook:pythoncode hanna$ pip install -r requirements.txt
Requirement already satisfied: lib_programname in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (1.1.8)
Requirement already satisfied: numpy!=1.19.4 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (1.19.5)
Requirement already satisfied: pyinstaller in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (4.2)
Requirement already satisfied: pyusb in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (1.1.1)
Requirement already satisfied: requests in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (2.25.1)
Requirement already satisfied: wxPython in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (4.1.1)
Requirement already satisfied: wheel in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.36.2)
Requirement already satisfied: cli-exit-tools in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from lib_programname->-r requirements.txt (line 1)) (1.1.8)
Requirement already satisfied: click in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from lib_programname->-r requirements.txt (line 1)) (7.1.2)
Requirement already satisfied: importlib-metadata in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from pyinstaller->-r requirements.txt (line 3)) (3.4.0)
Requirement already satisfied: setuptools in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from pyinstaller->-r requirements.txt (line 3)) (47.1.0)
Requirement already satisfied: altgraph in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from pyinstaller->-r requirements.txt (line 3)) (0.17)
Requirement already satisfied: pyinstaller-hooks-contrib>=2020.6 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from pyinstaller->-r requirements.txt (line 3)) (2020.11)
Requirement already satisfied: macholib>=1.8 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from pyinstaller->-r requirements.txt (line 3)) (1.14)
Requirement already satisfied: idna<3,>=2.5 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from requests->-r requirements.txt (line 5)) (2.10)
Requirement already satisfied: chardet<5,>=3.0.2 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from requests->-r requirements.txt (line 5)) (4.0.0)
Requirement already satisfied: certifi>=2017.4.17 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from requests->-r requirements.txt (line 5)) (2020.12.5)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from requests->-r requirements.txt (line 5)) (1.26.2)
Requirement already satisfied: six in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from wxPython->-r requirements.txt (line 6)) (1.15.0)
Requirement already satisfied: pillow in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from wxPython->-r requirements.txt (line 6)) (8.1.0)
Requirement already satisfied: zipp>=0.5 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from importlib-metadata->pyinstaller->-r requirements.txt (line 3)) (3.4.0)
Requirement already satisfied: typing-extensions>=3.6.4 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from importlib-metadata->pyinstaller->-r requirements.txt (line 3)) (3.7.4.3)
WARNING: You are using pip version 21.0; however, version 21.0.1 is available.
You should consider upgrading via the '/Library/Frameworks/Python.framework/Versions/3.7/bin/python3 -m pip install --upgrade pip' command.
Hannas-MacBook:pythoncode hanna$ python3 -m pip install --upgrade pip
Requirement already satisfied: pip in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (21.0)
Collecting pip
  Downloading pip-21.0.1-py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 1.3 MB/s 
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0
    Uninstalling pip-21.0:
      Successfully uninstalled pip-21.0
Successfully installed pip-21.0.1

goxmail, 2021-02-02:
And here' what the python crash report says:

Crashed Thread:        0  Dispatch queue: com.apple.root.user-interactive-qos

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000110
Exception Note:        EXC_CORPSE_NOTIFY

switchabl, 2021-02-02:
So sorry, that should have been `python -m pip list` (with the --).

The segmentation fault suggests this may be something deeper than a dependency issue though. Some more OS X-specific knowledge may be needed here (which I unfortunately don't have).

goxmail, 2021-02-02:
ah ;) Here we go:
Package                   Version
------------------------- ---------
altgraph                  0.17
certifi                   2020.12.5
chardet                   4.0.0
cli-exit-tools            1.1.8
click                     7.1.2
idna                      2.10
importlib-metadata        3.4.0
lib-programname           1.1.8
macholib                  1.14
numpy                     1.19.5
Pillow                    8.1.0
pip                       21.0.1
pygame                    2.0.1
pyinstaller               4.2
pyinstaller-hooks-contrib 2020.11
pyusb                     1.1.1
requests                  2.25.1
setuptools                47.1.0
six                       1.15.0
typing-extensions         3.7.4.3
urllib3                   1.26.2
wheel                     0.36.2
wxPython                  4.1.1
zipp                      3.4.0

No worries, I appreciate the input - if anyone else has ideas, happy to hear them...

switchabl, 2021-02-02:
Yes, that all looks fine...

Things I can think of:
- There seem to be issues with Python multiprocessing (which is used by FortiusANT as well) on Mac, supposedly fixed with Python 3.8 (https://bugs.python.org/issue33725). So maybe switching to 3.8 does the trick (should be in homebrew I guess?)
- There is a bug in FortiusANT that can cause crashes on Linux (#216, #225), maybe the same problem exists on OS X as well. The fix is not yet in the master branch but you can try the WIP fix (`git fetch && git checkout #216-GUI-crash-on-raspberry`, switch back with `git checkout master`).

I think @marcoveeneman uses OS X, maybe he can share some details on his python environment.

goxmail, 2021-02-02:
- I wasn't sure about the best Python version due to the note in the manual 
"Noted by @msjnaessens: “If you installed python 3.9, you cannot install numpy. I chose to install
python version 3.7.9 instead.”"
But 3.8 could definitely be worth a shot

- I'll do my noob best to get that fix working or will have to wait for it to be merged to master, respectively ;)

And, yes, from what I read so far in the issues, @marcoveeneman appears to be the only one having actual hands on OS X experience, so if he has any ideas, happy to hear them / try them out.

Again, many thanks!

harrykl, 2021-02-02:
I can start the python script on my mac und the UI starts. I become the error
bluetooth interface cannot be opened; [Errno 2] No such file or directory: 'node'

Any Idea?

tjerkh, 2021-02-02:
I usually run FortiusAnt on windows but I used MacOS in the past. To get the latest version running I had to install some new required libraries and upgrade wxPython, but can start the FortiusAnt GUI without any problems. (I did not test more.)
I'm using Python 3.9.0 with numpy 1.20.0 (can't remember if I had any issues installing this).

harrykl, 2021-02-03:
Thanks tjerkh. FortiusAnt run the same at my Mac. The tacx Trainer (1932) is found and Bluetooth ist connected (standard Bluetooth an Macbook). No data is shown, if I Try with my Flow with PC-Upgrade. If I tap stop the error (see above) occurs.
Greetings Harry

WouterJD, 2021-02-03:
Hi @harrykl 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

marcoveeneman, 2021-02-04:
Hi Guys, i'll share details of my setup here:

Python version:
```
Marcos-MacBook-Pro:~ marcoveeneman$ python3 --version
Python 3.7.9
```

Pip packages: (Note that i have a dirty developer environment with extra packages unrelated to FortiusANT)
```
Marcos-MacBook-Pro:~ marcoveeneman$ python3 -m pip list
Package                                           Version
------------------------------------------------- ---------
alabaster                                         0.7.12
Babel                                             2.8.0
beautifulsoup4                                    4.6.0
bleak                                             0.9.1
certifi                                           2020.6.20
chardet                                           3.0.4
docutils                                          0.16
gevent                                            20.9.0
google-services-api                               0.0.6
greenlet                                          0.4.17
idna                                              2.10
imagesize                                         1.2.0
Jinja2                                            2.11.2
keyboard                                          0.13.5
lxml                                              4.6.1
MarkupSafe                                        1.1.1
numpy                                             1.19.2
packaging                                         20.4
Pillow                                            8.0.1
pip                                               20.1.1
prettytable                                       1.0.1
py                                                1.9.0
PyBluez                                           0.23
pycrypto                                          2.6.1
pygame                                            1.9.6
Pygments                                          2.7.2
pynput                                            1.7.1
pyobjc                                            6.2.2
pyobjc-core                                       6.2.2
pyobjc-framework-Accounts                         6.2.2
pyobjc-framework-AddressBook                      6.2.2
pyobjc-framework-AdSupport                        6.2.2
pyobjc-framework-AppleScriptKit                   6.2.2
pyobjc-framework-AppleScriptObjC                  6.2.2
pyobjc-framework-ApplicationServices              6.2.2
pyobjc-framework-AuthenticationServices           6.2.2
pyobjc-framework-AutomaticAssessmentConfiguration 6.2.2
pyobjc-framework-Automator                        6.2.2
pyobjc-framework-AVFoundation                     6.2.2
pyobjc-framework-AVKit                            6.2.2
pyobjc-framework-BusinessChat                     6.2.2
pyobjc-framework-CalendarStore                    6.2.2
pyobjc-framework-CFNetwork                        6.2.2
pyobjc-framework-CloudKit                         6.2.2
pyobjc-framework-Cocoa                            6.2.2
pyobjc-framework-Collaboration                    6.2.2
pyobjc-framework-ColorSync                        6.2.2
pyobjc-framework-Contacts                         6.2.2
pyobjc-framework-ContactsUI                       6.2.2
pyobjc-framework-CoreAudio                        6.2.2
pyobjc-framework-CoreAudioKit                     6.2.2
pyobjc-framework-CoreBluetooth                    6.2.2
pyobjc-framework-CoreData                         6.2.2
pyobjc-framework-CoreHaptics                      6.2.2
pyobjc-framework-CoreLocation                     6.2.2
pyobjc-framework-CoreMedia                        6.2.2
pyobjc-framework-CoreMediaIO                      6.2.2
[... 92 lines cut ...]
six                                               1.15.0
snowballstemmer                                   2.0.0
Sphinx                                            3.4.3
sphinx-rtd-theme                                  0.5.0
sphinxcontrib-applehelp                           1.0.2
sphinxcontrib-devhelp                             1.0.2
sphinxcontrib-htmlhelp                            1.0.3
sphinxcontrib-jsmath                              1.0.1
sphinxcontrib-qthelp                              1.0.3
sphinxcontrib-serializinghtml                     1.1.4
urllib3                                           1.25.10
wcwidth                                           0.2.5
wheel                                             0.35.1
wxPython                                          4.1.0
zope.event                                        4.5.0
zope.interface                                    5.1.2
```

OS version:
```
Marcos-MacBook-Pro:~ marcoveeneman$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.15.7
BuildVersion:	19H2
```

marcoveeneman, 2021-02-04:
> I can start the python script on my mac und the UI starts. I become the error
> bluetooth interface cannot be opened; [Errno 2] No such file or directory: 'node'
> 
> Any Idea?

Did you download the complete FortiusANT package? The node folder should be located in \<FortiusANT directory\>/node, start FortiusANT.py from the \<FortiusANT directory\>pythoncode directory like: `python3 FortiusANT.py <your arguments>`, i start it with `python3 FortiusAnt.py -a -b -g` for example.

harrykl, 2021-02-04:
I downloaded the complete FortiusAnt Package as zip-File and I start it with `python3 FortiusAnt.py -a -b -g`. Everything looks fine but no data is recevied in FortiusAnt. Then I stop in the gui and the error occurs.

JoostBart, 2021-02-05:
More or less same issue here. See my description under #241 
MacOS Big Sur
Python 3.8
No data seems to be transferred to GUI.

tjerkh, 2021-02-05:
I tested my setup with the trainer and it looks I have the same problem. Everything looks fine but the GUI does not update. Interesting thing is that when I move the window from one to the other screen (I have multiple screens connected) the GUI is updated once and I can see the speed, power and cadence. Maybe related to #216 ?

harrykl, 2021-02-07:
I downloaded the latest Version an now the data ist shown in FortiusAnt. But the onboard BLE on Macbook Air don't work. Always the node error

goxmail, 2021-02-09:
Just wanted to update that, with the latest version, starting up is no issue and does work on my OS X machine with Python 3.7.9.!

Having some issues with the rundown as well as with getting found by Zwift / RGT, but the basics seem to work, so (at least from my perspective), this issue could be closed ;)

harrykl, 2021-02-15:
With Zwift it works for me too. 
After installing node to the system, you have to set it up for FortiusANT

$ cd ~/FortiusANT/node/
$ npm install

solved my Problem. From my perspective this issue could be closed

closed 2021-02-17 by WouterJD