WouterJD/FortiusANT #236: Can't connect the device

meiki1, 2021-02-02:
I have tried to connect the role with Zwift.
I can control the role manually in your program (Fortius ANT), but when I want to connect it in Zwift, it does not find anything. No matter what I select. I have read through the other problem cases but came up with no solution. Have also specially fetched the Garmin ANT stick. Have it now tried for several days / weeks, unfortunately unsuccessful. Can you help?
Thanks for your support

[image]

[image]
[image]
[image]

TacxBiker, 2021-02-03:
Hi,

Do you have only 1 ANT dongle? If you want to use Fortius ANT on 1 computer using ANT you will need 2 ANT dongles.

WouterJD, 2021-02-03:
Hi @meiki1 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

meiki1, 2021-02-03:
That was the clue I had missed. I thought the Tacx Fortius is ANT source 1 and the stick the second source. With a second stick it helped thanks.
Thanks also for the cool program and support. btw I am from Germany near Cologne

closed 2021-02-03 by meiki1