WouterJD/FortiusANT #237: Raspi4 setup without display

WouterJD, 2021-02-03:
[image]
Hello,
thank you for your nice Software. I am using it with a headless Raspi4 setup. Since I added the LED's I don't need the Display anymore. Now my wife is able to startup alone by switching on the powersupply.  
We use Tacx Flow T1902 and the Fortius T1932.
Greetings from Dassow in Germany
Jens-Peter

_Originally posted by @decodeais in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-770758893_

decodeais, 2021-02-07:
I am ready with my documentation for RasperryPi installation. 
It is possible to install it complete  headless. I checked it with Raspi4 and 3. 

The GUI is still freezing but it does not matter if you mount these Status LED's.

You start/stop it with Power On or Off. If you have the Raspi4 you can mount a Pushbutton for this.
Remote maintenance can be done by VNC or SSH.
The Parameter can be adjusted by the  shell script FortiusAntHeadless.sh

The Status LED's show :
USB connection/data transfer
Tacx  presence off ANT+ Dongle 
Candence from the REED-Sensor

Tested with Tacx Head T1932, Trainer T1901 and a Garmin Ant Dongle 1009

LED's should work for other setups with USB-Trainer and And ANT+ communication too. 
You can find the full description under this link:
https://github.com/decodeais/FortiusANT/blob/Raspi-Status-LED/doc/HeadlessRaspi.md

povey76, 2021-02-08:
Thanks @decodeais for the detailed information and guide, I have installed all the data onto the pi sd card and then installed into the pi, and updating via ssh (from a mac). Just waiting now for "pip3 install -r ~/FortiusANT-Raspi-Status-LED/pythoncode/requirements.txt " as it's taking a while to move on! I don't have any LED's connected (or a pushbutton) - I have a Pi 3B, which did you connect the led's to please - and will it work ok without the LED's? Ihave a spare tv in the pain cave so will connect it to that to see what is working! 

Great detail by the way - made it a lot simpler to follow!

decodeais, 2021-02-12:
Does it work ?  It should work without LED's ( dont forget the 470Ohm Resistor in Row to the LED) but the Button only works with the Raspi4.

WouterJD, 2021-02-22:
Hi @decodeais . I ordered an RPI4 today and will build a T1932 Fortius with it.
Curious what it will bring!

decodeais, 2021-02-23:
I think the most important thing for RPI4 would be to get the internal BLE working. It would be an easy way to get FortiusAnt working without all the different possible hardware configurations. You will see that there is still something wrong with the GUI but the  calibration part is working.

WouterJD, 2021-02-23:
Thanks, I bet I will need your support these days.
I will start with the guide you wrote. First parts have been delivered.

hasise, 2021-02-23:
Interesting to know exactly what is required and the installation steps. @decodeais manual is a very good start, not sure
about the USB port settings for a "normal" installation and the comment "The GUI is still freezing" is a bit scary since I'm
planning to use the UI. Maybe my Pi 3 is to slow but I will give it a try if it works ok on a Pi 4.

MeanHat, 2021-02-24:
Hi @WouterJD - In case it is of interest I have used a headless Raspberry Pi3B+ with an Adafruit Mini PiTFT 1.3" - 240x240 TFT (c£13) working with a Tacx Fortius and using Bluetooth. This has two buttons (one to start FortiusANT) and the other to stop and shutdown the Pi. The display tracks the start-up progress (note that the blue in the attached photos is actually dull grey in real-life) with text and colour changes as each step is completed. It is a better alternative to the power-off shim I used for a different FortiusANT implementation. Finishing off final testing. (hence the HDMI cable etc.).

[image]
[image]

TacxBiker, 2021-02-24:
@MeanHat Thats very nice. I'm trying to setup something like this as well. Im using a grafical screen. Besides the text I will also add the virtual gears.

As I'm new to python; how is the info of Fortius ANT passed to the display? I can write text to the display but only from its own script.
I know how i can use variables, but i don't know hoe to fill variables from another script

i use a screen like this:
https://github.com/SrBrahma/RPi-12864-LCD-ST7920-lib

regards, Mathijs

MeanHat, 2021-02-24:
Hi @TacxBiker / Mathijs - good luck with the screen you are using also. Looks like a nice solution.

I did not want to alter @WouterJD's core code so I started using data from FortiusAnt.py with subprocess but this did not provide the data I really wanted so I changed to pulling data from the logfile (-d1 argument in the command line provided the logfile data I wanted). I'm fairly new to python too so probably not very elegant. Does that help? If you would like me share anything please let me know.

Regards - Keith

MeanHat, 2021-02-28:
> Hi @TacxBiker / Mathijs - good luck with the screen you are using also. Looks like a nice solution.
>
> I did not want to alter @WouterJD's core code so I started using data from FortiusAnt.py with [...]

Hi @TacxBiker - in case it is of use to you I have uploaded the code to https://github.com/MeanHat/FortiusANT_controller

WouterJD, 2021-03-04:
@MeanHat it's too nice not to integrate; I will look into it!

WouterJD, 2021-03-04:
@MeanHat @TacxBiker I have ordered a 1.3" tft from AliEx.
So that's next step.

I will write documentation tonight and post for verification to complete FortiusAnt 6.0 (first  version with Raspberry integrated). Next version will then be with TFT integrated. 
Then users can chose: HDMI, LED or SPI :-)

Thanks for enthusiastic participation!

WouterJD, 2021-03-04:
--> https://github.com/WouterJD/FortiusANT/wiki/Raspberry-page-of-glory

TacxBiker, 2021-03-04:
@WouterJD 

Thank you for all your hard work. I looking forward to the new version

MeanHat, 2021-03-04:
@WouterJD - likewise. Look forward to v6 and love RPi page of glory

MeanHat, 2021-03-04:
> @MeanHat @TacxBiker I have ordered a 1.3" tft from AliEx.
> So that's next step.
>
> I will write documentation tonight and post for verification to complete FortiusAnt 6.0 (first  version with Raspberry [...]

Hi @WouterJD - sounds great. 
Unfortunately my miniTFT has developed a fault so will try to get it replaced. 
Also, when using Zwift the Bluetooth cuts out after about 10 mins. I think the problem is with my code so am troubleshooting. 
[image]

WouterJD, 2021-03-04:
@hasise, welcome!
> The comment "The GUI is still freezing" is a bit scary
The GUI is working fine on RPI4 (I don't have a 3), So give it a try.

WouterJD, 2021-03-04:
@MeanHat @TacxBiker @decodeais 
Branch Raspberyy_Release6 is published.

Currently I don't get BLE work; see https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-791026888.

To be continued in a new fresh day.

MeanHat, 2021-03-07:
@WouterJD - as requested in #192, I have updated the miniTFT Controller to v2 (https://github.com/MeanHat/FortiusANT_controller/blob/main/FA_Controller_v2.py). This now runs better, works in parallel with your FortiusAnt.py (rather than calling FortiusAnt.py as in v1), fixes the c10min Bluetooth drop out issue, fixes some other minor issues and keeps the Wastebasket under control (to prevent it becoming bloated with old logfiles).
Next step is to add ANT functionality (with display text to reflect this). Further steps may include addition of other head units once I can get a replacement T1902.

WouterJD, 2021-03-08:
Looking for
> a replacement T1902
?

Look at https://www.marktplaats.nl/q/tacx+fortius/ since Fortius full config offered there from EUR 25,= up :-)

I would almost think of buying them, add an RPI and sell as a Smart Trainer; but holding back because of support issues I would probably get.

MeanHat, 2021-03-08:
> Looking for
> 
> > a replacement T1902
> 
> ?
> 
> Look at https://www.marktplaats.nl/q/tacx+fortius/ since Fortius full config offered there from EUR 25,= up :-)
> 
> I would almost think of buying them, add an RPI and sell as a Smart Trainer; but holding back because of support issues I would probably get.

Thanks @WouterJD - That's a great price. I tried getting one from Germany in January but it's proving difficult persuading people to ship to the UK :(
I think you could make a good business out of refurbishing old Tacx trainers and fitting them with an RPi for use with CTPs! There is a strong demand. The other thing I have noticed is that many old trainers come with Tacx DVDs that resell for more than the combined price of the DVDs plus the trainer. That was the case for the Fortius I bought.
I'm hopefully close to securing a T1902 head unit... Just need to get BLE running on an RPi 0 (slowly working through all the errors).

MeanHat, 2021-03-09:
> Thanks @WouterJD - That's a great price. I tried getting one from Germany in January but it's proving difficult persuading people to ship to the UK :(
> I think you could make a good business out of [...]

Have successfully acquired a T1902 head unit

TacxBiker, 2021-03-17:
Hi,

step by step.. It took a while before I got the LCD display up and running. 
The idea is to get the main data on the screen: 
- actual power
- requested power
- speed
- cadence
- status text ("give kick to start calibrating" and so on)
- Virtual gear with a indicator above en below the gears indicating start gear and between the gears for the actual gear

The question I have (besides a lot of other things I need to handle) is how do I pass this information as supposed between the python scripts.
All the needed data is in fortiusANT and as @MeanHat is doing by pulling it out of the log file isn't working. not all the data is in the log file is it?

[image]

WouterJD, 2021-03-17:
Good work! I'm busy on some other business.
Will get back to this soon.

If you want to move ahead; SetValues() is the link between FortiusAntBody and FortiusAntGUI; you could start there.

TacxBiker, 2021-03-17:
Thanks, will look in to that. And see if I can understand the code.

As far the raspberry and all the differtent options in configuration is it probably better to use a generic system to get the data instead of make code into fortiusANT for all the different  screens/LCD's/OLED/ePaper

some sort of API? or is something like that a step to far?

MeanHat, 2021-03-17:
@TacxBiker - although it's not very elegant to interrogate the log file (created by - d1 parameter and therefore any odd number) it  contains speed, heart rate, current power, cadence and CTP requested target which was the data you wanted?

TacxBiker, 2021-03-17:
Hi MeanHat,

Of course it works. and independant of the FortiusANT code. It's more -like you say- not very elegant.
And (more important) form my understaning and learning python I'd like to give it a try the right way and learn how scripts communicate to each other

MeanHat, 2021-03-17:
Hi @TacxBiker - I'd like to get the display using data from the running python rather than the log file too. I was not successful launching FortiusAnt from my code (it needed more work for sure) and I did not want to modify the core FortiusAnt.py program(s). However, now that @WouterJD has just released the peripheral GPIO program I wondered whether I might modify that to drive the miniTFT display rather than LEDs. I would, of course, be very interested in your solution.

WouterJD, 2021-03-18:
My TFT has arrived; I will see how to integrate.
By the way; why not implement virtual leds on the TFT; five "bullets" in a row on the screen.

TacxBiker, 2021-03-18:
Nice!
for the record, what kind of TFT did you order?

And 5 vitual led's would be perfect, I was thinking on the LCD to use a dot and inverse dot

WouterJD, 2021-03-18:
https://nl.aliexpress.com/item/1005001685654439.html?spm=a2g0s.9042311.0.0.53334c4dPecGL4

WouterJD, 2021-04-13:
Also check the latest  sw branch which is close to getting released; leds are on GUI and TFT

TacxBiker, 2021-04-14:
Hi Wouter,

I did check the latest branch. The /raspberry/Test Raspberry Pi LEDS.sh works kind a. It shows the logo on the tft and rons a led test. then it stops. Buttons do not work.

I tried several things, running Fortius ANT and manualy  start raspberry.py no differents.

But as i am looking no there is a #5 version. I Tried version #41 

It seem for using the display the -l command is enough (as specified in the raspberry/FortiusANT.sh)

WouterJD, 2021-04-14:
leds: -l
display: -O display/rotation - rotation in degrees
leds/button mapping can be adjusted with -L

It works (on my raspberry pi 4) and I'm working on manual

It may be that the test-version (main code in raspberry.py) is not optimal.

TacxBiker, 2021-04-16:
It works also on a raspberry 3b. Very good work!!
A little testing between the outdoor rides at this moment, but very nice!
I had to play around with the command line options. I ended with : -a -b -g -O display/180 -D-1 -s 
just like you told.. 

I tweaked the output a little so that the gears are displayed and the requested power and actual power ar on the same line:

raspberry.py line 794 - 797

            self._DrawTextTable ( [ [ 'Speed'  , s                    ],\
                                    [ 'Cadence', "%i/min"    % iRevs  ],\
                                    [ 'Power', "%i" %iPower + " " + t       ],\
                                    [ 'Gear ', str(iCassetteIndex) + " " +str(iCrancksetIndex)]], True)

I still can't understand the code exacly , like what is happing with the % marks.
And I can't figure out why these line are with space in between (so the 5th line is over the LED's) 
Also the gear positions is displays not the tooth like the real fortiusANT gui.
I think i did found the code how that is converted, so that could be done on the little display as well.

The buttons aren't working. I haven't figure out why that is. or either I'm doning something wrong.

WouterJD, 2021-04-18:
"%i/min"    % iRevs 

% is an operator (like +, -, *)
For each occurrence of % in the string, a variable is filled
If there are multiple %formats, the variables must be enclosed in ()

See https://docs.python.org/2/library/stdtypes.html#string-formatting

WouterJD, 2021-04-18:
> And I can't figure out why these line are with space in between (so the 5th line is over the LED's)

See line 611; the last parameter you provide (True) governs this; but don't change to False that does not work (then the second parameter is the color). Perhaps not the most beautiful solution, but at times I also chose for quick over beauty.

```
                self.draw.text( (  0, 20 + i * 46), t[i][0], font=self.fontS, fill=GREY)
                self.draw.text( (120, 20 + i * 46), t[i][1], font=self.fontS, fill=WHITE)

```

Reduce 46 to some less and then you have less spacing; in your case 4/5 * 46 = 36 would do.

WouterJD, 2021-04-18:
> Also the gear positions is displays not the tooth like the real fortiusANT gui.

See FortiusAntGui.py#L1066

WouterJD, 2021-04-18:
> The buttons aren't working. I haven't figure out why that is. or either I'm doning something wrong.

I cannot help you from here :-)

Succes!

WouterJD, 2021-04-22:
@TacxBiker download current version; many TFT enhances. Will be merged into master shortly

TacxBiker, 2021-04-23:
Hi Wouter,

Thanks for the enhances. I was edditing the rasperry.py with your tips and I got it to work. In the meanwhile I see that you add the gear as well. thanks for that.
The onl thing I ran into was that the display was working well in simulate mode but not on the real trainer. It did the calibration but the display didn't go the the values screen.

Tonight I will try the new version.

WouterJD, 2021-04-23:
👍

TacxBiker, 2021-04-26:
It took a while. but I can confirm it works pefect. 
I only did a test ride but it seems perfectly

WouterJD, 2021-04-29:
Implemented in RaspBerryDisplay branch

WouterJD, 2021-04-29:
Released

closed 2021-04-29 by WouterJD